nordicsemi/IOS-nRF-Mesh-Library #168: Setting or disabling Relay

icarus31, 2019-10-03:
**Describe the bug**
It is not really a bug, but mostly a question related to the Relay that can be configured from Configuration Server.

If I set to 1 transmission, the result will be 2 transmissions because the spec says so. How can I set it to 0 retransmission? The slider does not allow me to do that.

Also, a suggestion: maybe you should have a slider label (at the top of it?) that display the number of retransmission and keep the one at the left side of the slider with the total of transmissions. It is bit confusing the first time you use that section so maybe it could help?

**To Reproduce**
N/A

**Expected behavior**
N/A

**Platform details:**
N/A

**Logs / Screenshots**
N/A

icarus31, 2019-10-09:
**Suggestions**

node.swift

```
internal init(_ request: ConfigRelaySet) {
    self.count = request.count  <--
    self.interval = UInt16(request.steps + 1) * 10
}
        
internal init(_ status: ConfigRelayStatus) {
    self.count = status.count  <--
    self.interval = UInt16(status.steps + 1) * 10
}
```

ConfigurationServerViewCell.swift

```
    @IBAction func relayCountDidChange(_ sender: UISlider) {
        let value = Int(sender.value)
        relayIntervalSlider.isEnabled = value > 1
        if value == 0 {
            relayCountLabel.text = "\(value) retransmission"
            relayIntervalLabel.text = "N/A"
        } else {
            relayCountLabel.text = "\(value) retransmissions"
            relayIntervalLabel.text = "\(Int(relayIntervalSlider.value) * 10) ms"
        }
    }
```

If the user put 0 retransmission, it means total of transmission will be 1 and no retransmission. Like written in the Bluetooth mesh profile documentation.

What do you think?

philips77, 2019-10-10:
I'll address is when I'm back from vacation. Thanks for the feedback!

philips77, 2019-11-01:
Hi @icarus31,
This is the most I could do. The database spec is something we can't modify, and it defines `count` as values 1-8, don't ask me why :)
But I fixed some comments and other issues I fond related to relay retransmit and network transmission objects.
This will be released in 2.0.1. The fix does not modify any behaviour, just clarifies the documentation.

closed 2019-11-01 by philips77

icarus31, 2019-11-03:
Just by curiosity, when you say 'database spec', you meant your database or the Bluetooth mesh spec? 

If Bluetooth mesh spec, where did you see the count within 1 and 8? Based on the Mesh Profile document (p 153), it says:

> For example, a value of 0b000 represents a single transmission with no retransmissions, and a value of 0b111 represents a single transmission and 7 retransmissions for a total of 8 transmissions. 

So, putting count to 0 = 1 transmission. Putting count to 1 = 1 transmission + 1 retransmission.

By the way, this is my understanding. Makes sense?

philips77, 2019-11-03:
We're using a database spec by Bluetooth SIG. You're a member of the sig, you may have access to it. It will be public when adoption is finished, until then we can't say more and it may change.