nordicsemi/IOS-nRF-Mesh-Library #169: Cannot Find nRF52DK with Light and Switch Merged Firmware

bobtaylorlsi, 2019-10-03:
Hi, cannot find in the Provision Device screen, the 2 nRF52DK running the firmware for the light or the switch. When I use the use the nRF Connect app, I see one nRF5x Mesh Switch (Connectable) with Mesh Proxy Service 1828 and one nRF5x Mesh Light (Connectable) with Mesh Proxy 1828. Please let me know what I missed.

Thanks
Bob

philips77, 2019-10-03:
Looks like your devices have already been provisioned? Otherwise they would advertise as unprovisioned devices with Mesh Provisioning Service instead. Try wiping out the memory with "nrfjprog -e" and flashing fw again.

philips77, 2019-10-09:
Hi, did you manage to solve the issue?

bobtaylorlsi, 2019-10-09:
Hi, no still unable to find the nodes - here is where I am now:

I erased the two PCA10040 with the nRF Connect (latest version) but I cannot figure out a way to flash the nRF52 dev kit. Was able on Monday with the previous version of j-link. However, I updated the j-link and now it will not do a drag-n-drop flash when I use your example merged hex files. nRF Connect gives an error when I try to use it to program. Any suggestions for how to flash the nRF52 on the dev kit?

bobtaylorlsi, 2019-10-09:
Hi, also tried nrfjprog, erase all, and then tried --program and I get a message that the example client and server merged hex files are not valid hex files (done individually)

philips77, 2019-10-10:
Check your USB connection. Perhaps connect to a different port. Also, for such questions, please use the DevZone, as this is not related to the iOS library.

closed 2019-10-15 by bobtaylorlsi