nordicsemi/IOS-nRF-Mesh-Library #175: Espressif timeout on sending

fiveagle, 2019-10-11:
**Describe the bug**
Hi,
I'm having some issue sending a CompositionDataGet to a Espressif gateway. It goes in TimeOut but reading the Gateway firmware log it looks like it is receiving too much segments. It continuously prints:

W (43991) BLE_MESH: Got segment for already complete SDU

and ends with:

W (60001) BLE_MESH: Client wrote 0x0000 instead enabling notify
W (99991) BLE_MESH: Client wrote 0x0000 instead enabling notify
W (140041) BLE_MESH: Client wrote 0x0000 instead enabling notify 

Thanks in advance for any help.

**Platform details:**
 - Device: iPhone8
 - OS:  iOS12.4.1
- Version: Last Commit e9be842730a0c0ae472279aa8d7e90115dc90ade

**Logs / Screenshots**

Nordic LOG:
W (384321) BLE_MESH: Got segment for already complete SDU
E (384821) BLE_MESH: send_seg, No multi-segment message contexts available
E (384821) BLE_MESH: dev_comp_data_get, Unable to send Config Composition Data Status
3:55 PM
W (443631) BLE_MESH: Client wrote 0x0000 instead enabling notify

Gateway Log:
I (647581) blemesh_prov_event_to_str:  EVENT: 4
I (647581) blemeshclient_prov_cb: event = 'Establish a BLE Mesh link event'
I (647581) blemeshclient_prov_cb: Node:Link:Open OK, bearer PB-GATT
W (648401) BLE_MESH: bt_mesh_attention, No Health Server context provided
I (648731) BLE_MESH: Primary Element: 0x0002
W (648741) BLE_MESH: bt_mesh_attention, No Health Server context provided
E (648761) BT_BTM: BTM_BleScan scan not active
E (648761) BLE_MESH: bt_le_scan_stop, Invalid status 6
E (648761) BLE_MESH: stopping scan failed (err -1)
I (648771) blemesh_prov_event_to_str:  EVENT: 5
I (648771) blemeshclient_prov_cb: event = 'Close a BLE Mesh link event'
I (648781) blemeshclient_prov_cb: ESP_BLE_MESH_NODE_PROV_LINK_CLOSE_EVT, bearer PB-GATT
I (648791) blemesh_prov_event_to_str:  EVENT: 10
I (648791) blemeshclient_prov_cb: event = 'Provisioning done event'
I (648801) blemeshclient_prov_complete: P406 Gateway is provisioned on mesh network
I (648811) blemeshclient_prov_complete: netkey index: 0x0025. gateway unicast address: 0x0002.
I (648821) blemeshclient_prov_complete: flags: 0x00, iv_index: 0x00000000
I (648831) NetKey: 53 b7 67 45 e8 88 aa 3f 35 36 13 19 8d 30 5d 8b
I (648831) fsm_event_handler: got event MESH_NODE_PROVISIONED
I (648841) ble_handler: registering netkey
I (648841) ble_handler: 53 b7 67 45 e8 88 aa 3f 35 36 13 19 8d 30 5d 8b
I (648851) BLE_GATT_service: custom_gatt_server_init: done
E (648861) nvs_init_and_open: Cannot open NVS 'plantconfnvs' with error: ESP_ERR_NVS_NOT_FOUND
I (648871) dispatch: old state 1
I (648871) dispatch: new state 2
I (648881) BLE_GATT_service: ESP_GATTS_CONNECT_EVT, conn_id = 0
I (648881) BLE_GATT_service: 69 96 f5 fb 83 dc
I (648891) BLE_GATT_service: create attribute table successfully, the number handle = 6
I (648901) BLE_GATT_service: SERVICE_START_EVT, status 0, service_handle 52
E (656531) BLE_MESH: send_seg, No multi-segment message contexts available
E (656531) BLE_MESH: dev_comp_data_get, Unable to send Config Composition Data Status
W (658521) BLE_MESH: Got segment for already complete SDU
W (659481) BLE_MESH: Got segment for already complete SDU
W (659981) BLE_MESH: Got segment for already complete SDU
W (700711) BLE_MESH: Client wrote 0x0000 instead enabling notify
W (730131) BT_GATT: attribute value too long, to be truncated to 20
W (740691) BLE_MESH: Client wrote 0x0000 instead enabling notify
I (748171) BLE_GATT_service: ESP_GATTS_DISCONNECT_EVT, reason = 0x13

philips77, 2019-10-11:
If it's not a duplicate of 174, all this question on Nordic DevZone, as this is more related to the firmware side. Or provide logs from the iOS mesh library if you think the issue is in this lib.

philips77, 2019-10-22:
Hello,
Do you have logs from the iOS library as well?

fiveagle, 2019-10-25:
Hi,
with the last commits it now stops before composition data.
It prints:

Error: Local Provisioner has no Unicast Address assigned

when binding the AppKey.

Apparently the provisioner node is not inside the nodes array of MeshNetwork and the library fails getting its unicastAddress in row 469 of MeshNetworkManager.

Thank You.

philips77, 2019-10-25:
When you create a new network, using `manager.createNewMeshNetwork(name;by)`, the first provisioner is automatically added. It gets the first unicast address from the address range of the given provisioner. If you import a network on another phone, which has never had this network, you have to add a new provisioner and assign an address to it. Have a look at: 
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/c6ba36d900ebe6b16a65d3e599a224a27ddf5a80/Example/nRFMeshProvision/View%20Controllers/Settings/SettingsViewController.swift#L201-L220

fiveagle, 2019-10-25:
I'm not importing a meshNetwork yet. The localProvisioner is the actual phone the app is running on. But it is not inside the node array. I have deleted and rebuild the app but it's the same.
The strange thing is that I successfully managed to provision a mesh device and that the gateway espressif logs indicates that it has been successfully provisioned.

Thanks

philips77, 2019-10-25:
Node is not necessary for Provisioning, just for configuration. To assign am address and create node for Provisioner use `network.assign(address;for)`>

philips77, 2019-10-25:
Can you post a snippet how do you create the network and Provisioner?

fiveagle, 2019-10-25:
Same as the Sample App.

func createNewMeshNetwork() {
        // TODO: Implement creator
        connection?.close()
        
        let provisioner = Provisioner(name: UIDevice.current.name,
                                      allocatedUnicastRange: [AddressRange(0x0001...0x199A)],
                                      allocatedGroupRange:   [AddressRange(0xC000...0xCC9A)],
                                      allocatedSceneRange:   [SceneRange(0x0001...0x3333)])
        
        _ = meshNetworkManager.createNewMeshNetwork(withName: "P406MeshNetwork", by: provisioner)
        
        let network = meshNetworkManager.meshNetwork!
        let address = MeshAddress(0xC000)
        
        do {
            let group = try Group(name: "P406MeshGroup", address: address)
            try network.add(group: group)
            
        } catch {
            switch error as! MeshNetworkError {
            case .invalidAddress:
                print("The address \(address.debugDescription) is not a valid group address.")
            case .groupAlreadyExists:
                print("Group with address \(address.debugDescription) already exists.")
            default:
                print("An error occurred while creating the mesh group.")
            }
        }
        
        _ = meshNetworkManager.save()
        
        meshNetworkDidChange()
    }
    
    /// Sets up the local Elements and reinitializes the `NetworkConnection`
    /// so that it starts scanning for devices advertising the new Network ID.
    func meshNetworkDidChange() {
        connection?.close()
        
        let meshNetwork = meshNetworkManager.meshNetwork!
        
        // Set up local Elements on the phone.
        let element0 = Element(name: "Primary Element", location: .first, models: [
            Model(sigModelId: 0x1000, delegate: GenericOnOffServerDelegate()),
            Model(sigModelId: 0x1002, delegate: GenericLevelServerDelegate()),
            Model(sigModelId: 0x1001, delegate: GenericOnOffClientDelegate()),
            Model(sigModelId: 0x1003, delegate: GenericLevelClientDelegate())
        ])
        let element1 = Element(name: "Secondary Element", location: .second, models: [
            Model(sigModelId: 0x1000, delegate: GenericOnOffServerDelegate()),
            Model(sigModelId: 0x1002, delegate: GenericLevelServerDelegate()),
            Model(sigModelId: 0x1001, delegate: GenericOnOffClientDelegate()),
            Model(sigModelId: 0x1003, delegate: GenericLevelClientDelegate())
        ])
        meshNetworkManager.localElements = [element0, element1]
        
        connection = NetworkConnection(to: meshNetwork)
        connection!.dataDelegate = meshNetworkManager
//        connection!.logger = self
        meshNetworkManager.transmitter = connection
        connection!.open()
    }

fiveagle, 2019-10-28:
Hi,
apparently I deleted the provisioner node while retrieving the node with address 2 but getting instead the node at address 1. Looks like this code below enabled this behaviour. Is this an expected behaviour? Should I be allowed to delete the provisioner node?

Anyway now I get` Request timed out` on send ConfigAppKeyAdd, both on gateway and on the devices. I have set `acknowledgmentMessageTimeout` to 120.

ThankYou very much.

```
var lastUnicastAddress: UInt16 {
        // Provisioner may not have any elements
        let allocatedAddresses = Address(elementsCount > 0 ? elementsCount : 1)
        return unicastAddress + allocatedAddresses - 1
    }
```

philips77, 2019-10-29:
Yes, it is possible to remove a Provisioner, even the last one. But this code you gave shouldn't do anything, it's not mutating.
For the second issue please post some logs.

fiveagle, 2019-10-29:
Ok. This is when I set 120 as timeout

```
Adding Application Key...
2019-10-28 17:58:59.488338+0100 P406[1928:1440681] [FoundationModel] Sending ConfigAppKeyAdd(networkKeyIndex: 10, applicationKeyIndex: 0, key: 16 bytes) to: 000B
2019-10-28 17:58:59.491127+0100 P406[1928:1440681] [Access] Sending Access PDU (opcode: 0x 0, parameters: 0x0A00001A9718F774F5341608E4B5B91B622165)
2019-10-28 17:58:59.497024+0100 P406[1928:1440681] [UpperTransport] Sending Upper Transport PDU (encrypted data: 0x26C7AB53B094B0925810639D59A004225B98CF2A, transMic: 0x249AB97D) encrypted using key: 1888_592's Device Key
2019-10-28 17:58:59.499186+0100 P406[1928:1440681] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 282, segO: 0, segN: 1, data: 0x26C7AB53B094B0925810639D)
2019-10-28 17:58:59.501745+0100 P406[1928:1440681] [Network] Sending Network PDU (ivi: 0, nid: 0x58, ctl: 0, ttl: 5, seq: 282, src: 0001, dst: 000B, transportPdu: 0xF985F108E88155C3BD4262FDDB7333E3, netMic: 0xB10D6DD1) encrypted using propels Network Key (index: 10)
2019-10-28 17:58:59.502967+0100 P406[1928:1440681] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 282, segO: 1, segN: 1, data: 0x59A004225B98CF2A249AB97D)
2019-10-28 17:58:59.505711+0100 P406[1928:1440681] [Network] Sending Network PDU (ivi: 0, nid: 0x58, ctl: 0, ttl: 5, seq: 283, src: 0001, dst: 000B, transportPdu: 0x985E1918D68AF548C5DCC3EF9480C9D2, netMic: 0x6A5BA897) encrypted using propels Network Key (index: 10)
2019-10-28 17:59:00.056036+0100 P406[1928:1440896] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 282, segO: 0, segN: 1, data: 0x26C7AB53B094B0925810639D)
2019-10-28 17:59:00.058835+0100 P406[1928:1440896] [Network] Sending Network PDU (ivi: 0, nid: 0x58, ctl: 0, ttl: 5, seq: 284, src: 0001, dst: 000B, transportPdu: 0x035698264AF157A4C93B26F32F22B4B2, netMic: 0x9A8C6B85) encrypted using propels Network Key (index: 10)
2019-10-28 17:59:00.060127+0100 P406[1928:1440896] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 282, segO: 1, segN: 1, data: 0x59A004225B98CF2A249AB97D)
2019-10-28 17:59:00.063845+0100 P406[1928:1440896] [Network] Sending Network PDU (ivi: 0, nid: 0x58, ctl: 0, ttl: 5, seq: 285, src: 0001, dst: 000B, transportPdu: 0xB86925B73242775752B8220C43013BF7, netMic: 0x6F65F63E) encrypted using propels Network Key (index: 10)
2019-10-28 17:59:00.610150+0100 P406[1928:1440819] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 282, segO: 0, segN: 1, data: 0x26C7AB53B094B0925810639D)
2019-10-28 17:59:00.613560+0100 P406[1928:1440819] [Network] Sending Network PDU (ivi: 0, nid: 0x58, ctl: 0, ttl: 5, seq: 286, src: 0001, dst: 000B, transportPdu: 0xC2BEAAD6538C7815987A37E3F5C58B6D, netMic: 0x77171BA0) encrypted using propels Network Key (index: 10)
2019-10-28 17:59:00.615112+0100 P406[1928:1440819] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 282, segO: 1, segN: 1, data: 0x59A004225B98CF2A249AB97D)
2019-10-28 17:59:00.618671+0100 P406[1928:1440819] [Network] Sending Network PDU (ivi: 0, nid: 0x58, ctl: 0, ttl: 5, seq: 287, src: 0001, dst: 000B, transportPdu: 0x59090A13B7E1D38D8CA1A1B35EE723B4, netMic: 0xCF1FCA15) encrypted using propels Network Key (index: 10)
2019-10-28 17:59:01.160200+0100 P406[1928:1440819] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 282, segO: 0, segN: 1, data: 0x26C7AB53B094B0925810639D)
2019-10-28 17:59:01.163653+0100 P406[1928:1440819] [Network] Sending Network PDU (ivi: 0, nid: 0x58, ctl: 0, ttl: 5, seq: 288, src: 0001, dst: 000B, transportPdu: 0x64E5460C5E60F8F14B8D495745080C6F, netMic: 0x1C4ED542) encrypted using propels Network Key (index: 10)
2019-10-28 17:59:01.165397+0100 P406[1928:1440819] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 282, segO: 1, segN: 1, data: 0x59A004225B98CF2A249AB97D)
2019-10-28 17:59:01.168887+0100 P406[1928:1440819] [Network] Sending Network PDU (ivi: 0, nid: 0x58, ctl: 0, ttl: 5, seq: 289, src: 0001, dst: 000B, transportPdu: 0x0E103CF792637845D55EF1C6F5F17582, netMic: 0x7872A627) encrypted using propels Network Key (index: 10)
2019-10-28 17:59:01.718256+0100 P406[1928:1440819] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 282, segO: 0, segN: 1, data: 0x26C7AB53B094B0925810639D)
2019-10-28 17:59:01.722104+0100 P406[1928:1440819] [Network] Sending Network PDU (ivi: 0, nid: 0x58, ctl: 0, ttl: 5, seq: 290, src: 0001, dst: 000B, transportPdu: 0x638D67C5B6D583087ED0696E43EBF590, netMic: 0xDC27EA92) encrypted using propels Network Key (index: 10)
2019-10-28 17:59:01.723610+0100 P406[1928:1440819] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 282, segO: 1, segN: 1, data: 0x59A004225B98CF2A249AB97D)
2019-10-28 17:59:01.727488+0100 P406[1928:1440819] [Network] Sending Network PDU (ivi: 0, nid: 0x58, ctl: 0, ttl: 5, seq: 291, src: 0001, dst: 000B, transportPdu: 0xD02156240EABA9A8563169A71539A969, netMic: 0xC3CF5884) encrypted using propels Network Key (index: 10)
2019-10-28 17:59:02.276882+0100 P406[1928:1440819] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 282, segO: 0, segN: 1, data: 0x26C7AB53B094B0925810639D)
[... cut ...] etwork Key (index: 10)
2019-10-28 17:59:18.519983+0100 P406[1928:1440945] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 306, segO: 1, segN: 1, data: 0xA3BDB3BE02BCC3E1DA048FB0)
2019-10-28 17:59:18.523837+0100 P406[1928:1440945] [Network] Sending Network PDU (ivi: 0, nid: 0x58, ctl: 0, ttl: 5, seq: 317, src: 0001, dst: 000B, transportPdu: 0x6CF6D8952F63E69B26086C2B62E09365, netMic: 0x11423859) encrypted using propels Network Key (index: 10)
Connection did close
```

philips77, 2019-10-29:
Do you have the `transmitter` set to a Bearer and `bearer.delegate` to `MeshNetworkManager`? Looks like either you are not sending any data, or the received data are not passed to the manager.

philips77, 2019-10-29:
You may also set the `logger` to the `NetworkConnection`, or the a `GattBearer` that you are using to get bearer logs as well.

fiveagle, 2019-10-29:
I have been updating my `NetworkConnection` with yours from the Sample app. 
I have introduced a delay between the provisioning and the first config send. I am pretty sure that my devices could and probably needs, to reboot during the process. When it happens I don't get any `GattBearerDelegate` calls in my class notifying the node is connected again. And then my process gets hanged till the timeout. I use the `whenConnected` method from your sample app.
My delegates seems to be set correctly because I receive the outside the configurations. I don't receive any callback also when I try to send a `VendorMessage`. 
When my other nordic devices don't reboot the config process goes successfully to the end.

Thansks

philips77, 2019-10-29:
Oh yes, after the device is provisioned it will reboot and needs some time before it starts advertising again. But this takes around 10 seconds in our mesh sdk 3.2. Then the connection, which also takes time as the connection interval is long. But eventually Composition Get succeeds.

philips77, 2019-11-04:
Hi, can this issue be closed now?

closed 2019-11-04 by fiveagle