nordicsemi/IOS-nRF-Mesh-Library #176: Importing Mesh json doesnt control Generic On Off model

sunilloharshoreline1, 2019-10-13:
If the mesh json file is imported into the Example project, it doesn't control the Generic On-Off control however the App through which the Mesh is created can control the Generic On-Off model.

PM-BubblyNet, 2019-10-16:
hi, i think you need to change the provisioner source address, the original / default os 7FFF so something like 6FFF would work I think?

philips77, 2019-10-17:
Yes, this may be the issue. Reach phone should be set as a new Provisioner with a unique unicast address. Before exporting you should create a new Provisioner in Settings, assign a unique address and ranges, than export and import on another phone. Go to Settings -> Provisioners, click Edit and drag the newly created one to the top to set it as a local one. Reach phone must be a different Provisioner as they must increment their sequence numbers Independently.

philips77, 2019-10-21:
Did you manage to solve your problem?

sunilloharshoreline1, 2019-10-23:
I haven't got time to try this. I will post my findings in couple of days. Thanks for your reply.

philips77, 2020-04-23:
I'm closing the issue due to inactivity. The possible answer has been given above.

closed 2020-04-23 by philips77