nordicsemi/IOS-nRF-Mesh-Library #177: How to config relay to all nodes (0xFFFF)

icarus31, 2019-10-14:
As a user, I want to be able to send Config Relay message to all nodes so that those nodes can be reconfigured to the new value.

I need some advice to how I could achieve that.

The function called looks like this:

```
delegate?.send(ConfigRelaySet(count: count, steps: steps),
               element: MeshNetworkManager.instance.localElements[0],
               destination: MeshAddress(0xFFFF),
               ttl: 127,
               applicationKey: MeshNetworkManager.instance.meshNetwork!.applicationKeys[0],
               description: "Sending Relay settings...")
```

This, always gives me 'Discarding packet' error:

```
[Model] Sending ConfigRelaySet(state: Enabled, count: 1, steps: 1) from: Primary Element (0001), to: FFFF
[Access] Sending Access PDU (opcode: 0x8027, parameters: 0x0109)
[UpperTransport] Sending Upper Transport PDU (encrypted data: 0x67757AC9, transMic: 0xA55F8559) encrypted using key: App Key 1 (index: 0)
[LowerTransport] Sending Segmented Access Message (akf: 1, aid: 0x12, szmic: 0, seqZero: 1466, segO: 0, segN: 0, data: 0x67757AC9A55F8559)
[Network] Sending Network PDU (ivi: 0, nid: 0x03, ctl: 0, ttl: 127, seq: 1466, src: 0001, dst: FFFF, transportPdu: 0xC56A1AD8A94CC9D199A36ED7, netMic: 0x8DFA430D) encrypted using Primary Network Key (index: 0)
[Network] Network PDU (ivi: 0, nid: 0x03, ctl: 0, ttl: 127, seq: 1466, src: 0001, dst: FFFF, transportPdu: 0xC56A1AD8A94CC9D199A36ED7, netMic: 0x8DFA430D) received
[LowerTransport] Segmented Access Message (akf: 1, aid: 0x12, szmic: 0, seqZero: 1466, segO: 0, segN: 0, data: 0x67757AC9A55F8559) receieved (decrypted using key: Primary Network Key (index: 0))
[LowerTransport] Access Message (akf: 1, aid: 0x12, szmic: 0, data: 0x67757AC9A55F8559) received
[UpperTransport] Upper Transport PDU (encrypted data: 0x67757AC9, transMic: 0xA55F8559) received
[Access] Access PDU (opcode: 0x8027, parameters: 0x0109) receieved (decrypted using key: App Key 1 (index: 0))
[FoundationModel] ConfigRelaySet(state: Enabled, count: 1, steps: 1) received from: 0001, to: FFFF
[Bearer] -> 0x0003786FE160D2171132C56A1AD8A94CC9D199A36ED78DFA430D
[LowerTransport] Sending Segmented Access Message (akf: 1, aid: 0x12, szmic: 0, seqZero: 1466, segO: 0, segN: 0, data: 0x67757AC9A55F8559)
[Network] Sending Network PDU (ivi: 0, nid: 0x03, ctl: 0, ttl: 127, seq: 1467, src: 0001, dst: FFFF, transportPdu: 0x35F732A2B0527AF75A999CD7, netMic: 0xAEBC980E) encrypted using Primary Network Key (index: 0)
[Network] Network PDU (ivi: 0, nid: 0x03, ctl: 0, ttl: 127, seq: 1467, src: 0001, dst: FFFF, transportPdu: 0x35F732A2B0527AF75A999CD7, netMic: 0xAEBC980E) received
[LowerTransport] Discarding packet (seqAuth: 1466, expected > 1466)
[Bearer] -> 0x00038124AB62F862C97735F732A2B0527AF75A999CD7AEBC980E
[Access] Resending Access PDU (opcode: 0x8027, parameters: 0x0109)
[UpperTransport] Sending Upper Transport PDU (encrypted data: 0xF4768293, transMic: 0xCA4EF2AB) encrypted using key: App Key 1 (index: 0)
[LowerTransport] Sending Segmented Access Message (akf: 1, aid: 0x12, szmic: 0, seqZero: 1468, segO: 0, segN: 0, data: 0xF4768293CA4EF2AB)
[Network] Sending Network PDU (ivi: 0, nid: 0x03, ctl: 0, ttl: 127, seq: 1468, src: 0001, dst: FFFF, transportPdu: 0xDFE870D36ECD44EACC748036, netMic: 0x6A5FBABC) encrypted using Primary Network Key (index: 0)
[Network] Network PDU (ivi: 0, nid: 0x03, ctl: 0, ttl: 127, seq: 1468, src: 0001, dst: FFFF, transportPdu: 0xDFE870D36ECD44EACC748036, netMic: 0x6A5FBABC) received
[LowerTransport] Segmented Access Message (akf: 1, aid: 0x12, szmic: 0, seqZero: 1468, segO: 0, segN: 0, data: 0xF4768293CA4EF2AB) receieved (decrypted using key: Primary Network Key (index: 0))
[LowerTransport] Access Message (akf: 1, aid: 0x12, szmic: 0, data: 0xF4768293CA4EF2AB) received
[UpperTransport] Upper Transport PDU (encrypted data: 0xF4768293, transMic: 0xCA4EF2AB) received
[Access] Access PDU (opcode: 0x8027, parameters: 0x0109) receieved (decrypted using key: App Key 1 (index: 0))
[FoundationModel] ConfigRelaySet(state: Enabled, count: 1, steps: 1) received from: 0001, to: FFFF
[Bearer] -> 0x0003AE73E953EBDBC554DFE870D36ECD44EACC7480366A5FBABC
[LowerTransport] Sending Segmented Access Message (akf: 1, aid: 0x12, szmic: 0, seqZero: 1468, segO: 0, segN: 0, data: 0xF4768293CA4EF2AB)
[Network] Sending Network PDU (ivi: 0, nid: 0x03, ctl: 0, ttl: 127, seq: 1469, src: 0001, dst: FFFF, transportPdu: 0x65AD5044EBB95B40FD14FA6E, netMic: 0x85914087) encrypted using Primary Network Key (index: 0)
[Network] Network PDU (ivi: 0, nid: 0x03, ctl: 0, ttl: 127, seq: 1469, src: 0001, dst: FFFF, transportPdu: 0x65AD5044EBB95B40FD14FA6E, netMic: 0x85914087) received
[LowerTransport] Discarding packet (seqAuth: 1468, expected > 1468)
[Bearer] -> 0x00030E0276D8637AFE9765AD5044EBB95B40FD14FA6E85914087
```

Thanks

philips77, 2019-10-17:
Hi, this is not possible by the spec. Config messages, including `ConfigRelaySet` must be signed with a device key, not application key, therefore must be sent to a single unicast address of a primary element in each device. You have to send one message per node. And use the other method, which takes Config Messages.

closed 2019-10-17 by philips77

icarus31, 2019-10-17:
Is it the same for the vendor messages?

philips77, 2019-10-17:
No, generic or vendor messages are signed with an app key, so they may be sent to any address. Only Config messages, that are sent to Configuration Server or Client use device key and may only be sent by the Provisioner, which knows the keys.

icarus31, 2019-10-17:
All the great information that you just gave me is inside the Bluetooth mesh profile document?

philips77, 2019-10-18:
Yes it is. It reads like a we written novel ;)