nordicsemi/IOS-nRF-Mesh-Library #179: When I frequently send the switch control command for a while, the SDK crashes. Can you help me?

codepgq, 2019-10-23:
[image]

``` swift
    lazy var btnONOFF: PQButton = {
        let view = PQButton()
        //view.backgroundColor = UIColor.gray
        view.setImage(UIImage(named: "button_off"), for: .normal)
        view.setImage(UIImage(named: "button_on"), for: .selected)
        view.buttonClick { [weak self] (button) in
            self?.colorPanelView.isHasOverlay = button.isSelected
            button.isSelected = !button.isSelected
            
            /// send color
            if let model = self?.node.onoffModel {
                let colorMsg = GenericOnOffSet(button.isSelected)
                do {
                    try MeshNetworkManager.instance.send(colorMsg, to: model)
                } catch {
                    print("send  onoff failed", error)
                }
            }
        }
        return view
    }()
```

philips77, 2019-10-23:
Sure, thank you for submitting the issue. I will look into it soon.

philips77, 2019-10-25:
Hi, could you try with the latest commit? I added some synchronisation around the contexts, I hope it will help.

codepgq, 2019-10-25:
I will test it tomorrow

philips77, 2019-10-25:
I'm planning to release 2.0 today (fingers crossed!). If you'll find any issues, they'll be in 2.0.1+.

codepgq, 2019-10-25:
OK！

philips77, 2019-10-28:
Any success here?

philips77, 2019-10-30:
The 2.0.0 has been released and it's available on CocoaPods and Carthage. As I said, if you'll find any more issues they'll be addresses in future versions.

codepgq, 2019-10-30:
I'm sorry to have replied to you now, we haven't tested the problem since your last fix. One phenomenon, when I have several model binding applicationkeys, some models may not be bound on, this is not always happening. But after 0.1s each time the data is spaced, it's work better, and I remember there was a message queue on your sdk

codepgq, 2019-10-31:
hi,I have this problem again now, after I quickly control many times of ctl instructions.

[image]

[image]

philips77, 2019-10-31:
It's one layer below. Looks like the same issue with concurrent modification, isn't it?

philips77, 2019-11-01:
I added thread synchronisation on Upper Transport Layer. I'll release it in 2.0.1 on Monday. If you could try the *develop* branch and see if it works for you?

codepgq, 2019-11-01:
I'll be testing it next Monday, and now I don't have the device?.

philips77, 2019-11-06:
Fix has been released in 2.1.0. I'm closing the issue now, feel free to reopen it or create a new one if you still hit this issue.

closed 2019-11-06 by philips77