nordicsemi/IOS-nRF-Mesh-Library #180: cocoapods for nrf mesh library

deverickasih, 2019-10-23:
Hi, i have tried to create a new xcode project and using the following in podfile :

pod 'OpenSSL-Universal', '1.0.2.13'
pod 'SwiftLint', '0.27.0'
pod 'nRFMeshProvision', '1.0.4'

but i always get the error of 'opensslhelper' being undefined.
[Use of unresolved identifier 'OpenSSLHelper']
has this been fixed?

Thanks

sunilloharshoreline1, 2019-10-23:
try using : pod 'OpenSSL' instead of pod 'OpenSSL-Universal'

philips77, 2019-10-23:
Btw, please use the version 2, not 1.0.4. So far it's not on CocoaPods, but I'll publish before the end of this week. The old version is not maintained anymore.

deverickasih, 2019-10-23:
@sunilloharshoreline1 i get the same unresolved identifier
@philips77 thx, be waiting for it

philips77, 2019-10-30:
The library 2.0.0 is now out on CocoaPods. Please test and confirm it's working for you.

deverickasih, 2019-11-01:
i'm using xcode 10.2.1 -> deployment target 10.0

inside podfile include : 
platform :ios, '10.0'
pod 'nRFMeshProvision', '2.0.0'

The library is usable with a couple of fixable errors as follows :
(1) Model.swift line [162...165] missing 'return' before 'modelId'
(2) LowerTransportLayer.swift line 186 & 188, add '?' so that it will look like the following :
           case .lowerTransportPdu(let pdu)?:
           case .acknowledgement(let ack)?:

Aside from that if i were to copy the AppDelegate from the example project into my new xcode project with the pod above, i may need to also import manually the 'Mesh Network' folder (from the example project) to be able to get a perfect running build.

Thanks very much for the library 2.0.0

philips77, 2019-11-01:
Yes, these are all Swift 5.1 features. The easiest solution for you is to update Xcode to 11.
Regarding the second part: you don't need `NetworkConnection` or any of the model delegates. They are application level settings. You should define your own models that your iPhone should support. For example, of your sending data to some vendor model, you should have the corresponding vendor client model.

philips77, 2019-11-01:
I'm applying the changes you mentioned to 2.0.1, so it will be compatible with Xcode 10.x. Did you have to modify any other files?

philips77, 2019-11-01:
The changes you requested will be fixed in 2.0.1 so it should be compatible with Xcode 10.x. Do you see any other issues?

philips77, 2019-11-06:
After all, I released 2.1.0, not 2.0.1. I hope it works for you even in Xcode 10.

closed 2019-11-06 by philips77