nordicsemi/IOS-nRF-Mesh-Library #183: crash when app launch

heywil, 2019-10-25:
Hi, 

I just downloaded the latest version and run the example app, the app crashed just after launch. I tied on both iOS 12.2 and 13.1.3, both have the same result.

The following message was shown in the log:
Fatal error: 'try!' expression unexpectedly raised an error: nRFMeshProvision.MeshNetworkError.addressNotAvailable: file /Users/willieip/Downloads/IOS-nRF-Mesh-Library-version2/nRFMeshProvision/Classes/MeshNetworkManager.swift, line 169
2019-10-25 17:36:13.877129+0800 nRFMeshProvision_Example[479:25283] Fatal error: 'try!' expression unexpectedly raised an error: nRFMeshProvision.MeshNetworkError.addressNotAvailable: file /Users/willieip/Downloads/IOS-nRF-Mesh-Library-version2/nRFMeshProvision/Classes/MeshNetworkManager.swift, line 169

philips77, 2019-10-25:
Yes, I forgot to push one commit yesterday. Please try now with the latest changes.

philips77, 2019-10-29:
This should be fixed now.

closed 2019-10-29 by philips77