nordicsemi/IOS-nRF-Mesh-Library #188: Can't edit network key

icarus31, 2019-10-30:
**Describe the bug**
Can't edit the network key because a new one is generated each time I open the alert view.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new network key
2. Tap the 'key' row
3. The key shown is a new one
4. See error

**Expected behavior**
Should show the previous key if the key already exists and create a new one if none. Just add key?.key to the presentKeyDialog like below?

`

    func presentKeyDialog() {
        let title = "New Key"
        let message = "The key must be a 32-character hexadecimal string."
        
        presentKeyDialog(title: title, message: message, key: key?.key) { key in
            self.newKey = key
            self.tableView.reloadRows(at: [.key], with: .fade)
        }
    }`

philips77, 2019-10-30:
Indeed, there's a bug. Also, seems like I can't edit the primary network key. I will fix it right away.

philips77, 2019-10-30:
Hi,
So I updated the code as you suggested. However, the current version of the library does not support Key Refresh procedure, so it does not allow to modify an existing key, even it it's unused. You need to remove it and create a new again. You may only edit the key when you are adding it and the key hasn't been added to the network yet.

closed 2019-10-30 by philips77

icarus31, 2019-10-31:
Doing the way you did (not including !isKeyUsed in the condition), does not solve the problem that I described. I want to be able to enter a new key (this is handled), but after I did 'Done' AND the key is not used, I want to be able to edit it (I did a typo, for example).

Maybe I did not understand properly what you described in the previous comment, but I think it won't hurt putting back !isKeyUsed.

Anyway, this is your code, so you do what you want, right? :-) You did a VERY good job!

philips77, 2019-10-31:
Yes, I understand, but right now there's no API that would allow to change the key in a Key. The `key` property is read-only from outside of the library. This looks like a feature for one of next versions. Ideally, changing the key should propagate the key using `ConfigXXXKeyUpdate` to all non-blacklisted nodes. Allowing only changing the key when it's not used can be done by removing it and creating a new one without any modifications in the library.

philips77, 2019-10-31:
Btw, out of curiosity. How easy it is to use the library now? Let's call it a User Poll :)

icarus31, 2019-11-03:
To be honest with you, the jump between version 1 and 2 was easier for me to understand all the parts and logging is much better too! :-) By the way, does removing the encryption is something that would be easy to remove temporally? Debugging encrypted data is not always easy to understand! ;-)

Now, because I am only a basic Swift developer, the big refactoring you did lately gave me trouble to fully understand what you did and the modification needed to migrate my code.

I created a new branch to try something based on that change and for some reason I was not able to process what the node sent me. I did not have time to work on it since you gave me a 'step-by-step' tutorial to handle it.

One thing I could suggest is if during the provisioning you have more than one nodes in the mesh, staying connected to the first node is not always a good thing. Getting composition data is a very 'heavy' process and the chance to fail is high and can become very long. I modified the original code to reconnect to the node I want to provision when going to configuration process. It is stable and much faster. I understand the way you did it, but I wanted to share with you this.

Beside that, I think it is a good library to play with.