nordicsemi/IOS-nRF-Mesh-Library #190: Get Composition Data Anomaly

deverickasih, 2019-11-01:
**Describing the bug**
I have two boards : 
(1) board with 2 elements
- first element > health & configuration
- second element > generic on/off 
(2) board with 5 elements
- first element > health & configuration
- 2...5 element > generic on/off

By default the example project provides the provisioner with 2 elements meaning that the next available address is 0x0003

**First Case [Failed]**
I add the first board with 2 elements with success (identify > provision > get composition data). This makes the next available address to be 0x0005

And then I add the second board with 5 elements with success (identify > provision) only. When getting composition data, it keeps on resending and ends with a timeout. [is the time too short for 5 elements?]

But when i redo the above process, and for the second board implicitly change the address (after identifying and before provisioning) to 0x0008, it works. But it does not work with addresses [0x0005, 0x0006, 0x0007]

**Second Case [Success]**
When i add (at 0x0003) the node with 5 elements first and then followed by (at 0x0008) the node with 2 elements, it works. There is no 'Get Composition Data' anomaly happening there. I also tried adding (at 0x000A) a third node with 5 elements and also with success (identity > provision > get composition data). No problemo.

**Third Case [Failed]**
When I add (at 0x0003) a node with 2 elements and then followed by (at 0x0005) another node with 2 elements. The second node fails at 'Get Composition Data' - Timeout.

**Hypothesis & Questions**
Is there a 5 address gap from the adding of the first node (at 0x0003) to the next node (at 0x0008)?
Will this anomaly also be happening at other addresses?

**Expected behavior**
(1) We shouldn't need to explicitly change the address but rather allow the assigned next available address to work?
(2) I also believe the expected behavior is that the 'Get Composition Data' Timeout shouldn't be happening?

**Platform details:**
 - Device: iPhone6
 - OS: iOS 12.1.4
 - Library Tag 2.0.0

philips77, 2019-11-01:
The address selection shouldn't matter. It should work with the same probability on the address 5 like 8. For you try multiple times? Could you post the logs of a failure here? Are you sure it's not related to how your proxy relies message? Did you try provisioning first the mode with 5 elements in address 5?

deverickasih, 2019-11-01:
Yes, I did try the first case 4 times, and always ends with timeout when getting composition data.

Provisioning first the node with 5 elements in address 5 works fine.

**Posting Logs Below For [First Case] Failed**
[Bearer] Central Manager state changed to .poweredOn
[Bearer] Connecting to nRF5x Mesh Light 4 Gang...
[Bearer] Connected to nRF5x Mesh Light 4 Gang
[Bearer] Discovering services...
[Bearer] Service found
[Bearer] Discovering characteristrics...
[Bearer] Data In characteristic found
[Bearer] Data Out characteristic found
[Bearer] Enabling notifications...
[Bearer] Data Out notifications enabled
[Bearer] GATT Bearer open and ready
[Provisioning] Requesting Provisioning Capabilities
[Provisioning] Sending Provisioning Invite (attention timer: 5 sec)
[Bearer] -> 0x030005
[Bearer] <- 0x03010500010001000000000000
[Provisioning] Device Capabilities: Number of elements: 5
Algorithms: FIPS P-256 Elliptic Curve
Public Key Type: None
Static OOB Type: Static OOB Information Available
Output OOB Size: 0
Output OOB Actions: None
Input OOB Size: 0
Input OOB Actions: None received
[Provisioning] Provisioning Capabilities received
[Provisioning] Provisioning started
[Provisioning] Sending Provisioning Start (algorithm: FIPS P-256 Elliptic Curve, public Key: No OOB Public Key, authentication Method: No OOB)
[Bearer] -> 0x03020000000000
[Provisioning] Sending Provisioner Public Key (0x5AC2AE7BF0FB080F5E1E55F7A132B20293CDF9331060C17B5E896C9A5390F1EC08C1BF8AB180897143FABEE9792AB32C6E6ECB6A1CFF9F15DE7A5172A5930D07)
[Bearer] -> 0x03035AC2AE7BF0FB080F5E1E55F7A132B20293CDF9331060C17B5E896C9A5390F1EC08C1BF8AB180897143FABEE9792AB32C6E6ECB6A1CFF9F15DE7A5172A5930D07
[Bearer] <- 0x0303DE7372C97F45E8317C2D442DC33394BF75885226387D7F2E32CFF9EECB3D9724FC7EFCEE7C90407BACAEE37225A2DA39CD4AD87D7B67891CD1179864551BF248
[Provisioning] Device Public Key (0xDE7372C97F45E8317C2D442DC33394BF75885226387D7F2E32CFF9EECB3D9724FC7EFCEE7C90407BACAEE37225A2DA39CD4AD87D7B67891CD1179864551BF248) received
[Provisioning] Sending Provisioner Confirmation (0x18D5D36632CAAE5FD45EE3D6EEF7C5EB)
[Bearer] -> 0x030518D5D36632CAAE5FD45EE3D6EEF7C5EB
[Bearer] <- 0x030540C2F69F56D7A8CD219FF9CC7092386C
[Provisioning] Device Confirmation (0x40C2F69F56D7A8CD219FF9CC7092386C) received
[Provisioning] Sending Provisioner Random (0x292950E4E080A30D927216C327B5BA19)
[Bearer] -> 0x0306292950E4E080A30D927216C327B5BA19
[Bearer] <- 0x0306008D36589A4A2F1CB8022C3103503F71
[Provisioning] Device Random (0x008D36589A4A2F1CB8022C3103503F71) received
[Provisioning] Sending Encrypted Provisioning Data (0xA22ECEA62F8DAE6978DD7B9C0925456D37EDF515D72125E4F64C882EC0C6B94DF9)
[Bearer] -> 0x0307A22ECEA62F8DAE6978DD7B9C0925456D37EDF515D72125E4F64C882EC0C6B94DF9
[Bearer] <- 0x0308
[Provisioning] Provisioning Complete received
[Provisioning] Provisioning complete
[Bearer] Cancelling connection...
[Bearer] Disconnected from nRF5x Mesh Light 4 Gang
[Provisioning] Provisioner is ready
[CoreBluetooth] XPC connection invalid
[CoreBluetooth] XPC connection invalid
[FoundationModel] Sending ConfigCompositionDataGet(page: 0) to: 0005
[Access] Sending Access PDU (opcode: 0x8008, parameters: 0x00)
[UpperTransport] Sending Upper Transport PDU (encrypted data: 0x1BBE5B, transMic: 0x1FAE1760) encrypted using key: nRF5x Mesh Light 4 Gang's Device Key
[LowerTransport] Sending Access Message (akf: 0, szmic: 0, data: 0x1BBE5B1FAE1760)
[Network] Sending Network PDU (ivi: 0, nid: 0x50, ctl: 0, ttl: 5, seq: 12, src: 0001, dst: 0005, transportPdu: 0xB5AE2EDB85158B8A, netMic: 0xB9272EBF) encrypted using Primary Network Key (index: 0)
[Bearer] -> 0x0050F38E9E08A061A185B5AE2EDB85158B8AB9272EBF
[Bearer] <- 0x0101005DEAD1282FE89DCC0000000094AF0BAF00E6A8BC
[Network] Secure Network Beacon (Network ID: 5DEAD1282FE89DCC, IV Index: 0, Key Refresh Flag: false, IV Update active: false) receieved (decrypted using key: Primary Network Key (index: 0))
[Proxy] New Proxy connected
[Proxy] Sending SetFilterType(filterType: Whitelist) from: 0001 to: 0000
[Network] Sending Control Message (opCode: 0x00, data: 0x00)
[Network] Sending Network PDU (ivi: 0, nid: 0x50, ctl: 1, ttl: 0, seq: 13, src: 0001, dst: 0000, transportPdu: 0xDBEF, netMic: 0x83177CEB70F0D6FE) encrypted using Primary Network Key (index: 0)
[Bearer] -> 0x025004605675C2FC9790DBEF83177CEB70F0D6FE
[Bearer] <- 0x0250D443600BBD6F84B7B096961E871BF4C099C384B2
[Network] Network PDU (ivi: 0, nid: 0x50, ctl: 1, ttl: 0, seq: 12, src: 0003, dst: 0000, transportPdu: 0xB096961E, netMic: 0x871BF4C099C384B2) received
[Network] Control Message (opCode: 0x03, data: 0x000000) receieved (decrypted using key: Primary Network Key (index: 0))
[... 2 lines cut ...]
[Network] Sending Control Message (opCode: 0x01, data: 0xFFFF0001)
[Network] Sending Network PDU (ivi: 0, nid: 0x50, ctl: 1, ttl: 0, seq: 14, src: 0001, dst: 0000, transportPdu: 0x21D5078922, netMic: 0xC6B963B40C2A4E33) encrypted using Primary Network Key (index: 0)
[Bearer] -> 0x02503C59A8A82D407DEF21D5078922C6B963B40C2A4E33
[Bearer] <- 0x0250C61DB746B8D2BE5A50913C3AFB904E3818D02C75
[Network] Network PDU (ivi: 0, nid: 0x50, ctl: 1, ttl: 0, seq: 13, src: 0003, dst: 0000, transportPdu: 0x50913C3A, netMic: 0xFB904E3818D02C75) received
[Network] Control Message (opCode: 0x03, data: 0x000002) receieved (decrypted using key: Primary Network Key (index: 0))
[Proxy] FilterStatus(filterType: Whitelist, listSize: 2) received from: 0003 to: 0000
[Access] Resending Access PDU (opcode: 0x8008, parameters: 0x00)
[... cut ...]
[Network] Sending Network PDU (ivi: 0, nid: 0x50, ctl: 0, ttl: 5, seq: 17, src: 0001, dst: 0005, transportPdu: 0x16E5E9167B09769D, netMic: 0x4C8755B0) encrypted using Primary Network Key (index: 0)
[Bearer] -> 0x00506D45B6D120D7E68B16E5E9167B09769D4C8755B0
[Access] Response to Access PDU (opcode: 0x8008, parameters: 0x00) not received (timeout)
[FoundationModel] ConfigCompositionDataGet(page: 0) sent from: 0001, to: 0005 timed out
[Access] Cancelling messages with op code: 32776, sent from: 0001 to: 0005

philips77, 2019-11-01:
Hmm... You don't get any messages from the node. To which proxy were you connected?

Let me describe you the logs:

1. I think you didn't set `logger` to the bearer you are using, as some logs seem to be missing.
2. You start provisioning node with address 0x0005. I hope both nodes were provisioned with the same Network Key, otherwise the first node wouldn't be able to rely messages to the soon-to-be-provisioned node 0x0005.
```
[Bearer] Central Manager state changed to .poweredOn
[Bearer] Connecting to nRF5x Mesh Light 4 Gang...
[Bearer] Connected to nRF5x Mesh Light 4 Gang
[Bearer] Discovering services...
[Bearer] Service found
[Bearer] Discovering characteristrics...
[Bearer] Data In characteristic found
[Bearer] Data Out characteristic found
[Bearer] Enabling notifications...
[Bearer] Data Out notifications enabled
[Bearer] GATT Bearer open and ready
[Provisioning] Requesting Provisioning Capabilities
[Provisioning] Sending Provisioning Invite (attention timer: 5 sec)
[Bearer] -> 0x030005
[Bearer] <- 0x03010500010001000000000000
[Provisioning] Device Capabilities: Number of elements: 5
Algorithms: FIPS P-256 Elliptic Curve
Public Key Type: None
Static OOB Type: Static OOB Information Available
Output OOB Size: 0
Output OOB Actions: None
Input OOB Size: 0
Input OOB Actions: None received
[Provisioning] Provisioning Capabilities received
[Provisioning] Provisioning started
[Provisioning] Sending Provisioning Start (algorithm: FIPS P-256 Elliptic Curve, public Key: No OOB Public Key, authentication Method: No OOB)
[Bearer] -> 0x03020000000000
[Provisioning] Sending Provisioner Public Key (0x5AC2AE7BF0FB080F5E1E55F7A132B20293CDF9331060C17B5E896C9A5390F1EC08C1BF8AB180897143FABEE9792AB32C6E6ECB6A1CFF9F15DE7A5172A5930D07)
[Bearer] -> 0x03035AC2AE7BF0FB080F5E1E55F7A132B20293CDF9331060C17B5E896C9A5390F1EC08C1BF8AB180897143FABEE9792AB32C6E6ECB6A1CFF9F15DE7A5172A5930D07
[Bearer] <- 0x0303DE7372C97F45E8317C2D442DC33394BF75885226387D7F2E32CFF9EECB3D9724FC7EFCEE7C90407BACAEE37225A2DA39CD4AD87D7B67891CD1179864551BF248
[Provisioning] Device Public Key (0xDE7372C97F45E8317C2D442DC33394BF75885226387D7F2E32CFF9EECB3D9724FC7EFCEE7C90407BACAEE37225A2DA39CD4AD87D7B67891CD1179864551BF248) received
[Provisioning] Sending Provisioner Confirmation (0x18D5D36632CAAE5FD45EE3D6EEF7C5EB)
[Bearer] -> 0x030518D5D36632CAAE5FD45EE3D6EEF7C5EB
[Bearer] <- 0x030540C2F69F56D7A8CD219FF9CC7092386C
[Provisioning] Device Confirmation (0x40C2F69F56D7A8CD219FF9CC7092386C) received
[Provisioning] Sending Provisioner Random (0x292950E4E080A30D927216C327B5BA19)
[Bearer] -> 0x0306292950E4E080A30D927216C327B5BA19
[Bearer] <- 0x0306008D36589A4A2F1CB8022C3103503F71
[Provisioning] Device Random (0x008D36589A4A2F1CB8022C3103503F71) received
[Provisioning] Sending Encrypted Provisioning Data (0xA22ECEA62F8DAE6978DD7B9C0925456D37EDF515D72125E4F64C882EC0C6B94DF9)
[Bearer] -> 0x0307A22ECEA62F8DAE6978DD7B9C0925456D37EDF515D72125E4F64C882EC0C6B94DF9
[Bearer] <- 0x0308
[Provisioning] Provisioning Complete received
[Provisioning] Provisioning complete
[Bearer] Cancelling connection...
[Bearer] Disconnected from nRF5x Mesh Light 4 Gang
```
3. The node 0x0005 was provisioned successfully.
4. Here we have some logs that in my opinion should not be there. But this is not an issue.
```
[Provisioning] Provisioner is ready
[CoreBluetooth] XPC connection invalid
[CoreBluetooth] XPC connection invalid
```
5. I think you are now starting connecting to the node 0x0003 (not the new one). There are no logs, but perhaps you didn't set the `logger` to the `NetworkConnection` or `GattBearer` object, or you're using your own proxy implementation.
6. You're starting sending the message to 0x0005.
```
[FoundationModel] Sending ConfigCompositionDataGet(page: 0) to: 0005
[Access] Sending Access PDU (opcode: 0x8008, parameters: 0x00)
[UpperTransport] Sending Upper Transport PDU (encrypted data: 0x1BBE5B, transMic: 0x1FAE1760) encrypted using key: nRF5x Mesh Light 4 Gang's Device Key
[LowerTransport] Sending Access Message (akf: 0, szmic: 0, data: 0x1BBE5B1FAE1760)
[Network] Sending Network PDU (ivi: 0, nid: 0x50, ctl: 0, ttl: 5, seq: 12, src: 0001, dst: 0005, transportPdu: 0xB5AE2EDB85158B8A, netMic: 0xB9272EBF) encrypted using Primary Network Key (index: 0)
[Bearer] -> 0x0050F38E9E08A061A185B5AE2EDB85158B8AB9272EBF
[... 38 lines cut ...]
[UpperTransport] Sending Upper Transport PDU (encrypted data: 0xC57309, transMic: 0xE02BAF33) encrypted using key: nRF5x Mesh Light 4 Gang's Device Key
[LowerTransport] Sending Access Message (akf: 0, szmic: 0, data: 0xC57309E02BAF33)
[Network] Sending Network PDU (ivi: 0, nid: 0x50, ctl: 0, ttl: 5, seq: 16, src: 0001, dst: 0005, transportPdu: 0x0B419E94F7B7AC90, netMic: 0xDD5D2644) encrypted using Primary Network Key (index: 0)
[Bearer] -> 0x0050FB473644135B04320B419E94F7B7AC90DD5D2644
```
12. again and again...
```
[Access] Resending Access PDU (opcode: 0x8008, parameters: 0x00)
[UpperTransport] Sending Upper Transport PDU (encrypted data: 0xEB0FB1, transMic: 0x391B90E5) encrypted using key: nRF5x Mesh Light 4 Gang's Device Key
[LowerTransport] Sending Access Message (akf: 0, szmic: 0, data: 0xEB0FB1391B90E5)
[Network] Sending Network PDU (ivi: 0, nid: 0x50, ctl: 0, ttl: 5, seq: 17, src: 0001, dst: 0005, transportPdu: 0x16E5E9167B09769D, netMic: 0x4C8755B0) encrypted using Primary Network Key (index: 0)
[Bearer] -> 0x00506D45B6D120D7E68B16E5E9167B09769D4C8755B0
```
13. And the timeout happened.
```
[... cut ...]
1. The node 0x0003 has been provisioned using a different network key, not Primary Network Key (index: 0). The proxy cannot rely this message as it cannot decode it on the Network Layer.
2. The sequence number that the 0x0005 expects is higher than what you are actually sending. Messages with sequence number <= last seq are dismissed, so you are not getting anything here. As this is newly provisioned node, you may have some issues with not clearing flash? Try erasing the memory and flashing the fw again.
3. The node 0x0003 has Proxy Feature disabled. It will reply to you, but will not relay messages to ADV bearer.

deverickasih, 2019-11-04:
I've also build and run the example project ensuring that (1) there is only one network key and (2) enabling proxy through the 'Configuration Server' -> 'GATT Proxy', and it still fails even with the example project. It could be a sequence number problem but how do i check that?

The only way i could add the second node is if i turn off the power of the already provisioned first node before adding the second node, because if i allow the first node to be turned on, the provisioner auto-joins the provisioned first node and the anomaly happens. But i hope that i don't have to turn off the power to any provisioned nodes before adding an unprovisioned node.

**A probable solution for my problem**
Is there a way to disable the auto-join feature in version 2?

philips77, 2019-11-04:
1. Ok
2. A newly provisioned node should have sequence numbers restarted as well.
3. Ok.
Are those kids above from your app, or the sample app? Could you post from the latter, where the logger is also set for the proxy bearer?
Do you have any way of getting logs from the node 3 and 5? Do you see 5 getting and message from 3 over adv bearer?

philips77, 2019-11-04:
Btw, I managed to replicate the issue on my side. I'll investigate that's going on there and let you know.

philips77, 2019-11-04:
I don't know what happened then, but now I can't reproduce this issue. Perhaps I did disable GATT Proxy feature on the connected node to test it, as this is now the only way I can reproduce it, but you're saying you have it enabled.

deverickasih, 2019-11-05:
Hi again, the following are the logs from the example project that I had just downloaded from the master. I did enable the first node to be GATT Proxy but the results are the same for me.

```
[Bearer] Central Manager state changed to .poweredOn
[Bearer] Connecting to nRF5x Mesh Light 1 Gang...
[Bearer] Connected to nRF5x Mesh Light 1 Gang
[Bearer] Discovering services...
[Bearer] Service found
[Bearer] Discovering characteristrics...
[Bearer] Data In characteristic found
[Bearer] Data Out characteristic found
[Bearer] Enabling notifications...
[Bearer] Data Out notifications enabled
[Bearer] GATT Bearer open and ready
[Provisioning] Requesting Provisioning Capabilities
[Provisioning] Sending Provisioning Invite (attention timer: 5 sec)
[Bearer] -> 0x030005
[Bearer] <- 0x03010500010001000000000000
[Provisioning] Device Capabilities: Number of elements: 5
Algorithms: FIPS P-256 Elliptic Curve
Public Key Type: None
Static OOB Type: Static OOB Information Available
Output OOB Size: 0
Output OOB Actions: None
Input OOB Size: 0
Input OOB Actions: None received
[Provisioning] Provisioning Capabilities received
[Provisioning] Provisioning started
[Provisioning] Sending Provisioning Start (algorithm: FIPS P-256 Elliptic Curve, public Key: No OOB Public Key, authentication Method: No OOB)
[Bearer] -> 0x03020000000000
[Provisioning] Sending Provisioner Public Key (0x40563010A251B9B8B7888A63DEC15CE92942F26A33A0957A5E680624FB38EED61F7E0FDF2520CC2A7B42176DB60329C5D6828E9FC6ED06A5710EEFD02AC0C411)
[Bearer] -> 0x030340563010A251B9B8B7888A63DEC15CE92942F26A33A0957A5E680624FB38EED61F7E0FDF2520CC2A7B42176DB60329C5D6828E9FC6ED06A5710EEFD02AC0C411
[Bearer] <- 0x030371134EAD339F2062C242EC5E04CFCF0FBBAE4914DF30890C6366D680AB313EA84200807B4A4A359B19DAB6E67E3A5BF2B61246D98C821E21CBA89E4E0F6C0E7F
[Provisioning] Device Public Key (0x71134EAD339F2062C242EC5E04CFCF0FBBAE4914DF30890C6366D680AB313EA84200807B4A4A359B19DAB6E67E3A5BF2B61246D98C821E21CBA89E4E0F6C0E7F) received
[Provisioning] Sending Provisioner Confirmation (0x0D9E3C7DF1746E31BD9C15437C5839C0)
[Bearer] -> 0x03050D9E3C7DF1746E31BD9C15437C5839C0
[Bearer] <- 0x03054B57A3DF7A75F0F6D704B88A8167A8E6
[Provisioning] Device Confirmation (0x4B57A3DF7A75F0F6D704B88A8167A8E6) received
[Provisioning] Sending Provisioner Random (0x85209C43DBC3C50CB8593A58DB809A99)
[Bearer] -> 0x030685209C43DBC3C50CB8593A58DB809A99
[Bearer] <- 0x0306FA2E6914BC8772CF593EBFFC0B6D1E78
[Provisioning] Device Random (0xFA2E6914BC8772CF593EBFFC0B6D1E78) received
[Provisioning] Sending Encrypted Provisioning Data (0x3F3204914E34EDB7FDD035C0194A6E4F38B805413A8A47E70294CF2EDEE5BDE5FF)
[Bearer] -> 0x03073F3204914E34EDB7FDD035C0194A6E4F38B805413A8A47E70294CF2EDEE5BDE5FF
[Bearer] <- 0x0308
[Provisioning] Provisioning Complete received
[Provisioning] Provisioning complete
[Bearer] Cancelling connection...
[Bearer] Disconnected from nRF5x Mesh Light 1 Gang
[Provisioning] Provisioner is ready
[Bearer] Central Manager state changed to .poweredOn
[Bearer] Connecting to nRF5x Mesh Light 1 Gang...
[Bearer] Connected to nRF5x Mesh Light 1 Gang
[Bearer] Discovering services...
[CoreBluetooth] XPC connection invalid
[CoreBluetooth] XPC connection invalid
[Bearer] Service found
[Bearer] Discovering characteristrics...
[FoundationModel] Sending ConfigCompositionDataGet(page: 0) to: 0003
[Access] Sending Access PDU (opcode: 0x8008, parameters: 0x00)
[UpperTransport] Sending Upper Transport PDU (encrypted data: 0x856B72, transMic: 0x0AECDEA8) encrypted using key: nRF5x Mesh Light 1 Gang's Device Key
[LowerTransport] Sending Access Message (akf: 0, szmic: 0, data: 0x856B720AECDEA8)
[Network] Sending Network PDU (ivi: 0, nid: 0x25, ctl: 0, ttl: 5, seq: 0, src: 0001, dst: 0003, transportPdu: 0x7CEED5047D490250, netMic: 0x19929552) encrypted using Primary Network Key (index: 0)
[LowerTransport] The bearer is closed.
[Bearer] Data In characteristic found
[Bearer] Data Out characteristic found
[Bearer] Enabling notifications...
[Bearer] Data Out notifications enabled
[Bearer] GATT Bearer open and ready
[Bearer] <- 0x010100AD1EF1B37D5135A300000000E50A982B2485982F
[... 167 lines cut ...]
[UpperTransport] Sending Upper Transport PDU (encrypted data: 0xFF9685, transMic: 0x255C8FA6) encrypted using key: nRF5x Mesh Light 4 Gang's Device Key
[LowerTransport] Sending Access Message (akf: 0, szmic: 0, data: 0xFF9685255C8FA6)
[Network] Sending Network PDU (ivi: 0, nid: 0x25, ctl: 0, ttl: 5, seq: 14, src: 0001, dst: 0005, transportPdu: 0x3B6E76BCC7555864, netMic: 0x34BD0F2E) encrypted using Primary Network Key (index: 0)
[Bearer] -> 0x002598395B3A443D10C93B6E76BCC755586434BD0F2E
[Access] Resending Access PDU (opcode: 0x8008, parameters: 0x00)
[UpperTransport] Sending Upper Transport PDU (encrypted data: 0xC6003A, transMic: 0xF79C834D) encrypted using key: nRF5x Mesh Light 4 Gang's Device Key
[LowerTransport] Sending Access Message (akf: 0, szmic: 0, data: 0xC6003AF79C834D)
[Network] Sending Network PDU (ivi: 0, nid: 0x25, ctl: 0, ttl: 5, seq: 15, src: 0001, dst: 0005, transportPdu: 0xDA95968D92783004, netMic: 0xA44DF807) encrypted using Primary Network Key (index: 0)
[Bearer] -> 0x0025829F1CEBA7CC00D5DA95968D92783004A44DF807
[Access] Resending Access PDU (opcode: 0x8008, parameters: 0x00)
[UpperTransport] Sending Upper Transport PDU (encrypted data: 0xE5F91E, transMic: 0x1EBFBA53) encrypted using key: nRF5x Mesh Light 4 Gang's Device Key
[LowerTransport] Sending Access Message (akf: 0, szmic: 0, data: 0xE5F91E1EBFBA53)
[... cut ...]
[Access] Response to Access PDU (opcode: 0x8008, parameters: 0x00) not received (timeout)
[FoundationModel] ConfigCompositionDataGet(page: 0) sent from: 0001, to: 0005 timed out
[Access] Cancelling messages with op code: 32776, sent from: 0001 to: 0005
```

I don't think GATT proxy enabled or disabled makes any difference because if I were to add the first node with 5 elements first and then disable the GATT proxy of the first node, I could add the second node with 2 elements without any problems...

philips77, 2019-11-05:
Hmm... the bearer is sending messages, but doesn't get anything back. It's hard to say why exactly. I mentioned above the possible reasons. As I wrote before, I did manage to get the same results once, but never after that. As this works for you in some configurations, and doesn't in other, that indicates for me the issue with the device. Gathering logs or sniffer traces from them may help finding the root cause. I'm sorry, but I can't help you more :(

deverickasih, 2019-11-06:
Thanks for much assistance, i'll check it with the device. I'll see if i could find anything..

philips77, 2019-11-19:
Hi,
Did you manage to solve the issue?

deverickasih, 2019-11-20:
I don't think it could be called a fix but currently i provisioned each node 1-on-1 without using proxies.

Talking about proxies, maybe this could somehow be connected with my **Get Composition Data Anomaly** problem. Below is a case.

By default a provisioned node has proxy turned off right? So this means that i shouldn't be able to turn on the light of node B through connection with node A if i do not turn on their proxy feature. What could be the reason if i could do that without proxy feature turned on on any of the nodes? I ask this because in my own project, according to the logs, (1) i am connected to node A, (2) without any logs stating that i disconnected from node A and reconnected to node B, i can control the lights of node B.

philips77, 2019-11-20:
This depends on the implementation. In our Mesh SDK the proxy feature is enabled by default. In the sample app you should be able to go to the Node's Configuration Server model and do Pull to Refresh to obtain current state of the properties. When the Proxy Feature is disabled, you should still be able to control Node A, but it would not not relay messages to the ADV Bearer, nor would relay any messages sent over ADV Bearer to the phone.

igathedub, 2019-12-01:
We have issues that look similar:

Provisioning and configuring successfully several nodes.
Then, after 3-4 nodes, the provisioning of the next node is successful, but getting composition, resetting and adding app key is not possible.
BUT: if you manage to switch to the proxy of the new node, configuration/reset and so on works.
Logs show that acknowledgments are getting discarded because their sequence number is far behind of required.

philips77, 2019-12-02:
Could you post some logs and perhaps steps to reproduce?

philips77, 2020-04-23:
I'm closing the issue due to inactivity. Please create a new one, describe the problem and provide logs if you still encounter the issue.

closed 2020-04-23 by philips77