nordicsemi/IOS-nRF-Mesh-Library #191: pod 'nRFMeshProvision', '~> 2.0' gives error for OpenSSL lib

sunilloharshoreline1, 2019-11-01:
Using latest pod for Version 2  (**pod 'nRFMeshProvision', '~> 2.0'**) gives compile error 

'myProject/Pods/OpenSSL-Universal/ios/lib/libcrypto.a(cmac.o)' does not contain bitcode. You must rebuild it with bitcode enabled (Xcode setting ENABLE_BITCODE), obtain an updated library from the vendor, or disable bitcode for this target. for architecture arm64

The version used for OpenSSL is 1.0.2.19, i think using 1.0.2.17 might solve the issue.

philips77, 2019-11-01:
I did reproduce this issue. I will go with your suggestion to use the older OpenSSL version.

philips77, 2019-11-01:
I fixed it by specifying the *OpenSSL-Universal* version in the pod file:

```ruby
target '<MyProject>' do
    pod 'nRFMeshProvision'
    pod 'OpenSSL-Universal', '= 1.0.2.17'
end
```
Can you live with that? I had problems with building the app for AppStore using this version and updating to 1.0.2.19 fixed the issue (also some byte code related things), so I'm not sure about reverting. I can document it in the readme.

philips77, 2019-11-01:
Also, 1.0.2.18 also seems to work.

sunilloharshoreline1, 2019-11-01:
I'm updating the XCode now, will check and post my observations

sunilloharshoreline1, 2019-11-04:
It does work now with 1.0.2.17 and  1.0.2.18. Thanks.
One more suggestion, can you group POD library classes into different folders as you did earlier?
Currently, all the classes come under 'nrfMeshProvision'.

philips77, 2019-11-04:
Have a look at: https://github.com/CocoaPods/CocoaPods/issues/2145
Great that it works, I'm closing the issue.

closed 2019-11-04 by philips77