nordicsemi/IOS-nRF-Mesh-Library #192: didReceiveMessage for GenericOnOff MeshMessage returns UnknownMessage

sunilloharshoreline1, 2019-11-04:
If the Read button pressed from GenericOnOff Server, the " didReceiveMessage sentFrom to " method returns MeshMessage as UnknownMessage , here is the messge object :

UnknownMessage(opCode: 0x008204, parameters: 00000000)
  - opCode : 33284
  ▿ parameters : Optional<Data>
    ▿ some : 4 bytes
      - count : 4
      ▿ pointer : 0x000000016ba38d40
        - pointerValue : 6100847936
      ▿ bytes : 4 elements
        - 0 : 0
        - 1 : 0
        - 2 : 0
        - 3 : 0

philips77, 2019-11-04:
Hello,
Did you set `manager.localElements` after creating the `MeshNetworkManager`?
Have a look at:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/dae83c9197a2739c4c1842a4a3d9ddd796d76b50/Example/nRFMeshProvision/AppDelegate.swift#L99-L112
where the sample app defines 2 elements (it can be 1, there are 2 just to show how to create more than 1 if needed) and defines the local models. One of the models is assigned this delegate:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/dae83c9197a2739c4c1842a4a3d9ddd796d76b50/Example/nRFMeshProvision/Mesh%20Network/GenericOnOffClientDelegate.swift#L34-L43
where if declares that it supports messages with Op Code 0x8204 that should be instantiated as `GenericOnOffStatus` message.

philips77, 2019-11-04:
In other words, if you are sending `GenericOnOffSet`, `...Get` or `...SetUnacknowledged`, you must have the Generic OnOff Client model on the phone. This model has to define messages it supports and declare to what types should they be instantiated. It should also reply to acknowledged messages, if such are supported by this model.

philips77, 2019-11-04:
Hi, did it help you? Can we close the issue?

sunilloharshoreline1, 2019-11-05:
I'm confused about the Client Model on the phone. 
I m using nordic's dev board which has all Generic Models. I m sending On/Off from "Version 2 Example App" to the d, the commands i.e On/Off it receives are perfectly fine, also, once I press "read" from Generic On Off server I can see the log on the Firmware too that it sends back the status to the mobile app. Also, the mobile app receives it in delegate method didReceiveMessage but the message type is UnknownMessage.
I m using the sample app provided here. And it has all the necessary things you mentioned above in AppDelegate :

```
func meshNetworkDidChange() {
        connection?.close()
        
        let meshNetwork = meshNetworkManager.meshNetwork!
        
        // Set up local Elements on the phone.
        let element0 = Element(name: "Primary Element", location: .first, models: [
            Model(sigModelId: 0x1000, delegate: GenericOnOffServerDelegate()),
            Model(sigModelId: 0x1002, delegate: GenericLevelServerDelegate()),
            Model(sigModelId: 0x1001, delegate: GenericOnOffClientDelegate()),
            Model(sigModelId: 0x1003, delegate: GenericLevelClientDelegate())
        ])
        let element1 = Element(name: "Secondary Element", location: .second, models: [
            Model(sigModelId: 0x1000, delegate: GenericOnOffServerDelegate()),
            Model(sigModelId: 0x1002, delegate: GenericLevelServerDelegate()),
            Model(sigModelId: 0x1001, delegate: GenericOnOffClientDelegate()),
            Model(sigModelId: 0x1003, delegate: GenericLevelClientDelegate())
        ])
        meshNetworkManager.localElements = [element0, element1]
        
        connection = NetworkConnection(to: meshNetwork)
        connection!.dataDelegate = meshNetworkManager
        connection!.logger = self
        meshNetworkManager.transmitter = connection
        connection!.open()
    }

```

philips77, 2019-11-05:
Ok, I know where is the issue. The `GenericOnOffStatus` message you receive has 4 octets, where the Mesh Model 1.0.1 defines it may have 1 or 3 octets.
[image]
Parsing the message fails, so it can't be instantiated as `GenericOnOffStatus` type, therefore it's returned as `UnknownMessage`.

philips77, 2019-11-05:
And here's where the parsing fails:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/dae83c9197a2739c4c1842a4a3d9ddd796d76b50/nRFMeshProvision/Classes/Mesh%20Messages/Generic%20Messages/GenericOnOffStatus.swift#L74-L86

sunilloharshoreline1, 2019-11-05:
Yes. That's right.
The mobile app receives 4 bytes of Data and additional data was Transaction Number from device.
Thanks for your help.

closed 2019-11-05 by sunilloharshoreline1