nordicsemi/IOS-nRF-Mesh-Library #195: Question Concerning Node.appkeys.count returning 0

deverickasih, 2019-11-06:
Hi, i tried binding an appkey to a node :
```
try MeshNetworkManager.instance.send(ConfigAppKeyAdd(applicationKey: selectedAppKey), to: node)
```
it return successfully : 
```
MessageHandle(opCode: 0, source: 1, destination: 2, manager: Optional(nRFMeshProvision.MeshNetworkManager))
```
Then i tried to get the node and also placed a breakpoint after the following line of code :
```
let nodeToSend = MeshNetworkManager.instance.meshNetwork?.node(withAddress: node.unicastAddress)
```
When i checked the variable 'nodeToSend', the 'nodeToSend.appKeys' has 0 appkeys
Then when i try to get the list of applicationkeys with the following code :
```
let keysForNode : [ApplicationKey]! = selectedModel.parentElement?.parentNode?.applicationKeysAvailableFor(selectedModel)
```
keysForNode has no application keys..

I debugged this also with the project from master with code :
```
start("Adding Application Key...") {
        return try MeshNetworkManager.instance.send(ConfigAppKeyAdd(applicationKey: selectedAppKey), to: self.node)
}
```
when i checked self.node.appkeys.count using breakpoint, i get 0 after binding an appkey to it..

**Question**
Shouldn't there be at least one appkey since i have previously binded one?

What i planned on doing is after binding the appkey to node, i also want to immediately bind appkey to models inside the node with the following lines of code :
```
let message = ConfigModelAppBind(applicationKey: keysForNode[0], to: selectedModel)
try MeshNetworkManager.instance.send(message, to: selectedModel)
```
but keysForNode.count == 0 :(

philips77, 2019-11-06:
All those methods you call are asynchronous. Adding an App Key to a Node requires active proxy connection to the mesh network, the `transmitter` and `bearer.delegate` set correctly, etc. You will always get the `MessageHandle` immediately, as this can be used to cancel the pending request. But this does not mean that the Node has already received the key. You need to wait for the `ConfigAppKeyStatus` message in manager's delegate. When you receive it, and the status is success, the Node has received the key and the local database has been updated, so since then requesting app keys for this node will return it.

philips77, 2019-11-06:
In the sample app you have to put a breakpoint in the `meshNetworkManager(_:didReceiveMessage:sentFrom:to)` where the `ConfigAppKeyStatus` is received.

deverickasih, 2019-11-06:
Thanks for this insight!

closed 2019-11-06 by deverickasih