nordicsemi/IOS-nRF-Mesh-Library #198: Clarifications about StaticVendorMessage send

fiveagle, 2019-11-07:
**Describe the bug**
I'm following the example from PR [194](https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/pull/194) but I cannot underdstand how should I send a message to an already existing model on my devices.

Below there is my implementation:

- We created this custom local model:

```
let element0 = Element(name: "Primary Element", location: .first, models: [
                        Model(modelId: 0x0002 , companyId: 0x0760, delegate: GreenPropertyKeyModelDelegate()),
                    ])

```
(The server model is modelId: 0x0001 so we set modelId 0x0002)

- We created this delegate:

```
class GreenPropertyKeyModelDelegate: ModelDelegate {
    var messageTypes: [UInt32 : MeshMessage.Type]
    
    var isSubscriptionSupported: Bool = false
    
    init() {
        messageTypes = [GreenPropertySetKeyMessage.opCode : GreenPropertySetKeyMessage.self]
    }
    
    func model(_ model: Model, didReceiveAcknowledgedMessage request: AcknowledgedMeshMessage, from source: Address, sentTo destination: MeshAddress) -> MeshMessage {
        
        print("\(self) didReceiveAcknowledgedMessage: \(request)")
        fatalError("Not possible")
    }
    
    func model(_ model: Model, didReceiveUnacknowledgedMessage message: MeshMessage, from source: Address, sentTo destination: MeshAddress) {
        print("\(self) didReceiveUnacknowledgedMessage: \(message)")
    }
    
    func model(_ model: Model, didReceiveResponse response: MeshMessage, toAcknowledgedMessage request: AcknowledgedMeshMessage, from source: Address) {
        print("\(self) didReceiveResponse: \(response)")
    }
  
}
```

- This is the message:

```
public struct GreenPropertySetKeyMessage: StaticVendorMessage {
    
   
    enum WritingPosition: UInt8 {
        case first = 0x1C
        case second = 0x1D
        
        func asPropertySetPosition() -> GreenPropertySetButtonMessage.WritingPosition {
            switch self {
            case .first:
                return .first
            case .second:
                return .second
            }
        }
    }
    
    public static var opCode: UInt32 = 0xC2
    
    public var parameters: Data?
    
    public init?(parameters: Data) {
        self.parameters = parameters
    }
    
    init?(enoceanKey: String, position: WritingPosition) {
   
       var payload: [UInt8] = [position.rawValue, 0x00]
        
        guard let oobIndex = enoceanKey.range(of: "OOB")?.upperBound else {
            return nil
        }
        let endIndex = String.Index(utf16Offset: enoceanKey.count, in: enoceanKey)
        
        let key = enoceanKey[oobIndex ..< endIndex]
        let keyString = String(key)
        
        let keyArray = keyString.components(separatedBy: "")
        
        let groupedKey = keyArray.chunked(into: 2).compactMap { String($0.joined()) }
        
        let keyByteArray = groupedKey.compactMap {
            UInt8($0)
        }
        
        payload.append(contentsOf: keyByteArray)
        
        self.init(parameters: Data(fromArray: payload))
    }
}

```

We used this process to publish the local Element model (is it correct???)

 ```
let periodResolution: StepResolution = .hundredsOfMilliseconds
                    let publish = Publish(to: MeshAddress(address), using: (MeshNetworkManager.instance.meshNetwork?.applicationKeys.first)!,
                                                     usingFriendshipMaterial: false, ttl: 255,
                                                     periodSteps: 62, periodResolution: periodResolution,
                                                     retransmit: Publish.Retransmit(publishRetransmitCount: 0,
                                                                                    intervalSteps: 0))

 let confmessage: ConfigMessage =
                    ConfigModelPublicationSet(publish, to: propertyModel!) ??
                        ConfigModelPublicationVirtualAddressSet(publish, to: propertyModel!)!
                try MeshNetworkManager.instance.send(confmessage, to: propertyModel!)

```
`(Property model  Model(modelId: 0x0001 , companyId: 0x0760, delegate: GreenPropertyKeyModelDelegate()) )`

Cases	:

1) if we publish the model using the address = Address(65535) (ALL NODES)
and after we send the message,we get some sort of response

2) If we try to publish the on a specific node address
and after we send the message we don’t get a response.

3) also if we use the vendor model from the composition data process (that we expected is mapped on our custom model)
We don’t get response (in android they use the vendor model from composition data process! )

What we missing????

ThankYou

philips77, 2019-11-07:
Hello,
First of all, when you have server model with ID 0x0001, I would suggest using 0x0000 for client. Otherwise some methods may not work, for example this:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/68dc144aabdf99da472b809115d94358b843343d/nRFMeshProvision/Classes/Mesh%20API/Models.swift#L44-L47
Secondly, you need to have 24-bit op code for vendor messages. This:
```swift
public static var opCode: UInt32 = 0xC2
```
should be changed to:
```swift
public static var opCode: UInt32 = 0xC26007
```

philips77, 2019-11-07:
If you set the publication to any address that is not a Unicast Address of any of phone's local elements, you need to make sure this address is added to Proxy Filter. Otherwise, the connected proxy will not retransmit this message to you over GATT. For example, a good option would be to set the publication to a group address, and then either subscribe your local model to this groups address, or just add it to `ProxyFilter` (remember, that even if you subscribe your local model to this group address, you need to add it to proxy filter manually. It will be added automatically upon next connection, but from subscribing to reconnection it won't be there, so you'll not get messages targeting this address).
If you want to subscribe your local model to this group address, you need to 
1. set `isSubscriptionSupported` to `true` in the `ModelDelegate`,
2. send `ConfigModelSubscriptionAdd` message targeting your own unicast address with the local model given in the init.

philips77, 2019-11-07:
Regarding the model id (0x0002 -> 0x0000). The convention is that an even number is the server and odd is a corresponding client. I don't think it's a requirement, but some features in the lib may depend on this. But if you're not using them, this is not a requirement. I actually didn't try breaking this rule, so I'm curious if you find any issues.

fiveagle, 2019-11-07:
Ok, but I should use a model already present in my device, instead of create a custom model and publishing it to every node. My above implementation is the only way I manage to get an answer from a send message.

When I try to send a message to the Vimar model of the mesh device I don't get anything back.

My model from Vimar is: 

```
(lldb) po node.elements[0].models.last
▿ Optional<Model>
  ▿ some : <Model: 0x282c21950>

(lldb) po node.elements[0].models.last?.name
▿ Optional<String>
  - some : "Vendor Model"

(lldb) po node.elements[0].models.last?.modelIdentifier
▿ Optional<UInt16>
  - some : 1

(lldb) po node.elements[0].models.last?.companyIdentifier
▿ Optional<UInt16>
  - some : 1888

(lldb) po node.elements[0].models.last?.delegate
nil
```

And I set my model delegates this way:

```
let element0 = Element(name: "Primary Element", 
                                        location: .first, 
                                        models: [
                                               Model(modelId: 0x0001 , 
                                                          companyId: 0x0760, 
                                                          delegate: GreenPropertyKeyModelDelegate())])

meshNetworkManager.localElements = [element0]
```

philips77, 2019-11-07:
Your device has a vendor model 0x0001, and you should have 0x0000 on the phone locally. This actually really doesn't matter, but as the two models have different behavior, they should use different ids.

philips77, 2019-11-07:
It's like with Generic models. The Generic OnOff Client (0x1001) sends messages to Generic OnOff Server (0x1000). They do different things, so have different IDs. If some other node in the network do Composition Data Get on your phone (iOS lib supports Configuration Server, so it will reply with the Composition Data Status), it would find out that you are a server, not a client in this pair, or would have no idea what features the phone supports if both remote and local model would have the same IDs by design.

fiveagle, 2019-11-07:
But is the publish required?
I mean... without a publish looks like that the mesh device model doesn't have a delegate set. How can I receive any callback from it when I send?

philips77, 2019-11-07:
You have to set the delegate only for local models, that is those that your phone supports. For example, if your app would allow users to switch lights, it should have 1 or mode Generic OnOff Client models (there can be only one model with given id on an elements, so if the app needs to have N buttons to control N light groups, N local elements are required).
The remote device, the one that you have provisioned, already has some elements and models, specified by your company specification. Let's say you develop smart bulbs, so each bulb has a single element with Generic OnOff Server. The implementation of this model is built-in in the bulb, so you can't do anything about it. You just have to request Composition Data from this bulb to get information about its features. In this scenario, you'd find out it has Configuration Server, Health Server (those 2 are required on all nodes) and Generic OnOff Server, so it will understand Generic OnOff Get/Set/Set Unacknowledged messages and it will return Generic OnOff Status (or publish it, if publication is set to this remote model).

That you have to do is:
1. In your app define let's say one element with Generic OnOff Client model and set the `ModelDelegate` to it (you have to set delegates to local models). This delegate should declare Generic OnOff Status message in `messageTypes`, as this is the message your local model will be receiving.
2. Provision the bulb and do Composition Data Get (library will store bulb's elements and models in the db).
3. Add App Key to the remote node and bind it to the Generic OnOff Server model on it.
4. You may now start sending Generic OnOff XXX to the remote node. The response will be instantiated as Generic OnOff Status based on the `messageTypes` of your local model.
5. You may also set publication on the remote Generic OnOff Server model to publish to some group address (or your address, or all nodes, or anything). Whenever the state of this bulb changes it will send Generic OnOff Status message to this address.

philips77, 2019-11-07:
Also, local models don't support publication on their own. If you want to support this, you need to call `manager.publish(...)` whenever the model's state changes, or periodically. In the sample app none of the models do this.
You don't send messages "from a models" in this iOS mesh library, but rather using any of `manager.send(...)` methods.

philips77, 2019-11-07:
So setting a publication to local node makes only sense it other nodes need to know or may modify to what address you send messages from this model (using `manager.publish(...)`. They can send `ConfigPublicationGet` to retrieve this information. But in your app using this library you may send any messages using `manager.send(...)`, that is independent from any model you have locally. You may now actually send a message that, based on your local models, you shouldn't be able to send to addresses that are not even in the mesh.

fiveagle, 2019-11-13:
Solved.
Apparently the requirement to set the opCode as 
0xC26007 = 0xC00000 | (0x02 << 16) | (0x02 my opcode) (0x0760 in Little Endian)
is enough to send a message successfully and having it back as UnknowMessage.

closed 2019-11-13 by fiveagle