nordicsemi/IOS-nRF-Mesh-Library #200: Stuck while requesting CompostionData

sunilloharshoreline1, 2019-11-11:
**App stuck once the composition data is requested after provisioning**

didReceiveMessage is called but the MeshMessage is of type UnknownMessage. It should be ConfigCompositionDataStatus

UnknownMessage(opCode: 0x000002, parameters: 0059000000000028000300000011000000020000100410061007100210001301130312041203130413071308130C130D130000020006130210000002000A130210000002000B13021000000F0000100410061007100210001301130312041203130413071308130C130D130000020006130210000002000A130210000002000B13021000000F0000100410061007100210001301130312041203130413071308130C130D130000020006130210000002000A130210000002000B130210)
  - opCode : 2
  ▿ parameters : Optional<Data>
    ▿ some : 189 bytes
      - count : 189
      ▿ pointer : 0x00000001c0172840
        - pointerValue : 7517710400

------------------------------------------------------------------
here are the logs : 

Connecting...
Discovering services...
Initializing...
bearer open
Identifying...
State Change to : Requesting Provisioning Capabilities
Identifying...
State Change to : Provisioning Capabilities received
Identification Success
public Key:Optional(No OOB Public Key)
Authentication method:Optional(No OOB)
Unicast Address:Optional(28)
Provisioning...
State Change to : Provisioning started
Progress : 12
Provisioning is in progress
State Change to : Provisioning complete
Progress : 24
Provisioning is complete, now starting the Configuration...
Device disconnected..
Provisioning Success
State Change to : Provisioner is ready
2019-11-11 14:16:09.831628+0530 Aether Ble[328:39080] Warning: Attempt to present <UIAlertController: 0x1070bd200> on <Aether_Ble.ProvisionedNodeConfigurator: 0x107065600> whose view is not in the window hierarchy!
Progress : 60
Provisioner is ready
2019-11-11 14:16:09.841764+0530 Aether Ble[328:39080] [Bearer] Central Manager state changed to .poweredOn
2019-11-11 14:16:09.843629+0530 Aether Ble[328:39080] [Bearer] Connecting to nRF5x...
2019-11-11 14:16:09.857749+0530 Aether Ble[328:39120] [FoundationModel] Sending ConfigCompositionDataGet(page: 0) to: 001C
2019-11-11 14:16:09.859097+0530 Aether Ble[328:39120] [Access] Sending Access PDU (opcode: 0x8008, parameters: 0x00)
2019-11-11 14:16:09.866238+0530 Aether Ble[328:39120] [UpperTransport] Sending Upper Transport PDU (encrypted data: 0xCE3F00, transMic: 0x5EF93182) encrypted using key: nRF5x's Device Key
2019-11-11 14:16:09.867430+0530 Aether Ble[328:39120] [LowerTransport] Sending Access Message (akf: 0, szmic: 0, data: 0xCE3F005EF93182)
2019-11-11 14:16:09.872177+0530 Aether Ble[328:39120] [Network] Sending Network PDU (ivi: 0, nid: 0x04, ctl: 0, ttl: 5, seq: 62, src: 0001, dst: 001C, transportPdu: 0x819C728E393B759B, netMic: 0x79B37F2F) encrypted using Primary Network Key (index: 0)
2019-11-11 14:16:09.874391+0530 Aether Ble[328:39120] [LowerTransport] The bearer is closed.
2019-11-11 14:16:10.140202+0530 Aether Ble[328:39080] [Bearer] Connected to nRF5x
2019-11-11 14:16:10.140434+0530 Aether Ble[328:39080] [Bearer] Discovering services...
2019-11-11 14:16:10.440539+0530 Aether Ble[328:39080] [Bearer] Service found
2019-11-11 14:16:10.440663+0530 Aether Ble[328:39080] [Bearer] Discovering characteristrics...
2019-11-11 14:16:10.560452+0530 Aether Ble[328:39080] [Bearer] Data In characteristic found
2019-11-11 14:16:10.560574+0530 Aether Ble[328:39080] [Bearer] Data Out characteristic found
2019-11-11 14:16:10.560670+0530 Aether Ble[328:39080] [Bearer] Enabling notifications...
2019-11-11 14:16:10.739483+0530 Aether Ble[328:39080] [Bearer] Data Out notifications enabled
2019-11-11 14:16:10.739607+0530 Aether Ble[328:39080] [Bearer] GATT Bearer open and ready
2019-11-11 14:16:10.741032+0530 Aether Ble[328:39080] [Bearer] <- 0x0101001B1A7E285017407E00000000E9497702A2275AE4
2019-11-11 14:16:10.744139+0530 Aether Ble[328:39120] [Network] Secure Network Beacon (Network ID: 1B1A7E285017407E, IV Index: 0, Key Refresh Flag: false, IV Update active: false) receieved (decrypted using key: Primary Network Key (index: 0))
2019-11-11 14:16:10.745974+0530 Aether Ble[328:39120] [Proxy] New Proxy connected
2019-11-11 14:16:10.748527+0530 Aether Ble[328:39265] [Proxy] Sending SetFilterType(filterType: Whitelist) from: 0001 to: 0000
2019-11-11 14:16:10.749639+0530 Aether Ble[328:39265] [Network] Sending Control Message (opCode: 0x00, data: 0x00)
2019-11-11 14:16:10.754837+0530 Aether Ble[328:39265] [Network] Sending Network PDU (ivi: 0, nid: 0x04, ctl: 1, ttl: 0, seq: 63, src: 0001, dst: 0000, transportPdu: 0x6CB3, netMic: 0x9BE84B7C5C6F1C3D) encrypted using Primary Network Key (index: 0)
2019-11-11 14:16:10.757142+0530 Aether Ble[328:39265] [Bearer] -> 0x02042E7132041AB4DCEC6CB39BE84B7C5C6F1C3D
2019-11-11 14:16:10.830231+0530 Aether Ble[328:39080] [Bearer] <- 0x0204E86663BC2B33F871DD14E549CDBD19DF7A6AE366
2019-11-11 14:16:10.832710+0530 Aether Ble[328:39265] [Network] Network PDU (ivi: 0, nid: 0x04, ctl: 1, ttl: 0, seq: 0, src: 001C, dst: 0000, transportPdu: 0xDD14E549, netMic: 0xCDBD19DF7A6AE366) received
2019-11-11 14:16:10.834462+0530 Aether Ble[328:39265] [Network] Control Message (opCode: 0x03, data: 0x000000) receieved (decrypted using key: Primary Network Key (index: 0))
2019-11-11 14:16:10.836008+0530 Aether Ble[328:39265] [Proxy] FilterStatus(filterType: Whitelist, listSize: 0) received from: 001C to: 0000
2019-11-11 14:16:10.838650+0530 Aether Ble[328:39265] [Proxy] Sending AddAddressesToFilter(addresses: Set([65535, 1])) from: 0001 to: 0000
2019-11-11 14:16:10.839700+0530 Aether Ble[328:39265] [Network] Sending Control Message (opCode: 0x01, data: 0xFFFF0001)
[... 111 lines cut ...]
2019-11-11 14:16:18.136864+0530 Aether Ble[328:39265] [Access] Response Access PDU (opcode: 0x 2, parameters: 0x0059000000000028000300000011000000020000100410061007100210001301130312041203130413071308130C130D130000020006130210000002000A130210000002000B13021000000F0000100410061007100210001301130312041203130413071308130C130D130000020006130210000002000A130210000002000B13021000000F0000100410061007100210001301130312041203130413071308130C130D130000020006130210000002000A130210000002000B130210) receieved (decrypted using key: nRF5x's Device Key)
2019-11-11 14:16:18.138298+0530 Aether Ble[328:39120] [Bearer] -> 0x00045063269B2BF1EEE1C0F7D6618E0ED41535934A948C26B0
2019-11-11 14:16:18.152068+0530 Aether Ble[328:39080] [Bearer] <- 0x0004A23A1473DC4DC72D69B1D7685CE4C6198B56C2FF8B5840BA8DE685E1
2019-11-11 14:16:18.153592+0530 Aether Ble[328:39265] [Network] Network PDU (ivi: 0, nid: 0x04, ctl: 0, ttl: 4, seq: 59, src: 001C, dst: 0001, transportPdu: 0x69B1D7685CE4C6198B56C2FF8B5840BA, netMic: 0x8DE685E1) received
2019-11-11 14:16:18.155397+0530 Aether Ble[328:39265] [LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 2, segO: 15, segN: 16, data: 0x58C61408DCBB1B6B7943F73F) receieved (decrypted using key: Primary Network Key (index: 0))
2019-11-11 14:16:18.156106+0530 Aether Ble[328:39265] [LowerTransport] Message already acknowledged, sending ACK again
2019-11-11 14:16:18.156495+0530 Aether Ble[328:39265] [LowerTransport] Sending ACK (seqZero: 2, blockAck: 0x0001FFFF)
2019-11-11 14:16:18.159462+0530 Aether Ble[328:39265] [Network] Sending Network PDU (ivi: 0, nid: 0x04, ctl: 1, ttl: 5, seq: 70, src: 0001, dst: 001C, transportPdu: 0xC6C01EC05B0087, netMic: 0x4C10F1BF1D39669C) encrypted using Primary Network Key (index: 0)
2019-11-11 14:16:18.160307+0530 Aether Ble[328:39265] [Bearer] -> 0x0004AC16DCBE71256D35C6C01EC05B00874C10F1BF1D39669C
2019-11-11 14:16:18.209419+0530 Aether Ble[328:39080] [Bearer] <- 0x00042BEE0416DA139C04804E3C8FE3F10F2D09BF
2019-11-11 14:16:18.210964+0530 Aether Ble[328:39265] [Network] Network PDU (ivi: 0, nid: 0x04, ctl: 0, ttl: 4, seq: 60, src: 001C, dst: 0001, transportPdu: 0x804E3C8FE3F1, netMic: 0x0F2D09BF) received
2019-11-11 14:16:18.212503+0530 Aether Ble[328:39265] [LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 2, segO: 16, segN: 16, data: 0x5D2B) receieved (decrypted using key: Primary Network Key (index: 0))
2019-11-11 14:16:18.212830+0530 Aether Ble[328:39265] [LowerTransport] Message already acknowledged, sending ACK again
2019-11-11 14:16:18.213156+0530 Aether Ble[328:39265] [LowerTransport] Sending ACK (seqZero: 2, blockAck: 0x0001FFFF)
2019-11-11 14:16:18.216008+0530 Aether Ble[328:39265] [Network] Sending Network PDU (ivi: 0, nid: 0x04, ctl: 1, ttl: 5, seq: 71, src: 0001, dst: 001C, transportPdu: 0xB6E05F7DBDC731, netMic: 0x711A3F53571875D7) encrypted using Primary Network Key (in [... cut ...]
2019-11-11 14:16:18.273398+0530 Aether Ble[328:39265] [LowerTransport] Sending ACK (seqZero: 2, blockAck: 0x0001FFFF)
2019-11-11 14:16:18.276698+0530 Aether Ble[328:39265] [Network] Sending Network PDU (ivi: 0, nid: 0x04, ctl: 1, ttl: 5, seq: 72, src: 0001, dst: 001C, transportPdu: 0x003E8671082A85, netMic: 0x3DC6ED5D113B5227) encrypted using Primary Network Key (index: 0)
2019-11-11 14:16:18.277699+0530 Aether Ble[328:39265] [Bearer] -> 0x00044C294752C92F6986003E8671082A853DC6ED5D113B5227

------------------------------------------------------------------

philips77, 2019-11-11:
You have to set the `localElements` after you create your `MeshNetworkManager`.
Have a look at: https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/master/Documentation/GETTING_STARTED.md#local-elements-and-models 
and https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/440e47bbcbe28b4ea8e0377bdb0d502c927124fe/Example/nRFMeshProvision/AppDelegate.swift#L99-L115 in sample app. After setting the local elements (even if they are empty), the delegates for Configuration Server and Configuration Client models are set.

sunilloharshoreline1, 2019-11-12:
Yes. I didn't set the localElements. If I set it as empty it works. 
Thanks for your help.

closed 2019-11-12 by sunilloharshoreline1