nordicsemi/IOS-nRF-Mesh-Library #201: Scenes & Scheduling of Scenes?

deverickasih, 2019-11-12:
**Describe the solution you'd like**
I'd like to be able to create a scene for the whole mesh network (which node lights up, which node is dimmed and which node has its lights off). I'd also like to be able to set the time of day a particular scene will take place at - so that when the time comes, each node in the network changes its state(s) without input from the provisioner.

**Describe alternatives you've considered**
Setting up two groups, (1) a group that will tell all its nodes to turn off & (2) a group that will tell all its nodes to turn on. And upon activating the scene, i turn on one group and turn off the other group. **Problem :** if I do it this way, what about dimmer models and other models, I may have to set up a group for every single distinct state available in the particular scene. Or is that actually the idea of a scene?

philips77, 2019-11-12:
Scenes are more like what's in your first desired solution. They're not implemented yet in this library (we gladly accept PRs!). You may read about them in the Bluetooth Mesh Model 1.0.1 specification.
You'd have to implement some required messages in the Generic Messages, as far as I remember. However, scenes should also be saved in the database, and the spec for this isn't public and final yet (that what we have least it for the future).
Also, scenes must be supported on your devices by enabling Scene Models.

philips77, 2019-11-19:
I will try to implement Scene messages in the next version, but without any UI in the sample app.

deverickasih, 2019-11-20:
Great!!

philips77, 2019-11-20:
Have a look at #209.

philips77, 2020-08-29:
Fyi: I'm starting implementing scenes. Hopefully the server and client impl + UI on the sample app will be ready in few weeks.

olsky, 2020-08-29:
Could you please point to the mesh models of scenes?

philips77, 2020-08-31:
Are you talking about the spec? They're defined in Bluetooth Mesh Model 1.0.1 as Scene Server and Scene Setup Server, I guess. I've just started reading spec...

philips77, 2021-02-05:
Scenes are supported in version 3.

closed 2021-02-05 by philips77