nordicsemi/IOS-nRF-Mesh-Library #202: Generic Message for not already implemented models

igathedub, 2019-11-13:
**Is your feature request related to a problem? Please describe.**
Right now we are trying to test Time [Setup] Server (and after that Scheduler) model. The communication between the nodes is encrypted, so we wanted your example app to be the client (request initiator)

**Describe the solution you'd like**
It would be extremely helpful if we could use something like Vendor Model's "Controls" part for all models but without 0xC0 restriction. 

**Describe alternatives you've considered**
"Generic Message Sender"-pane where you can choose Unicast Address + Model and send Generic Message to it (Encrypted with App & Net-Keys)

**Additional context**
Can you give us hints or even how we can implement it?

philips77, 2019-11-19:
Hi,
This is a good idea. I will try proposing it on our backlog planning.

philips77, 2021-02-05:
We're about to release version 3.0, but this feature is not in it. Just letting you know that I still remember.