nordicsemi/IOS-nRF-Mesh-Library #203: Direct connection provisioning

ximplarandy, 2019-11-14:
**Is your feature request related to a problem? Please describe.**
When provisioning a new device with 3 or more provisioned device, the app would connect to any one of them, and provision the new device via mesh messaging. However, it seems very unreliable, and would die more frequently as the number of devices increase.

**Describe the solution you'd like**
Can we have the v1 way?  Like connecting to the new device directly, which rarely fail.

**Describe alternatives you've considered**
Nothing much except powering down other cards, which is not efficient. 

**Additional context**
--

philips77, 2019-11-14:
This has been added in version 2.1, available now on master branch and released on AppStore. On Proxy tab disable the Automatic Connection switch and select target proxy manually.

ximplarandy, 2019-11-15:
Thank you very much.  Found the setting and seems to be much more reliable.

closed 2019-11-15 by ximplarandy