nordicsemi/IOS-nRF-Mesh-Library #204: Logging page

olsky, 2019-11-15:
It would be good to have a separate Log-Messages-Page (nice-to-have: (a) setting for log level DBG/INF/WRN/ERR; (b) enable/disable).

All the printfs would go there:
- in device scan
- in device-discovery
- node provisioning
- setup
- models interaction

this would help debugging.

Alternatively, one needs to connect the xcode and uart-trace on the other side.

Would it be good-feature? :)

philips77, 2019-11-16:
You may connect your iPhone to Mac, run Console app, select the phone on the left menu. Then perform some action on nRF Mesh, e.g. send some messages, and you'll see the logs there. You may set a filter on Console app to only display logs from this app, or any of the components.

olsky, 2019-11-16:
Thanks for the hint!

Do you think the page in the app could be an alternative? 

In this case there is no need to connect to the laptop.

philips77, 2019-11-16:
Yes, I was thinking about it, but didn't have time eventually. Perhaps later.

olsky, 2019-11-16:
> Yes, I was thinking about it, but didn't have time eventually. Perhaps later.

I'm not a swift guru, but if you draft a macro flow -- can try it :)