nordicsemi/IOS-nRF-Mesh-Library #206: Send ConfigAppKey callback

fiveagle, 2019-11-18:
**Describe the bug**
I'm trying to send a ConfigAppKey message but I don't receive anything back.

**To Reproduce**
Steps to reproduce the behavior:
I try to send the message this way:

```swift
try MeshNetworkManager.instance.send(ConfigAppKeyAdd(applicationKey: self.appKey), to: self.node)
```

But I only see the log below.

Thanks

**Expected behavior**
A clear and concise description of what you expected to happen.

**Platform details:**
 - Device: iPhone6s
 - OS: iOS13.2
 - Library Version: 2.1.0

**Logs / Screenshots**
```
2019-11-18 17:13:39.631530+0100 P406[1077:832746] [Bearer] -> 0x0047C612AAC50572DD91BD7F26092086F8844C4979F14C8A009590D3EEBD
2019-11-18 17:13:39.644700+0100 P406[1077:832746] [Bearer] -> 0x00475DF51AFBC5CC3758DAFEB15575DF719206E7612F2857735559D8F454
2019-11-18 17:13:40.522126+0100 P406[1077:832746] [Bearer] -> 0x0047C55FD0AD030C70E109908CA3AC24C0D8AEB50AE7C7A2ECC9E4D15E31
2019-11-18 17:13:40.527975+0100 P406[1077:832746] [Bearer] -> 0x0047AD175222D419BC3C5537AF60C8C762245D594EC1B53E838FD1D44172
2019-11-18 17:13:41.447101+0100 P406[1077:833014] [Bearer] -> 0x00476681A58E2DB059FC3BDE267A3C1930BB8D994B72D88FB4FB78B3C597
2019-11-18 17:13:41.454060+0100 P406[1077:833014] [Bearer] -> 0x0047A878FA9E84EBAA584523E4D5A8ADF4AF82B1B1C8676D5B3472CF092C
2019-11-18 17:13:44.187289+0100 P406[1077:832735] [Bearer] -> 0x0047973E6E67A01B3F8CB556E7DD05391A674662AE3DA7433FAC22395020
2019-11-18 17:13:44.192951+0100 P406[1077:832735] [Bearer] -> 0x004741F8A5C5E3931F330245DD73B3977335C98BE93E9A3E93CB9DB4EADF
2019-11-18 17:13:45.054127+0100 P406[1077:832746] [Bearer] -> 0x00470D15D91F91E5F1796C86AFBE44FB1619EA82C6383E33E9B02EE37146
2019-11-18 17:13:45.059563+0100 P406[1077:832746] [Bearer] -> 0x0047D02125662F917CE6450B244A52C1CE91D56C536A0163441E7D9F9824
2019-11-18 17:13:45.929002+0100 P406[1077:832746] [Bearer] -> 0x0047D352DD86E225ABAFC91B9A573630365C089ED9360278A7975CE39E5B
2019-11-18 17:13:45.934995+0100 P406[1077:832746] [Bearer] -> 0x0047958ECA283316CC9C43423B319D0896A7FFB07E6B82418B198C4D4D5E
2019-11-18 17:13:53.378681+0100 P406[1077:832735] [Bearer] -> 0x0047CEE2A0AE5F2E81B008714637E0E01934BAC365C73CA0AD52E9F7692A
2019-11-18 17:13:53.384626+0100 P406[1077:832735] [Bearer] -> 0x0047EDDC544BFB6124D53F884EED85C95AB3B7DF737A10E89F927E9E337E
2019-11-18 17:13:54.313565+0100 P406[1077:832746] [Bearer] -> 0x00478836E6096CCFEC8917159B54B902BEED31325977C5DBBEF844C1389C
2019-11-18 17:13:54.318125+0100 P406[1077:832746] [Bearer] -> 0x0047EF6B79C54F714986AC357BC0A43F4D858FE4D752D57B96018A1D7A16
2019-11-18 17:13:55.193633+0100 P406[1077:833061] [Bearer] -> 0x00476D830AE4C872B4AF44E3533353877BF0A882C33B2DDBD96AA21073B2
2019-11-18 17:13:55.198521+0100 P406[1077:833061] [Bearer] -> 0x00476613E165B5F71BCEB4C7314DC4EA27B91E691829542CF059BD3A8AF9
```

philips77, 2019-11-18:
Possible reasons are:
1. Too low sequence number, the messages are ignored by the receiver,
2. No node with target address,
3. The proxy node doesn't know the network key that's being used to encrypt those messages on network layer (first network key of the target node is used),
4. Proxy node has disabled proxy feature,
5. Target node not in range,
6. Too low TTL and the messages don't reach target node,
7. The target node doesn't know the network key used to encrypt the message (the key might have been deleted by some other provisioner).

fiveagle, 2019-11-19:
Ok thanks

closed 2019-11-19 by fiveagle

philips77, 2019-11-22:
Ok, there is a reason number 8.

8. The source Element's Unicast Address is not added to the Proxy Filter of the proxy. Your phone will not receive any messages targeting addresses that are not on the White List Proxy Filter or that are not on the Black List Proxy Filter.

Version 2.2 will add a detection of a "proxy with limited proxy filter capabilities". Some devices support only 1 device in their proxy filter. The nRF Mesh library by default adds at least 2 addresses: Unicast Addresses of all local Elements, Groups address to which at least one local Model is subscribed and address 0xFFFF (All Nodes). If the list count in returned status is still 1, the library will make sure that it is the 0th Element's Unicast Address that is successfully added. It will also call `ProxyFilerDelegate.limitedProxyFilterDetected(maxSize:)` with parameter 1, so you can switch to `.blacklist` filter there if you need to receive other messages as well.

2.2 is to be released shortly, check out **develop** branch for commits.