nordicsemi/IOS-nRF-Mesh-Library #207: Reading network messages via the proxy.

ximplarandy, 2019-11-19:
**Is your feature request related to a problem? Please describe.**
When using the proxy service, the network messages are not shown on the app

**Describe the solution you'd like**
It would be great if we could see the network messages on the app directly, as it is much easier when moving around.  Just plain output would help a lot!

philips77, 2019-11-19:
You may connect your iPhone to a Mac, start Console app on the Mac, select the phone there on the left menu and filter for subsystem `com.nordicsemi.nRFProvision`. You will see all the logs there. From my experience, you will have to disconnect and reconnect the phone from time to time.
We were thinking about doing a log screen, but displaying logs on Mac is much handier as the screen is bigger, filtering is already implemented, etc. Right now we don't have time to add log pane in the app, but we have it in our backlog.

philips77, 2019-11-19:
Duplicate of #204.

closed 2019-11-19 by philips77