nordicsemi/IOS-nRF-Mesh-Library #208: Crash during ConfigModelAppBind

fiveagle, 2019-11-19:
**Describe the bug**
I was sending a ConfigModelAppBind to each model server on my device.
All my steps are managed in code and I'm sending them one after the other with a delay of 0.4s between each.

Is possible that my delay is lower than the TTL and it creates this situtation leading to the crash? I've also tried with 0.7.

It is not a systematic crash.

Thank You.

```
2019-11-19 11:29:42.697228+0100 P406[1217:894513] Fatal error: Unexpectedly found nil while implicitly unwrapping an Optional value: file Pods/nRFMeshProvision/nRFMeshProvision/Classes/Mesh Messages/Foundation Messages/Configuration Messages/ConfigModelAppBind.swift, line 53
```

My steps are:
Provisioning
Get Composition Data
Get TTL
Adding AppKey (ConfigAppKeyAdd)
Iterate all the models and bind the app key on each.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Platform details:**
 - Device: iPhone6s
 - OS: iOS13.2
 - Library Version 2.1.0

**Logs / Screenshots**

```
Prepare provision
Start Scanning
Registered NetworkConnection Delegate: Optional<BearerDelegate>
Discovering services...
Initializing...
Identifying...
New state: Requesting Provisioning Capabilities
Identifying...
New state: Provisioning Capabilities received
CapabilitiesReceived:
Number of elements: 1
Algorithms: FIPS P-256 Elliptic Curve
Public Key Type: None
Static OOB Type: Static OOB Information Available
Output OOB Size: 0
Output OOB Actions: None
Input OOB Size: 0
Input OOB Actions: None
Provisioning...
New state: Provisioning started
New state: Provisioning complete
Provisioning complete.
Registered NetworkConnection Delegate: Optional<BearerDelegate>
Finish provisioning node 1888_593!!!
Prepare key binding
Registered NetworkConnection Delegate: Optional<BearerDelegate>
493 bytes

2019-11-19 11:29:29.457351+0100 P406[1217:894513] [Bearer] Central Manager state changed to .poweredOn
2019-11-19 11:29:29.458158+0100 P406[1217:894513] [Bearer] Connecting to 1888_593...
2019-11-19 11:29:29.781472+0100 P406[1217:894513] [Bearer] Connected to 1888_593
2019-11-19 11:29:29.781807+0100 P406[1217:894513] [Bearer] Discovering services...
Config Key Discovering services...
2019-11-19 11:29:30.836064+0100 P406[1217:894513] [Bearer] Service found
2019-11-19 11:29:30.836200+0100 P406[1217:894513] [Bearer] Discovering characteristrics...
2019-11-19 11:29:32.075845+0100 P406[1217:894513] [Bearer] Data In characteristic found
2019-11-19 11:29:32.075995+0100 P406[1217:894513] [Bearer] Data Out characteristic found
2019-11-19 11:29:32.076167+0100 P406[1217:894513] [Bearer] Enabling notifications...
Config Key Initializing...
2019-11-19 11:29:33.062694+0100 P406[1217:894513] [Bearer] Data Out notifications enabled
2019-11-19 11:29:33.062883+0100 P406[1217:894513] [Bearer] GATT Bearer open and ready
Network Connection changed isOpen: true
Config Key Bearer did open
2019-11-19 11:29:33.079902+0100 P406[1217:894865] [Bearer] -> 0x002C596FE9A08FEEE03A2CAB87D2E2A22A7C641D192F
2019-11-19 11:29:33.310157+0100 P406[1217:894513] [Bearer] <- 0x0101003F7F0EF505136F4200000000215AD786573F87A5
2019-11-19 11:29:33.313406+0100 P406[1217:894875] [Bearer] -> 0x022CBF205D7121D145CAF76E29ED367CE0645DFA
2019-11-19 11:29:33.559411+0100 P406[1217:894513] [Bearer] <- 0x002C3B6DAD26591AE5EA56EA61FCB16CC1439872EF99B90E34F0FDD8B49F
2019-11-19 11:29:33.975604+0100 P406[1217:894865] [Bearer] -> 0x002C71CBF1BE7FF9D21D9B4A62173394C7EF3DC8BA5DE36A1B
2019-11-19 11:29:34.053870+0100 P406[1217:894513] [Bearer] <- 0x002C364E3AE5CC9DBDFE369318D34DCD6D0B49F9415DC0C2E4A5885F530C
2019-11-19 11:29:34.481225+0100 P406[1217:894875] [Bearer] -> 0x002C38AAACBF42046AE94C59414A2B3276172C25D3AB46FE19
2019-11-19 11:29:34.549447+0100 P406[1217:894513] [Bearer] <- 0x002C33610F31B20896D2B80355A61735D39B2C36CA10DE309CECD58AD2FD
2019-11-19 11:29:34.962987+0100 P406[1217:894877] [Bearer] -> 0x002CCCA479CC44A5470C09F840EA0363BE4F9F5C059894035E
2019-11-19 11:29:35.042433+0100 P406[1217:894513] [Bearer] <- 0x002C0B8014AA597C38FE119BA1C81FD326B5AB75
2019-11-19 11:29:35.045998+0100 P406[1217:894875] [Bearer] -> 0x002C438FD7F47018D672D3C5307E6BDFAE272AD0C88479C732
2019-11-19 11:29:35.291982+0100 P406[1217:894513] [Bearer] <- 0x022C29A882A1990E8CB4AE130DBEE30ABD0714C578C7
2019-11-19 11:29:35.296172+0100 P406[1217:894865] [Bearer] -> 0x022C22203A016A94D0E9AD5DA3DFDED6A8092AA6A5E567
Requesting default TTL...
2019-11-19 11:29:35.483226+0100 P406[1217:894865] [Bearer] -> 0x002C9B35CEC8262A30A68E6FB6AD5FE9460ABB64F9
2019-11-19 11:29:35.542894+0100 P406[1217:894513] [Bearer] <- 0x002C5F37CCD68C4C5849955E2EE6C959DABA18927A6B085D8BA4115EFAA5
2019-11-19 11:29:35.546447+0100 P406[1217:894877] [Bearer] -> 0x002C5906D856DC8BFC92C636052E489B0C8DD756D5B9D5EE09
2019-11-19 11:29:36.032337+0100 P406[1217:894513] [Bearer] <- 0x002C5DBB77B344982442EAF2C4F40FBB05115E286996D5C8C41705CE785D
2019-11-19 11:29:36.035717+0100 P406[1217:894877] [Bearer] -> 0x002CB211637727C29E0C0DD5BD08891B3CE7A7065F1060B1CE
2019-11-19 11:29:36.529252+0100 P406[1217:894513] [Bearer] <- 0x002C8DA4C24C4F662C32D4E9BFBDFB8C8C496FF5
2019-11-19 11:29:36.532845+0100 P406[1217:894865] [Bearer] -> 0x002C499F88414A8E65E486C68B72C700F23138944325231C46
2019-11-19 11:29:36.779636+0100 P406[1217:894513] [Bearer] <- 0x022C2E6E2750BA20E185286ADB08A9FB536FB00C3714
10ab_dbmesh succed
2019-11-19 11:29:37.271501+0100 P406[1217:894513] [Bearer] <- 0x002C641D3E0E713D75858F420D8D815E1D23F234CEAF
Node TTL aquired
10ab_dbmeshnetwork succed
10ab_dbmeshnetwork succed
10ab_dbmeshnetwork succed
Adding Application Key...
2019-11-19 11:29:37.736046+0100 P406[1217:894865] [Bearer] -> 0x002CB25299D3784F3B5D430116AC0CF5BCA515EB64232B4D0D9F1E679777
2019-11-19 11:29:37.740959+0100 P406[1217:894865] [Bearer] -> 0x002C9383576E6029983E1D0E7822B38F1928FC876751A325ADC9C153FA8E
2019-11-19 11:29:38.012739+0100 P406[1217:894513] [Bearer] <- 0x002CF05C4AF23F73DF8BC8511B56A1A7428A8F35D4B3CDC88C
MeshNetworkManager failed to send Bind Message - error: Request timed out.
Riprovo il binding
2019-11-19 11:29:38.508655+0100 P406[1217:894513] [Bearer] <- 0x002CFAA3ACBEDA5A0D4A878E385765684716EB5FEECF8DE9C8
Binding...
Health Server keys: [testing Application Key (index: 0)]
2019-11-19 11:29:38.973718+0100 P406[1217:894879] [Bearer] -> 0x002C021E7CA5DA49D52BB0AC74DE0391A2DC9B996DCE26D796175B
10ab_dbmeshnetwork succed
2019-11-19 11:29:39.253725+0100 P406[1217:894513] [Bearer] <- 0x002CBB8F611FBFCE5702C7021FD9D614D66E4303B1298CAF2F623DA2
Success get config
Binding...
10ab_dbmeshnetwork succed
Scene Server keys: [testing Application Key (index: 0)]
2019-11-19 11:29:39.738619+0100 P406[1217:894879] [Bearer] -> 0x002CD704A17D7983F1B1F0CBDA6420D8886F303E479C1F7023FEC7
2019-11-19 11:29:39.929758+0100 P406[1217:894877] [Bearer] -> 0x002CA0F380114874A6E0095956988189715382E1AD89
2019-11-19 11:29:39.994635+0100 P406[1217:894513] [Bearer] <- 0x002CD9C7471FA1A76F52429055EFE8A1370A3F71963076630441B6EE
Success get config
Binding...
10ab_dbmeshnetwork succed
Scene Setup Server keys: [testing Application Key (index: 0)]
2019-11-19 11:29:40.453087+0100 P406[1217:894879] [Bearer] -> 0x002CC573C0DCA7CAC68D9B52741CB333CBECC33CD85357EB6EB710
2019-11-19 11:29:40.493375+0100 P406[1217:894513] [Bearer] <- 0x002CBCF80B259205620611B90260C03957BBAAEBBE310FC59A0ADE80B03F
2019-11-19 11:29:40.914347+0100 P406[1217:894879] [Bearer] -> 0x002CBFFF649C4028B131B5551BFFD9A87CB4EA5C74C4AC13AB
2019-11-19 11:29:40.986845+0100 P406[1217:894513] [Bearer] <- 0x002C468DB716627CCCBEE8937922DC30C7791B281117D58AD6900E7E35E4
2019-11-19 11:29:41.403789+0100 P406[1217:894875] [Bearer] -> 0x002C42148007D58148FEFDD013981761EED3516B6DA0ECF9FA
2019-11-19 11:29:41.478897+0100 P406[1217:894513] [Bearer] <- 0x002C1919A7E16FED32186445CA80A25588DA57D64D36AE78E412AE2E97CF
2019-11-19 11:29:41.903181+0100 P406[1217:894875] [Bearer] -> 0x002C134342A2091AE640B4EDC1AB5BB93F36C0397083E7C07B
2019-11-19 11:29:41.973755+0100 P406[1217:894513] [Bearer] <- 0x002C130DC39D7B5C0CA6A6E93E28876C820BB6D6
2019-11-19 11:29:42.003792+0100 P406[1217:894875] [Bearer] -> 0x002C1A34EE4558F11A6A530849648B51E12CABEB9212CEE3C7
2019-11-19 11:29:42.223443+0100 P406[1217:894513] [Bearer] <- 0x002C8B9323E328139A2B34B9A1338D5194A087923EACBF1B1AF0B926
Success get config
Binding...
Requesting default TTL...
2019-11-19 11:29:42.472750+0100 P406[1217:894879] [Bearer] -> 0x002C4687B9FC1F8CDB9A2A21290E28167DC3EDAA77
Scene Client keys: [testing Application Key (index: 0)]
2019-11-19 11:29:42.697228+0100 P406[1217:894513] Fatal error: Unexpectedly found nil while implicitly unwrapping an Optional value: file Pods/nRFMeshProvision/nRFMeshProvision/Classes/Mesh Messages/Foundation Messages/Configuration Messages/ConfigModelAppBind.swift, line 53
```

philips77, 2019-11-19:
Hi,
This indeed looks like a bug. For some reason the model you're using has `parentElement` set to `nil`. It's a `weak` variable to avoid cyclic reference, but the the mesh network should hold references to everything, so they should not be disposed. As a temporary fix, make sure that all `model`s, that you put to `ConfigModelAppBind` messages have `parentElement` set.

philips77, 2019-11-20:
Why do you have
```
Binding...
Requesting default TTL...
```
at the end?
Is the TTL request something you intend to do at the same time?

philips77, 2019-11-22:
The latest commit adds some protection against this crash. The `parentElement` in `Model`, as it's `weak`, became optional instead of implicitly unwrapped, so you have to ensure that the message gets created.
I still don't know how it is possible that you had a `model` without `parentElement` set, but this at least will fix the crash.

closed 2019-11-22 by philips77