nordicsemi/IOS-nRF-Mesh-Library #210: can't display two or more provisioned devices

zhangjinliang1985, 2019-11-20:
Hello. I am using the example with the version 2.1.0. It seems that the APP can not display two nodes 
after two devices are provisioned. Is there anything I missed?

philips77, 2019-11-20:
Has the provisioning finished successfully? Did you use unique Unicast Address for each of them? 
You need to provide more information, like screenshots, logs, etc.

zhangjinliang1985, 2019-11-20:
Yeah. The provisioning finished successfully.
1. The first node's Unicast Address is 0x0001. You can see it on the screenshot below:
[image]

2. But after the second node has been provisioned successfully, the first node was disappeared. Only the second node is shown in the app. 
 
[image]

philips77, 2019-11-20:
This is weird. I just repeated what you did and I see 2 devices (+ the provisioner). Can you send us the hex files you are using for both nodes? Can they be flashed onnRF52 DK?

zhangjinliang1985, 2019-11-21:
yeah. i am using zephyr-based device. i will give you the hex files. There are two hex files. One is the boot hex file and the other is app hex file. You should flash both of them on nRF52 DK.  I have sent you a email. Please check it.

zhangjinliang1985, 2019-11-21:
Oh. I just remember that the demo app of the version 1.0.2 does not have this problem.

zhangjinliang1985, 2019-11-21:
The hex files are for the nrf52832 chip.

zhangjinliang1985, 2019-11-21:
I also have tried the android app of the newest version. It also has this problem.

philips77, 2019-11-21:
Thank you, I did get the hex files and will check them today.

philips77, 2019-11-21:
I think the issue is that both your devices advertise with the same UUID, so the app assumes it's the same device and just replaces it. Each device should have a random UUID instead.

YoYo0304, 2019-11-21:
OK. let me try. thank you

philips77, 2019-11-22:
Did it work?

YoYo0304, 2019-11-29:
yeah. Thank you very much

closed 2019-11-29 by YoYo0304