nordicsemi/IOS-nRF-Mesh-Library #211: How do you create a Guest Non-Provisioner User?

deverickasih, 2019-11-20:
According to the example app, a provisioner without a unicast address cannot configure any nodes. This means a provisioner without a unicast address can be used for Guests?

I am trying to understand how to define a Guest User - who do not configure the nodes but should be able to interact with the nodes (e.g. turn on/off nodes, activate scene, turn on/off groups) in the mesh network given to him/her to access.

My questions (with regards to the example project) are :

(1) why does a provisioner who is not going to configure anything, need to allocate ranges for unicast address, group address and scene?

(2) why does a provisioner who does not have a unicast address be able to change the name of the element but not turn it on/off?

(3) Or is there another way i could use to create a **Guest User** who can turn on/off nodes and groups and activate scenes without having to add a second provisioner to the mesh network?

philips77, 2019-11-20:
> According to the example app, a provisioner without a unicast address cannot configure any nodes.

Yes. It cannot send any messages to the network. Not only configure them, but also control them. It only can provision new nodes and assign them Unicast Address and primary Network Key. It's really stupid provisioner :)

> This means a provisioner without a unicast address can be used for Guests?

No. For Guests you should use normal Provisioner with Unicast Address assigned. You should not expose the configuration capabilities to guests, but this is only on application layer of your app. You may still use this mesh library in guest app, but through the UI allow only selected actions, like turning on/off nodes, etc. You may have any model you want, e.g. give users permissions to control only lights, locks, configure their rooms, etc.

> (1) why does a provisioner who is not going to configure anything, need to allocate ranges for unicast address, group address and scene?

For Unicast Address range it needs to have at least its own addresses assigned. Then it will not be able to provision any more nodes, as it has used 100% of it's own range. For other ranges, they can be empty (at least I hope so, I removed a validation check that ensured at least 1 address/scene was added).

> (2) why does a provisioner who does not have a unicast address be able to change the name of the element [isn't this configuring the element?] but does not have the ability to turn it on/off [on/off lights shouldn't be seen as a configuration] ?

The name is not a configuration element. It's just saved locally on the phone and in the JSON. But after the network is shared, each provisioner can rename the same node to be identified by any name they want.

(3) Or is there another way i could use to create a Guest User who can turn on/off nodes and groups and activate scenes without having to add a second provisioner to the mesh network?

By spec, each Node that should be able to send any messages must have a Unicast Address assigned. There are 2 ways of adding new Nodes: provisioning a new one or just declaring it and sharing the network configuration oob (out of band, for example using JSON). This library does not support server-side provisioning (so a phone with an app that's using this library cannot be provisioned by other provisioner). Instead, the mesh network configuration should be shared with this app using export/import feature. You don't have to call this new Node a Provisioner. In the library it will be in fact a Provisioner (with very limited Unicast Address range and other ranges empty), but it will never perform any configuration as this will be hidden in the guest app. Also, if you share the network configuration to such Guest node, you may limit the amount of information you share in the JSON file. This is not yet supported in the library, but I imagine that a guest could receive only information about nodes that have the guest key added and all other nodes, including other guest nodes and provisioners) would be excluded from the JSON. So they would not be able to mess up with your network after, let's say, intercepting the JSON file and using nRF Mesh app (where all the configuration features are exposed). 

Keep in mind, that if a user intercepts the JSON, they can:
1. extend their ranges and provision new nodes,
2. reset all the nodes they have access to (unless there is a protection in the firmware prohibiting resetting using non-primary network key), and provision them with their own keys,
3. reconfigure the nodes to make them flood the network (e.g. by setting periodical publication with short interval),
3. if other nodes were included in the JSON (including the home-owner master provisioner), they can reset their phones remotely, as the Device Keys are in the JSON.

Guests should receive only this part of the network, that they can destroy and you should include some protection on the fw side if you are afraid that guests can do damage.

philips77, 2019-11-20:
A better way to secure nodes from being reconfigured is not to share the real Device Keys with the guests. As the Device Key field is required in the Node, you should change it to a dummy value before sharing the JSON, so even without any protection, the Nodes are safe (configuration requires Device Key).

philips77, 2019-11-20:
Or, even better, exclude all Node objects (except the guest node) from the JSON when sharing, and just use groups. Users should be able to turn off/on bulbs only by using groups or virtual labels.

philips77, 2019-11-20:
So, answering your initial question, what differentiate the Provisioner and the Node is just the information it has. If you give a Node the primary network key and device keys, it becomes a Provisioner, but if you have a Provisioner that knows nothing except groups and guest keys, it's happy it can at least turn light on.

philips77, 2019-11-20:
One last thing. Removing the guest from the network requires an update or removal of the Guest Network Key from all the Nodes. Otherwise, the guest could cause problems in the future. So you have to make sure that with the JSON file the guest got it will no longer be able to send any messages.

closed 2019-11-21 by deverickasih