nordicsemi/IOS-nRF-Mesh-Library #214: Vendor model companyIdentifier is zero

codepgq, 2019-11-23:
[image]

My Vendor model ID is 0x02110000, but after this conversion it becomes 0, it should be 0x0211.

philips77, 2019-11-23:
I think this is correct. Your company identifier is 0x0211 (Telink Semiconductor Co. Ltd) and model of is 0x0000. There's no company id with number 0x0000. On the other hand, Model IDs usually start from 0 (this depends on the company, of course).

philips77, 2019-11-23:
Vendor model IDs are defined by 2 values: the company id, defined here: https://www.bluetooth.com/specifications/assigned-numbers/company-identifiers/, and 16-bit model id, assigned by the company.

closed 2019-11-23 by philips77