nordicsemi/IOS-nRF-Mesh-Library #216: How to import the first & second time?

deverickasih, 2019-11-25:
What i mean by the title is how to import (1) the first time when there is no mesh network yet and (2) the second time, replacing the current mesh network.

Below is the function i use. I am quite confused with the outcome of the code. Sometimes it works, and sometimes it doesn't.
``
func initializeMeshNetwork(withName: String, withData: Data) {

     // setting up the mesh network like in the example code
     meshNetworkManager = MeshNetworkManager()
     ...

     // import data from cloud if there is data
     if (!withData.isEmpty) {
          do {
               let result = try meshNetworkManager.import(from: withData)
               print("result 01 = \(result)")
          } catch {
          }
     }

     // loading the mesh network from cloud
     var loaded = false
     do {
          loaded = try meshNetworkManager.load()
          print("result 02 = \(loaded)")
     } catch {
     }
     
     if (!loaded) {
          createNewMeshNetwork(name: withName)
     } else {
          print("Using Network from cloud with name - \(withName)")
          meshNetworkDidChange()
          print("mesh network current name is \(meshNetworkManager.meshNetwork?.meshName)")
     }

     ...
}
``

**Question 1** : After i import, do i need to load (as done above) or is it automatically loaded with the import function (so i don't have to load again)?
**Question 2** : When 'withData' is not empty, i sometimes get "result 02 = false". [Because of this i end up creating a new mesh network although i already have a network gotten from the cloud.] What is the normal cause for this?
**Question 3** : For every mesh network that i export to cloud, i set it a name. When i import a different mesh network over an existing mesh network, the name from "meshNetworkManager.meshNetwork?.meshName" is from the previous mesh network. Why is that? What am i doing wrong?

philips77, 2019-11-25:
**Ad. 1.** No, you actually can't call `load(from:)` after importing, as it will override the just imported network. Import replaces the local `meshNetwork` variable, just like `load()` does. However, you have to call `save()`** in order the network to be saved. 

Have a look how the sample app does it around here:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/10acde1cadc78f6ca08082cea8002578495d6491/Example/nRFMeshProvision/View%20Controllers/Settings/SettingsViewController.swift#L220

**Ad. 2.** `load()` returns `false` when there was no network saved before.

**Ad. 3.** I think you just should remove `load()` after importing. 

Moreover, you may extend the `Storage` protocol and create a class/struct that will keep the data in the cloud. You don't have to use the default `LocalStorage` and use import/export instead, but of course this will also work. But this is the reason I made the `Storage` protocol open, so you can implement your own.

deverickasih, 2019-11-28:
Removing Load after Import solves the issue. Great info!! Thanks.

closed 2019-11-28 by deverickasih

philips77, 2019-11-28:
One last comment. The Json scheme isn't final yet, so it may change. We will adapt both Android and iOS libraries when it hits public, until then you shouldn't assume it's final. Of course only minor changes will be made, I hope...