nordicsemi/IOS-nRF-Mesh-Library #217: Provisioning Problem.

deverickasih, 2019-11-25:
I have recently found a persisting error when trying to provision a device. The following are the logs. After the end of the logs, the provisioning just stops.

```
[Provisioning] Provisioning started
[Provisioning] Sending Provisioning Start (algorithm: FIPS P-256 Elliptic Curve,
public Key: No OOB Public Key, authentication Method: No OOB)
[Bearer] -> 0x03020000000000
[Provisioning] Sending Provisioner Public Key
(0x989E389F9A934E6B6DBC032722FEDB37A3930428EED3B55AB19BF8227C6A46220D20CA5060AE6E9F62432C7B33DE301588ACD7DF87F63BFE29446953B25A4D61)
[Bearer] ->
0x0303989E389F9A934E6B6DBC032722FEDB37A3930428EED3B55AB19BF8227C6A46220D20CA5060AE6E9F62432C7B33DE301588ACD7DF87F63BFE29446953B25A4D61
[Bearer] <-
0x03035A401C565CC79EF24A16AD6EADB7B5913DDFEBC99B2C967F7DEC91594DA02FCA2603F01533B7041999B59B12CE443BCD330D7E1D42F254AE641579CFE475834B
[Provisioning] Device Public Key
(0x5A401C565CC79EF24A16AD6EADB7B5913DDFEBC99B2C967F7DEC91594DA02FCA2603F01533B7041999B59B12CE443BCD330D7E1D42F254AE641579CFE475834B)
received
[Provisioning] Sending Provisioner Confirmation (0x909BA367E8328AADD2F7B8A89356F6A0)
[Bearer] -> 0x0305909BA367E8328AADD2F7B8A89356F6A0
[Bearer] <- 0x030563AAA367913EE5FDC676351B01796669
[Provisioning] Device Confirmation (0x63AAA367913EE5FDC676351B01796669) received
[Provisioning] Sending Provisioner Random (0xC978216FFBD924F89EE441534DF9C60A)
[Bearer] -> 0x0306C978216FFBD924F89EE441534DF9C60A
[Bearer] <- 0x03063F6509E8919222FDE1F944EA4B13C90E
[Provisioning] Device Random (0x3F6509E8919222FDE1F944EA4B13C90E) received
[Provisioning] Sending Encrypted Provisioning Data
(0x23D4FC9A836908864B65D97D4DF2B0D0894158F26C921C3E8EEB98E3FE8A2D80D0)
[Bearer] -> 0x030723D4FC9A836908864B65D97D4DF2B0D0894158F26C921C3E8EEB98E3FE8A2D80D0
[Bearer] <- 0x0308
[Provisioning] Provisioning Complete received
[Provisioning] Provisioning complete
Jan : Provisioning complete.
[Bearer] Cancelling connection...
Jan : Proceeding to configuration of node...
Jan : Requesting Composition Data...
Jan : Result of ConfigCompositionDataGet = MessageHandle(opCode: 32776, source: 1,
destination: 2, manager: Optional(nRFMeshProvision.MeshNetworkManager))
[FoundationModel] Sending ConfigCompositionDataGet(page: 0) to: 0002
[Bearer] Disconnected from AFB504-C148BA22199A
[Provisioning] Provisioner is ready
[Access] Sending Access PDU (opcode: 0x8008, parameters: 0x00)
[UpperTransport] Sending Upper Transport PDU (encrypted data: 0x51E5C5, transMic: 0xDACAC53D) encrypted using key: AFB504-C148BA22199A's Device Key
[LowerTransport] Sending Access Message (akf: 0, szmic: 0, data: 0x51E5C5DACAC53D)[Network] Sending Network PDU (ivi: 0, nid: 0x7E, ctl: 0, ttl: 5, seq: 0, src: 0001, dst: 0002, transportPdu: 0x176DB9C6FCD91EAA, netMic: 0x3B5A45FF) encrypted using Primary Network Key (index: 0)
[Network] Secure Network Beacon (Network ID: 0D75DE242995A748, IV Index: 0, Key Refresh Flag: false, IV Update active: false) receieved (decrypted using key: Primary Network Key (index: 0))
[Proxy] New Proxy connected
[Proxy] Sending SetFilterType(filterType: Whitelist) from: 0001 to: 0000
[Network] Sending Control Message (opCode: 0x00, data: 0x00)
[Network] Sending Network PDU (ivi: 0, nid: 0x7E, ctl: 1, ttl: 0, seq: 1, src: 0001, dst: 0000, transportPdu: 0xD3E7, netMic: 0x1462BB0AED420092) encrypted using Primary Network Key (index: 0)
[Network] Network PDU (ivi: 0, nid: 0x7E, ctl: 0, ttl: 4, seq: 0, src: 0002, dst: 0001, transportPdu: 0x0FE0C794307982E340A3842150DDAAB5, netMic: 0x61344544) received
[LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 0, segO: 0, segN: 3, data: 0x446798BC22B1C0B051796828) receieved (decrypted using key: Primary Network Key (index: 0))
[Network] Network PDU (ivi: 0, nid: 0x7E, ctl: 0, ttl: 4, seq: 1, src: 0002, dst: 0001, transportPdu: 0xD3CE3952A7B449E38600A73A23588345, netMic: 0x7ADEE945) received
[LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 0, segO: 1, segN: 3, data: 0x9821927FD59AFA7A34476C62) receieved (decrypted using key: Primary Network Key (index: 0))
[LowerTransport] Sending ACK (seqZero: 0, blockAck: 0x00000003)
[Network] Sending Network PDU (ivi: 0, nid: 0x7E, ctl: 1, ttl: 5, seq: 2, src: 0001, dst: 0002, transportPdu: 0xC60056CEEC6F85, netMic: 0xB8C1EC1D19B4B02A) encrypted using Primary Network Key (index: 0)
[Network] Network PDU (ivi: 0, nid: 0x7E, ctl: 0, ttl: 4, seq: 2, src: 0002, dst: 0001, transportPdu: 0xA7BB7C30976AB97F2BE6118788AC0F4C, netMic: 0xECABD821) received
[LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 0, segO: 2, segN: 3, data: 0x7E7EAC8973B374DDFDA6DF00) receieved (decrypted using key: Primary Network Key (index: 0))
[LowerTransport] Sending ACK (seqZero: 0, blockAck: 0x00000007)
[Network] Sending Network PDU (ivi: 0, nid: 0x7E, ctl: 1, ttl: 5, seq: 3, src: 0001, dst: 0002, transportPdu: 0xA636944EB76690, netMic: 0x932FA109CBC004CA) encrypted using Primary Network Key (index: 0)
[Network] Network PDU (ivi: 0, nid: 0x7E, ctl: 0, ttl: 4, seq: 3, src: 0002, dst: 0001, transportPdu: 0x3061D60E7387820C35639DF7E872BBD6, netMic: 0x2379F826) received
[LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 0, segO: 3, segN: 3, data: 0x7309234905AE3D39CAB77295) receieved (decrypted using key: Primary Network Key (index: 0))
[LowerTransport] Access Message (akf: 0, szmic: 0, data: 0x446798BC22B1C0B0517968289821927FD59AFA7A34476C627E7EAC8973B374DDFDA6DF007309234905AE3D39CAB77295) received
[LowerTransport] Sending ACK (seqZero: 0, blockAck: 0x0000000F)
[UpperTransport] Upper Transport PDU (encrypted data: 0x446798BC22B1C0B0517968289821927FD59AFA7A34476C627E7EAC8973B374DDFDA6DF007309234905AE3D39, transMic: 0xCAB77295) received
[Access] Response Access PDU (opcode: 0x 2, parameters: 0x00590000000000280003000000020000000200000001000010000001000010000001000010000001000010) receieved (decrypted using key: AFB504-C148BA22199A's Device Key)
[Network] Sending Network PDU (ivi: 0, nid: 0x7E, ctl: 1, ttl: 5, seq: 4, src: 0001, dst: 0002, transportPdu: 0xD43DED05E9BB51, netMic: 0xCA6437755B7FF6B7) encrypted using Primary Network Key (index: 0)
[Network] Network PDU (ivi: 0, nid: 0x7E, ctl: 1, ttl: 0, seq: 4, src: 0002, dst: 0000, transportPdu: 0x6A396BB5, netMic: 0x7C9B882E38F2405E) received
[Network] Control Message (opCode: 0x03, data: 0x000000) receieved (decrypted using key: Primary Network Key (index: 0))
[Proxy] FilterStatus(filterType: Whitelist, listSize: 0) received from: 0002 to: 0000
[Proxy] Sending AddAddressesToFilter(addresses: Set([1, 65535])) from: 0001 to: 0000
[Network] Sending Control Message (opCode: 0x01, data: 0x0001FFFF)
[Network] Sending Network PDU (ivi: 0, nid: 0x7E, ctl: 1, ttl: 0, seq: 5, src: 0001, dst: 0000, transportPdu: 0x183A30CEED, netMic: 0x172914911BEE65C5) encrypted using Primary Network Key (index: 0)
[Network] Network PDU (ivi: 0, nid: 0x7E, ctl: 0, ttl: 4, seq: 9, src: 0002, dst: 0001, transportPdu: 0x3B66EF83993DCC1FB27CB7D2D5AA1918, netMic: 0x42EDD7D4) received
[LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 0, segO: 2, segN: 3, data: 0x7E7EAC8973B374DDFDA6DF00) receieved (decrypted using key: Primary Network Key (index: 0))
[LowerTransport] Message already acknowledged, sending ACK again
[LowerTransport] Sending ACK (seqZero: 0, blockAck: 0x0000000F)
[Network] Sending Network PDU (ivi: 0, nid: 0x7E, ctl: 1, ttl: 5, seq: 6, src: 0001, dst: 0002, transportPdu: 0xEA3C8B3F526F6B, netMic: 0x734C5A6F0B9E6BEA) encrypted using Primary Network Key (index: 0)
[Network] Network PDU (ivi: 0, nid: 0x7E, ctl: 0, ttl: 4, seq: 10, src: 0002, dst: 0001, transportPdu: 0xB79FF155F5BC24A762573C08EAAF24A4, netMic: 0x47FBC091) received
[LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 0, segO: 3, segN: 3, data: 0x7309234905AE3D39CAB77295) receieved (decrypted using key: Primary Network Key (index: 0))
[LowerTransport] Message already acknowledged, sending ACK again
[LowerTransport] Sending ACK (seqZero: 0, blockAck: 0x0000000F)
[Network] Sending Network PDU (ivi: 0, nid: 0x7E, ctl: 1, ttl: 5, seq: 7, src: 0001, dst: 0002, transportPdu: 0xA35A4472C0B3E6, netMic: 0x0F99DD159D6B8BE5) encrypted using Primary Network Key (index: 0)
[Network] Network PDU (ivi: 0, nid: 0x7E, ctl: 0, ttl: 4, seq: 11, src: 0002, dst: 0001, transportPdu: 0x896C06D17197E8BC5B745B1D2F963FAC, netMic: 0xCDFDB399) received
[... cut ...]
[Network] Sending Network PDU (ivi: 0, nid: 0x7E, ctl: 1, ttl: 5, seq: 8, src: 0001, dst: 0002, transportPdu: 0xA57CF9E56398C8, netMic: 0x80C11890969BA995) encrypted using Primary Network Key (index: 0)
[Network] Network PDU (ivi: 0, nid: 0x7E, ctl: 1, ttl: 0, seq: 14, src: 0002, dst: 0000, transportPdu: 0xFD4831D3, netMic: 0x73BE61816BA2211D) received
[Network] Control Message (opCode: 0x03, data: 0x000002) receieved (decrypted using key: Primary Network Key (index: 0))
[Proxy] FilterStatus(filterType: Whitelist, listSize: 2) received from: 0002 to: 0000
```

philips77, 2019-11-25:
Do you set the `localElements` in `MeshNetworkManager`? If can be set to an empty array if you don't need any, but has to be set so that the local model delegates are initialized correctly.

deverickasih, 2019-11-25:
O ok.. i didn't set it. Now its flawless. Thanks

closed 2019-11-25 by deverickasih