nordicsemi/IOS-nRF-Mesh-Library #218: Packet drop with proxy

ximplarandy, 2019-11-25:
**Describe the bug**
We are using the proxy to relay the information sending to group C002 from nodes to the mobile app. When the number of nodes increases to about 20, with intervals of about 1 per minute, some packet loss is observed. 

From the following log below, you can see that the message from node 007B has only part of the message received and never decoded.  This is a vendor model.

I would like to know if there are settings etc that I could change to allow more throughput?

**Expected behavior**
I would expect the packets to be received and decrypted.  The messages are only happening 1-3 seconds apart, which I would expect to be OK.

**Platform details:**
iPad with iOS 13

**Logs / Screenshots**
`Nov 25 14:14:05 Segmented Access Message (akf: 1, aid: 0x37, szmic: 0, seqZero: 1070, segO: 0, segN: 1, data: 0xFA7C52F2B02740420734842A, src: 00AB receieved (decrypted using key: Primary Network Key (index: 0))
Nov 25 14:14:05 Segmented Access Message (akf: 1, aid: 0x37, szmic: 0, seqZero: 1070, segO: 1, segN: 1, data: 0x52C7C17B04, src: 00AB receieved (decrypted using key: Primary Network Key (index: 0))
Nov 25 14:14:05 unknowMessage reveiced : (opcode: 00C25900, parameters: 72030068009400720005) from src: 00AB, des: C002
Nov 25 14:14:06 Segmented Access Message (akf: 1, aid: 0x37, szmic: 0, seqZero: 1030, segO: 0, segN: 1, data: 0xBCFA40B0C9BD4F70FF20F988, src: 0065 receieved (decrypted using key: Primary Network Key (index: 0))
Nov 25 14:14:06 Segmented Access Message (akf: 1, aid: 0x37, szmic: 0, seqZero: 1030, segO: 1, segN: 1, data: 0xA098412ED4, src: 0065 receieved (decrypted using key: Primary Network Key (index: 0))
Nov 25 14:14:06 unknowMessage reveiced : (opcode: 00C25900, parameters: 67030068009400720000) from src: 0065, des: C002
Nov 25 14:14:07 Discarding packet (seqAuth: 33798, expected > 33798), seq 33798, source 0065
Nov 25 14:14:07 Discarding packet (seqAuth: 33798, expected > 33798), seq 33798, source 0065
Nov 25 14:14:08 Discarding packet (seqAuth: 33798, expected > 33798), seq 33798, source 0065
Nov 25 14:14:08 Discarding packet (seqAuth: 33798, expected > 33798), seq 33798, source 0065
Nov 25 14:14:09 Segmented Access Message (akf: 1, aid: 0x37, szmic: 0, seqZero: 6190, segO: 0, segN: 1, data: 0x4F0FA437CC1021CAAE171304, src: 007B receieved (decrypted using key: Primary Network Key (index: 0))
Nov 25 14:14:10 Segmented Access Message (akf: 1, aid: 0x37, szmic: 0, seqZero: 1030, segO: 1, segN: 1, data: 0x3EDA68E700, src: 00A3 receieved (decrypted using key: Primary Network Key (index: 0))
Nov 25 14:14:10 Segmented Access Message (akf: 1, aid: 0x37, szmic: 0, seqZero: 1030, segO: 1, segN: 1, data: 0x3EDA68E700, src: 00A3 receieved (decrypted using key: Primary Network Key (index: 0))
Nov 25 14:14:11 Segmented Access Message (akf: 1, aid: 0x37, szmic: 0, seqZero: 1030, segO: 1, segN: 1, data: 0x3EDA68E700, src: 00A3 receieved (decrypted using key: Primary Network Key (index: 0))
Nov 25 14:14:11 Segmented Access Message (akf: 1, aid: 0x37, szmic: 0, seqZero: 1030, segO: 0, segN: 1, data: 0xD9F369DEDCE10A6424FB2E19, src: 00A3 receieved (decrypted using key: Primary Network Key (index: 0))
Nov 25 14:14:11 unknowMessage reveiced : (opcode: 00C25900, parameters: 67030010006800940000) from src: 00A3, des: C002`

ximplarandy, 2019-11-25:
Actually, may I know what is the max length to relay without fragmentation?  Thanks

philips77, 2019-11-25:
> Actually, may I know what is the max length to relay without fragmentation? Thanks

It's 11 bytes. Have a look at this method:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/10acde1cadc78f6ca08082cea8002578495d6491/nRFMeshProvision/Classes/Layers/Access%20Layer/AccessPdu.swift#L57-L62

and the spec:
[image]
[image]

Regarding your first question and the log. There are some settings you may set in your network that will make the messages more reliable. You may set "relay count and interval" for all relays and "network transmit" for all nodes from where messages originate. Find more info about those parameters in the Bluetooth Mesh Profile specification 1.0., chapters 4.2.20 and 4.2.19. Those parameters can be modified using the sample app, after opening a primary element and Configuration Server model on each node.

You may also send your messages as segmented. Then the lower transport layer will try to make sure all parts are delivered and confirmed, but only when they are sent to a Unicast Address. In your case you are sending to a Group address, so the ACK don't apply. In that case you may send the message as Acknowledged Vendor Message, but this will generate more trafic, as the phone will have to ack all received messages. And sometimes it is not possible to alter the specification of your messages and making them acknowledged. Besides, it is not recommended to send acknowledged messages to group addresses as this will generate more trafic.

philips77, 2019-11-25:
And, the last thing. I see you are receiving your messages as `UnknownMessage`. I would recommend using a typed message instead. Have a look how the Simple OnOff vendor model support has been added in #194.

ximplarandy, 2019-11-25:
Thank you very much.  The vendor model is definitely something I will implement when things are reliable and not changing daily, understand it will make the data more structure and easier to maintain. 

On the message length part, strange that my data length should be 10 bytes already, I wonder why it is fragmented. 

My structure for feeding into the message_create function is like this:

`typedef struct __attribute((packed))
{
    uint8_t tid;  
    uint8_t operation;
    uint8_t td;
    uint16_t param1;
    uint16_t param2;
    uint16_t param3;
    uint8_t issues;
}`

Or the 10 bytes includes some other overhead?

philips77, 2019-11-25:
10 byte long message may be sent as a single unsegmented message or 2 segmented messages. It's up to the sender to decide. Looks like your nodes send the message as segmented, and as it has 10 bytes, not 8, must be sent in 2 segments. I would highly recommend changing to non-segmented messages in that case, as you are not relying on ACK either way (messages are sent to group addresses), so all the benefit of having short segmented message is gone.

ximplarandy, 2019-11-25:
Thank you again for the fast response!!

closed 2019-11-25 by philips77

ximplarandy, 2019-11-25:
Just confirm one more thing.  If the message is sent from node A, to group C002.  Node B is a proxy subscribe and whitelist C002.  If the message is set to be unfragmented, then the mobile app will read from node B with the same setting via the proxy?  Or the proxy needs some setting?

Sorry for the many questions.

philips77, 2019-11-26:
If you add C002 to proxy white list filter, you'll receive a callback in Mesh Network Delegate each time it was received. Until you'll implement the model delegate and the message type, it will be an unknown message. When implemented, you'll receive the type specified in the model delegate for this opcode, and you'll also receive this message. You'll have an option also to subscribe you'll local model to this group address, after that you'll be also getting this message in your model delegate. To subscribe local model to c002, you have to send message config model subscribe to the local model.