nordicsemi/IOS-nRF-Mesh-Library #220: vendor model custom message

chy1025507207, 2019-11-25:
Can I add a vendor model to send custom messages??

philips77, 2019-11-25:
There is already just possibility. Have a look at the documentation: https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/master/Documentation/GETTING_STARTED.md

After instantiating the `MeshNetworkManager` you have to set up your local elements with local models. To send/receive vendor messages you need to have at least one vendor model. Here's how the sample app is doing this: https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/10acde1cadc78f6ca08082cea8002578495d6491/Example/nRFMeshProvision/AppDelegate.swift#L107

Here you have the supported messages and the model delegate: https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/tree/master/Example/nRFMeshProvision/Mesh%20Network/Simple%20OnOff

And here's how it sends one of those messages when a button in a cell is clicked: https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/master/Example/nRFMeshProvision/View%20Controllers/Control/Model/SimpleOnOffClientCell.swift

closed 2019-11-25 by philips77

chy1025507207, 2019-11-25:
You are right, thanks