nordicsemi/IOS-nRF-Mesh-Library #225: Can not set publication address to a unicast addr?

YoYo0304, 2019-11-29:
There are a sensor model server  and a sensor client in our bluetooth mesh net. But i can not set the 
publication addr of the sensor model server to the sensor modle client's addr.   
[image]

YoYo0304, 2019-11-29:
but when using the android app,  the same configuration is ok

philips77, 2019-11-29:
The element will appear on the list of it has a compatible model which is bound to selected application key. Make sure that the server is bound to this key before setting publication.

YoYo0304, 2019-12-02:
yeah. The server has been bound to the app key. 
[image]

philips77, 2019-12-02:
Och.. then it's a bug in the library. I assumed that all "matching" models (server-client pairs) have IDs differing by 1 (where even is the server and odd is the client), but in this case it's different:
[image]
You will have to modify the source code of the app to make it work. I'll fix it in the upcoming bug-fixing release.

YoYo0304, 2019-12-03:
OK. thanks

closed 2020-04-14 by philips77