nordicsemi/IOS-nRF-Mesh-Library #226: Not able to set proxy filter after changing connection to another proxy

heywil, 2019-12-02:
HI, 

I was trying to use the Manual connection management function, everything works fine when I connect to the first proxy, but when I try to switch to connect another proxy, the proxy white list could not be set after the connection was done. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the "Proxy" tab
2. Turn off "Automatic Connection"
3. Tap the proxy cell
4. Select a proxy and connect to it
5. Tap the proxy cell again
6. Select another proxy and connect to it
7. Press the "+" button to add an address
8. Select an address and press "Done"
9. Dialog "Adding addresses..." pop up, but nothing happens afterwards.

**Platform details:**
 - Device: iPad Air 2
 - OS: 13.1
 - Library Version (or Commit): 2.2.0

**Logs / Screenshots**
2019-12-02 12:40:24.261874+0800 nRFMeshProvision_Example[14313:1425731] [Proxy] Sending SetFilterType(filterType: Whitelist) from: 0001 to: 0000
proxy send message
2019-12-02 12:40:24.262425+0800 nRFMeshProvision_Example[14313:1425731] [Network] Sending Control Message (opCode: 0x00, data: 0x00)
2019-12-02 12:40:24.266570+0800 nRFMeshProvision_Example[14313:1425731] [Network] Sending Network PDU (ivi: 0, nid: 0x1F, ctl: 1, ttl: 0, seq: 14, src: 0001, dst: 0000, transportPdu: 0x0FB8, netMic: 0xCC6CF24C4D5CDB3B) encrypted using Keycard Default Key (index: 0)
2019-12-02 12:40:24.267652+0800 nRFMeshProvision_Example[14313:1425731] [Bearer] -> 0x021FBEE6EC564EF126190FB8CC6CF24C4D5CDB3B
2019-12-02 12:40:24.268089+0800 nRFMeshProvision_Example[14313:1425731] [CoreBluetooth] API MISUSE: <CBPeripheral: 0x2806a06e0, identifier = 1BA348D0-C772-8F35-C779-AAF6CE808775, name = (Node_Name), state = disconnecting> can only accept commands while in the connected state

philips77, 2019-12-02:
Thanks for the report. I'll look into it. For now make sure you disconnect from the first proxy before switching to the next one (reset it?), this should work.

philips77, 2019-12-17:
I tried to replicate your issue, but I was able to add the address in point 8 and 9.
Are you still able to reproduce this issue, or was it a one time thing?

heywil, 2019-12-17:
I was able to reproduce the issue each time I tried. 

I did made some changes to the program, as the nodes was not provisioned on the iPad i was using, I change the network key and application key to a fix one. 

One thing I noticed, when I connect to the first proxy, 3 address were added to the whitelist automatically (Elemtent 1 - 0x0001, Secondary Element - 0x0002, All Nodes - 0FFFF), but when I connect to the second proxy, non of them shows up. The list remain empty. 

The program seems to be still using the first proxy to send the SetFilterType message when it connected to the second proxy.

philips77, 2019-12-17:
Can you please provide some logs?

philips77, 2019-12-17:
When in step 4. you connect to the second Proxy, the sample app will first connect to it, ensuring that it supports Mesh Proxy Service, and if it does, will call this method:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/45e12224e8b5f2d45659e99dddda966b1ba17789/Example/nRFMeshProvision/Mesh%20Network/NetworkConnection.swift#L144-L160
The `NetworkConnection` class is always using the first proxy on the list, but, as you may see, it appends the new proxy to the end of the list, first closing the connection with others. The `close()` method should finish very quickly, causing the proxy removal from the list, so only the new one remains and is ready to be used.
In your case, it seems like you don't get `peripheral(didDisconnect...)` callback, so the first proxy is sill the wrong one. Try changing `append`  to `insert(proxy, at: 0)` and check if it helps. Do you keep the connections to proxies somewhere else as well? Even if you do, and the physical connection is open, you should still get this callback, as the client gets disconnected.

philips77, 2020-04-14:
Any progress with this issue?

closed 2020-04-23 by philips77