nordicsemi/IOS-nRF-Mesh-Library #227: Remaining time 0x3F should be displayed as Unknown

philips77, 2019-12-02:
**Describe the bug**
In the sample app, the Remaining time 0x3F in `GenericLevelStatus` should be displayed as *Unknown time*, not 6300 ms.

**Platform details:**
 - Library Version (or Commit): 2.1.0

The bug was originally reported here: https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/issues/282

vikrant8051, 2019-12-02:
Not only in case of "GenericLevelStatus" but whenever Server Node send Remaining time as 0x3F
it consider as Unknown.

philips77, 2019-12-17:
The issue you mentioned seems to be working fine on nRF Mesh for iOS.
The sample app supports only Generic Level and Generic OnOff models that use `GenericLevelStatusMessage`. In both view controllers responsible for displaying the returned status the 0x3F is taken under consideration:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/10acde1cadc78f6ca08082cea8002578495d6491/Example/nRFMeshProvision/View%20Controllers/Network/Configuration/Model/GenericLevelViewCell.swift#L143-L147
and
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/10acde1cadc78f6ca08082cea8002578495d6491/Example/nRFMeshProvision/View%20Controllers/Network/Configuration/Model/GenericOnOffViewCell.swift#L108-L112
Please, attach a screenshot showing the value being displayed incorrectly.

Screenshot from Generic Level configuration screen from version 2.2.0:
[image]

closed 2019-12-17 by philips77