nordicsemi/IOS-nRF-Mesh-Library #229: Not getting a on/off response from node elements after re-importing mesh network.

deverickasih, 2019-12-03:
Hi,

**Problem**
I have an app. When network is not created i create a new network. If network exists, i load the existing network. The first time i add a new device, i update the mesh network to cloud. Then without reimporting the mesh network, using the app i can interact with the device (turn lights on/off). When i exit the app and re-enter the app, i have to get data from cloud and re-import the mesh network into the app. After re-importing the network, i can't seem to interact with the device i previously added.

**Logs**
Every time i toggle on/off these logs are sent
```
Jan : Element Address tapped = 0003
Jan : Result for toggle On of 0003 = MessageHandle(opCode: 33282, source: 1,
destination: 3, manager: Optional(nRFMeshProvision.MeshNetworkManager))
[Model] Sending GenericOnOffSet(tid: Optional(243), isOn: true, transitionTime: nil, delay: nil) from: Element 0 (0x0001), to: 0003
[Access] Sending Access PDU (opcode: 0x8202, parameters: 0x01F3)
[UpperTransport] Sending Upper Transport PDU (encrypted data: 0xE8E38732, transMic: 0x85125A74) encrypted using key: 1stKey (index: 0)
[LowerTransport] Sending Access Message (akf: 1, aid: 0x32, szmic: 0, data: 0xE8E3873285125A74)
[Network] Sending Network PDU (ivi: 0, nid: 0x5E, ctl: 0, ttl: 5, seq: 1110, src: 0001, dst: 0003, transportPdu: 0x1371F27F4BC4C9EFED, netMic: 0x25C7ED37) encrypted using Primary Network Key (index: 0)
[Access] Resending Access PDU (opcode: 0x8202, parameters: 0x01F3)
[UpperTransport] Sending Upper Transport PDU (encrypted data: 0x2F09A6B3, transMic: 0x66DF5B1F) encrypted using key: 1stKey (index: 0)
[LowerTransport] Sending Access Message (akf: 1, aid: 0x32, szmic: 0, data: 0x2F09A6B366DF5B1F)
[Network] Sending Network PDU (ivi: 0, nid: 0x5E, ctl: 0, ttl: 5, seq: 1111, src: 0001, dst: 0003, transportPdu: 0xDCBE35492BC73A70C1, netMic: 0xF80A465E) encrypted using Primary Network Key (index: 0)
[Access] Resending Access PDU (opcode: 0x8202, parameters: 0x01F3)
[UpperTransport] Sending Upper Transport PDU (encrypted data: 0x8EE4F5C0, transMic: 0xE40695E2) encrypted using key: 1stKey (index: 0)
[LowerTransport] Sending Access Message (akf: 1, aid: 0x32, szmic: 0, data: 0x8EE4F5C0E40695E2)
[Network] Sending Network PDU (ivi: 0, nid: 0x5E, ctl: 0, ttl: 5, seq: 1112, src: 0001, dst: 0003, transportPdu: 0xB723F54B30D7B728B9, netMic: 0xCAF26A52) encrypted using Primary Network Key (index: 0)
[Access] Resending Access PDU (opcode: 0x8202, parameters: 0x01F3)
[UpperTransport] Sending Upper Transport PDU (encrypted data: 0xDF400A7A, transMic: 0xCE4A14E9) encrypted using key: 1stKey (index: 0)
[LowerTransport] Sending Access Message (akf: 1, aid: 0x32, szmic: 0, data: 0xDF400A7ACE4A14E9)
[Network] Sending Network PDU (ivi: 0, nid: 0x5E, ctl: 0, ttl: 5, seq: 1113, src: 0001, dst: 0003, transportPdu: 0xFE60B1F0A876F8EC73, netMic: 0x9B35F3AB) encrypted using Primary Network Key (index: 0)
[Access] Response to Access PDU (opcode: 0x8202, parameters: 0x01F3) not received (timeout)
[Model] GenericOnOffSet(tid: Optional(243), isOn: true, transitionTime: nil, delay: nil) sent from: 0001, to: 0003 timed out
[Access] Cancelling messages with op code: 33282, sent from: 0001 to: 0003
```
```
Jan : Element Address tapped = 0003
Jan : Result for toggle Off of 0003 = MessageHandle(opCode: 33282, source: 1,
destination: 3, manager: Optional(nRFMeshProvision.MeshNetworkManager))
[Model] Sending GenericOnOffSet(tid: Optional(244), isOn: false, transitionTime: nil, delay: nil) from: Element 0 (0x0001), to: 0003
[Access] Sending Access PDU (opcode: 0x8202, parameters: 0x00F4)
[UpperTransport] Sending Upper Transport PDU (encrypted data: 0x9BADB8F1, transMic: 0xA20B82E8) encrypted using key: 1stKey (index: 0)
[LowerTransport] Sending Access Message (akf: 1, aid: 0x32, szmic: 0, data: 0x9BADB8F1A20B82E8)
[Network] Sending Network PDU (ivi: 0, nid: 0x5E, ctl: 0, ttl: 5, seq: 1114, src: 0001, dst: 0003, transportPdu: 0x65EDEFD2E59010098D, netMic: 0x73E7A612) encrypted using Primary Network Key (index: 0)
[Access] Resending Access PDU (opcode: 0x8202, parameters: 0x00F4)
[UpperTransport] Sending Upper Transport PDU (encrypted data: 0x27ECB652, transMic: 0xAB371498) encrypted using key: 1stKey (index: 0)
[LowerTransport] Sending Access Message (akf: 1, aid: 0x32, szmic: 0, data: 0x27ECB652AB371498)
[Network] Sending Network PDU (ivi: 0, nid: 0x5E, ctl: 0, ttl: 5, seq: 1115, src: 0001, dst: 0003, transportPdu: 0xD9C04FEFAB72AE1A48, netMic: 0x92CDC527) encrypted using Primary Network Key (index: 0)
[Access] Resending Access PDU (opcode: 0x8202, parameters: 0x00F4)
[UpperTransport] Sending Upper Transport PDU (encrypted data: 0x004A52F4, transMic: 0xC724EF01) encrypted using key: 1stKey (index: 0)
[LowerTransport] Sending Access Message (akf: 1, aid: 0x32, szmic: 0, data: 0x004A52F4C724EF01)
[Network] Sending Network PDU (ivi: 0, nid: 0x5E, ctl: 0, ttl: 5, seq: 1116, src: 0001, dst: 0003, transportPdu: 0xD79372321E08BD6D9F, netMic: 0x63B93DD9) encrypted using Primary Network Key (index: 0)
[Access] Resending Access PDU (opcode: 0x8202, parameters: 0x00F4)
[UpperTransport] Sending Upper Transport PDU (encrypted data: 0xEE80FA0B, transMic: 0x4571C110) encrypted using key: 1stKey (index: 0)
[LowerTransport] Sending Access Message (akf: 1, aid: 0x32, szmic: 0, data: 0xEE80FA0B4571C110)
[Network] Sending Network PDU (ivi: 0, nid: 0x5E, ctl: 0, ttl: 5, seq: 1117, src: 0001, dst: 0003, transportPdu: 0xB297C1EA18E76B423E, netMic: 0xA7BFD3BA) encrypted using Primary Network Key (index: 0)
[Access] Response to Access PDU (opcode: 0x8202, parameters: 0x00F4) not received (timeout)
[Model] GenericOnOffSet(tid: Optional(244), isOn: false, transitionTime: nil, delay: nil) sent from: 0001, to: 0003 timed out
[Access] Cancelling messages with op code: 33282, sent from: 0001 to: 0003
```

**Possible Mistake**
In my code :
```
let genericOnOffModel = 0
let lightButton = CustomLightButton()

let elementAddress : Address = UInt16("0003", radix: 16)
let parentAddress : Address = UInt16("0002", radix: 16)

lightButton.model = MeshNetworkManager.instance.meshNetwork?.node(withAddress: parentAddress)!.element(withAddress: elementAddress)!.model[genericOnOffModel]
```

**Question**
I believe i made a mistake in setting the elementAddress and parentAddress. What should be the right way to set them?
Or is the problem something else?

philips77, 2019-12-03:
Have a look at https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/issues/206#issuecomment-555231853 and https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/issues/206#issuecomment-557514116 for possible reasons why your packets don't reach the destination. 
It is difficult to say why the packets are being discarded and where in your situation. Are you sure you have the address 0x0001 added to the proxy filter on the connected proxy node? Do you have logs from your target node? Does it receive packets? Are you directly connected to it using GATT Bearer or is there another proxy node?

deverickasih, 2019-12-05:
I don't remember what i did but now there is response from the nodes that i have previously added to mesh network even after re-importing mesh network to my project.

I believe that i changed the provisioner ttl to 5 & set the network retransmission limit to 5.

I have got 2 other questions, i hope you don't mind me putting them in this same thread :

(1) If i set an element-model to publish it's state to all nodes every 10 seconds, is there a time limit for it to stay like this or does it publish it's state to all nodes until the day it is set otherwise? I ask this because yesterday i set an element-model to publish to all nodes every 10 seconds. But today when i open my app, i can interact with the nodes through the app and through their physical buttons and everything works fine, but the publishing-every-10-seconds settings that i set to that element-model no longer publishes its state every 10 seconds.. I am not quite sure what's the difference between today and yesterday? Is there a time limit of some sort?

(2) How do i make sure that i only connect to one node at any given time? I ask this because i currently have 2 nodes in the mesh network and everytime i import the meshnetwork, i automatically connect to both nodes. I use nrf connect app to verify this issue. But when i turn off my bluetooth in my iPhone, then only do both nodes appear connectable in the nrf connect app.

philips77, 2019-12-05:
1. There should be no timeout. The device should still advertise. Are you sure the device isn't publishing, or just you don't get them in the app?

2. The sample app may connect to more than one node, but you'd have to modify Network connection class. By default it's just 1. At least I hope so.

deverickasih, 2019-12-06:
1. We recently tried again today. My firmware programmer told me that in the sample app, it does work, that is because it was left on throughout the night. But when we turn off the board and turn it back on, the board itself does not publish its state every 10 seconds. According to him, every time we do a power cycle on the board, its publishing settings get a reset. So we have to set it manually again after a power cycle before it can start publishing its state every 10 seconds. Personally, i'd like to know if this is something we have to custom from our part or is this a bug from the board itself? Is this power cycle thing an intended result from Nordic?

2. After setting the maxConnections to 1 in the Network Connection class, i was capable of ensuring only one connection at any given time. Thanks

philips77, 2019-12-06:
Regarding 1, I don't know. You need to ask on DevZone, our support team will know things like that.
Closing the issue now.

closed 2019-12-06 by philips77