nordicsemi/IOS-nRF-Mesh-Library #232: Grouping Logs Analysis

deverickasih, 2019-12-12:
I've got two logs, one from my project and another from the sample app. My Problem is that I bind only 1 element to a group. And when I turn the group on, it lights all the elements under the node that element is in. This only happens in my own project not in the sample app but the logs I receive from both my project and the sample app is quite the same and this leaves me confused.

My app uses nrfMesh 2.0.0 and I also downloaded the sample project with tag : 2.0.0

My own project logs : 
- 0x0009 is the address I added to group
- 0x0007 is the node address of the element 0x0009
- group address is C000 (49152)
```
[FoundationModel] Sending ConfigModelSubscriptionAdd(address: 49152, elementAddress: 9, modelIdentifier: 4096, companyIdentifier: nil) to: 0007

Jan : subscribe with result = MessageHandle(opCode: 32795, source: 1, destination: 7, manager: Optional(nRFMeshProvision.MeshNetworkManager))

[Access] Sending Access PDU (opcode: 0x801B, parameters: 0x090000C00010)
[UpperTransport] Sending Upper Transport PDU (encrypted data: 0x934643D91C86FFFE, transMic: 0x2FEB55C2) encrypted using key: qwertyuio's Device Key
[LowerTransport] Sending Access Message (akf: 0, szmic: 0, data: 0x934643D91C86FFFE2FEB55C2)
[Network] Sending Network PDU (ivi: 0, nid: 0x0F, ctl: 0, ttl: 5, seq: 963, src: 0001, dst: 0007, transportPdu: 0xE7E4C177D1885071F4F11EDD03, netMic: 0xEA2EBF77) encrypted using Primary Network Key (index: 0)
[Network] Network PDU (ivi: 0, nid: 0x0F, ctl: 0, ttl: 4, seq: 49154, src: 0007, dst: 0001, transportPdu: 0x6509579C4D3CEBACB8CAA757E474, netMic: 0x5DB21383) received
[LowerTransport] Access Message (akf: 0, szmic: 0, data: 0xDF64D19632C01760D5A3A64251) receieved (decrypted using key: Primary Network Key (index: 0))
[UpperTransport] Upper Transport PDU (encrypted data: 0xDF64D19632C01760D5, transMic: 0xA3A64251) received
[Access] Response Access PDU (opcode: 0x801F, parameters: 0x00090000C00010) receieved (decrypted using key: qwertyuio's Device Key)
[FoundationModel] ConfigModelSubscriptionStatus(status: Success, address: 49152, elementAddress: 9, modelIdentifier: 4096, companyIdentifier: nil) received from: 0007

Jan : didReceiveMessage = ConfigModelSubscriptionStatus(status: Success, address: 49152, elementAddress: 9, modelIdentifier: 4096, companyIdentifier: nil), from = 0007, to = 0001, withParameters = Optional("00090000C00010")
Jan : Yes received
Jan : End Subscription of models
Jan : Send message to group = MessageHandle(opCode: 33283, source: 1, destination: 49152, manager: Optional(nRFMeshProvision.MeshNetworkManager))

[Model] Sending GenericOnOffSetUnacknowledged(tid: Optional(26), isOn: true, transitionTime: nil, delay: nil) from: Element 0 (0x0001), to: C000
[Access] Sending Access PDU (opcode: 0x8203, parameters: 0x011A)
[UpperTransport] Sending Upper Transport PDU (encrypted data: 0xA2C2B7E7, transMic: 0xDD455E6E) encrypted using key: 1stKey (index: 0)
[LowerTransport] Sending Access Message (akf: 1, aid: 0x0A, szmic: 0, data: 0xA2C2B7E7DD455E6E)
[Network] Sending Network PDU (ivi: 0, nid: 0x0F, ctl: 0, ttl: 5, seq: 964, src: 0001, dst: C000, transportPdu: 0x60C0E8273D589C4C09, netMic: 0x266E4C07) encrypted using Primary Network Key (index: 0)
[Network] Network PDU (ivi: 0, nid: 0x0F, ctl: 0, ttl: 5, seq: 964, src: 0001, dst: C000, transportPdu: 0x60C0E8273D589C4C09, netMic: 0x266E4C07) received
[LowerTransport] Access Message (akf: 1, aid: 0x0A, szmic: 0, data: 0xA2C2B7E7DD455E6E) receieved (decrypted using key: Primary Network Key (index: 0))
[UpperTransport] Upper Transport PDU (encrypted data: 0xA2C2B7E7, transMic: 0xDD455E6E) received
[Access] Access PDU (opcode: 0x8203, parameters: 0x011A) receieved (decrypted using key: 1stKey (index: 0))

Jan : didReceiveMessage = UnknownMessage(opCode: 0x008203, parameters: 011A), from = 0001, to = C000, withParameters = Optional("011A")

[Network] Network PDU (ivi: 0, nid: 0x0F, ctl: 0, ttl: 5, seq: 49155, src: 0008, dst: FFFF, transportPdu: 0xE7C9B54ED9A31957, netMic: 0x804CBEBF) received
[LowerTransport] Access Message (akf: 1, aid: 0x0A, szmic: 0, data: 0x82F54DD0AD469F) receieved (decrypted using key: Primary Network Key (index: 0))
[UpperTransport] Upper Transport PDU (encrypted data: 0x82F54D, transMic: 0xD0AD469F) received
[Access] Access PDU (opcode: 0x8204, parameters: 0x01) receieved (decrypted using key: 1stKey (index: 0))

Jan : didReceiveMessage = UnknownMessage(opCode: 0x008204, parameters: 01), from = 0008, to = FFFF, withParameters = Optional("01")

[Network] Network PDU (ivi: 0, nid: 0x0F, ctl: 0, ttl: 5, seq: 49156, src: 0009, dst: FFFF, transportPdu: 0xAF7ABA9E5948E148, netMic: 0x8F14F470) received
[LowerTransport] Access Message (akf: 1, aid: 0x0A, szmic: 0, data: 0x5B9DE54630467F) receieved (decrypted using key: Primary Network Key (index: 0))
[UpperTransport] Upper Transport PDU (encrypted data: 0x5B9DE5, transMic: 0x4630467F) received
[Access] Access PDU (opcode: 0x8204, parameters: 0x01) receieved (decrypted using key: 1stKey (index: 0))

Jan : didReceiveMessage = UnknownMessage(opCode: 0x008204, parameters: 01), from = 0009, to = FFFF, withParameters = Optional("01")
```

The logs from the sample app : 
- 0x0006 is the element address I added to group
- 0x0003 is the node address of the element 0x0006
- group address is C000 (49152)
```
[FoundationModel] Sending ConfigModelSubscriptionAdd(address: 49152, elementAddress: 6, modelIdentifier: 4096, companyIdentifier: nil) to: 0003
[Access] Sending Access PDU (opcode: 0x801B, parameters: 0x060000C00010)
[UpperTransport] Sending Upper Transport PDU (encrypted data: 0xD8AEAC3D893E825D, transMic: 0xB0A73866) encrypted using key: AFB503-EEFA01A77116's Device Key
[LowerTransport] Sending Access Message (akf: 0, szmic: 0, data: 0xD8AEAC3D893E825DB0A73866)
[Network] Sending Network PDU (ivi: 0, nid: 0x67, ctl: 0, ttl: 5, seq: 66, src: 0001, dst: 0003, transportPdu: 0x5E66731D5833A37D2D12F101BE, netMic: 0x00C3F62A) encrypted using Primary Network Key (index: 0)
[Bearer] -> 0x0067468F5D5515A91B525E66731D5833A37D2D12F101BE00C3F62A
[Bearer] <- 0x006724CB37415ECECA847A42D95E1B2CBFF81E31187357E2FC1C7AF0
[Network] Network PDU (ivi: 0, nid: 0x67, ctl: 0, ttl: 4, seq: 16439, src: 0003, dst: 0001, transportPdu: 0x7A42D95E1B2CBFF81E31187357E2, netMic: 0xFC1C7AF0) received
[LowerTransport] Access Message (akf: 0, szmic: 0, data: 0x66CC0FCED514529701E8B8F183) receieved (decrypted using key: Primary Network Key (index: 0))
[UpperTransport] Upper Transport PDU (encrypted data: 0x66CC0FCED514529701, transMic: 0xE8B8F183) received
[... 11 lines cut ...]
[Access] Access PDU (opcode: 0x8203, parameters: 0x018B0A14) receieved (decrypted using key: App Key 1 (index: 0))

[Model] (tid: Optional(139), isOn: true, transitionTime: Optional(1 sec), delay: Optional(20)) received from: 0001, to: C000
[Bearer] -> 0x0067B97525C8165E66CF8BF3FA381F0B2AE98945A92AF18C60
[Bearer] <- 0x0067D007321A936A0069CC6BD5BF1D4B0BC1E211B05468AE

[Network] Network PDU (ivi: 0, nid: 0x67, ctl: 0, ttl: 4, seq: 16440, src: 0006, dst: FFFF, transportPdu: 0xCC6BD5BF1D4B0BC1E211, netMic: 0xB05468AE) received
[LowerTransport] Access Message (akf: 1, aid: 0x3F, szmic: 0, data: 0x1F49B94677E78A7609) receieved (decrypted using key: Primary Network Key (index: 0))
[UpperTransport] Upper Transport PDU (encrypted data: 0x1F49B94677, transMic: 0xE78A7609) received
[Access] Access PDU (opcode: 0x8204, parameters: 0x01010A) receieved (decrypted using key: App Key 1 (index: 0))

[Model] GenericOnOffStatus(isOn: true, targetState: Optional(true), remainingTime: Optional(1 sec)) received from: 0006, to: FFFF
[Bearer] <- 0x0067C073B1899A3A3352A41B5404D30B87D0E18B1DD2

[Network] Network PDU (ivi: 0, nid: 0x67, ctl: 0, ttl: 4, seq: 16441, src: 0006, dst: FFFF, transportPdu: 0xA41B5404D30B87D0, netMic: 0xE18B1DD2) received
[LowerTransport] Access Message (akf: 1, aid: 0x3F, szmic: 0, data: 0xF43B0408E4E1AD) receieved (decrypted using key: Primary Network Key (index: 0))
[UpperTransport] Upper Transport PDU (encrypted data: 0xF43B04, transMic: 0x08E4E1AD) received
[Access] Access PDU (opcode: 0x8204, parameters: 0x01) receieved (decrypted using key: App Key 1 (index: 0))

[Model] GenericOnOffStatus(isOn: true, targetState: nil, remainingTime: nil) received from: 0006, to: FFFF
```

I am missing the [Bearer] and [Model] loggings from my own project but overall I find both these logs similar. Another thing to note is that I set publish to all nodes for all elements in both the sample app and my own project.

The strange part is that in my project I get 0x0008 and 0x0009 turned on when the group only consists of 0x0009 - only have one element. But also in the sample project I get GenericOnOffStatus called twice from 0x0006 to 0xFFFF

philips77, 2019-12-12:
You either must be doing something wrong before, or the nodes you test with your app have bugs. Try setting the same nodes that with sample app, if they're different.
Regarding the double publication, one is sent immediately, you can see the remaining time is 1s, the second when the state finally settles. As you see, in that case the default transition time is 1s.

deverickasih, 2019-12-17:
Recently I've tried again, and then I receive 'Insufficient Resources' when I try to subscribe a group to a model.

```
[FoundationModel] Sending ConfigModelSubscriptionAdd(address: 49152, elementAddress: 17, modelIdentifier: 4096, companyIdentifier: nil) to: 000E
[Access] Sending Access PDU (opcode: 0x801B, parameters: 0x110000C00010)
[UpperTransport] Sending Upper Transport PDU (encrypted data: 0x3F32C77D965D1A46, transMic: 0xD9FCE310) encrypted using key: qasse's Device Key
[LowerTransport] Sending Access Message (akf: 0, szmic: 0, data: 0x3F32C77D965D1A46D9FCE310)
[Network] Sending Network PDU (ivi: 0, nid: 0x0F, ctl: 0, ttl: 5, seq: 44, src: 0001, dst: 000E, transportPdu: 0xC3261FAF936A64BE076D646E2E, netMic: 0xA9E19730) encrypted using Primary Network Key (index: 0)
[Bearer] -> 0x000F0A8948C134D53FBBC3261FAF936A64BE076D646E2EA9E19730
[Bearer] <- 0x000F2482D09773F2BCB6B55E2557F0A3C6727559D7B901EB6BE70104
[Network] Network PDU (ivi: 0, nid: 0x0F, ctl: 0, ttl: 4, seq: 16461, src: 000E, dst: 0001, transportPdu: 0xB55E2557F0A3C6727559D7B901EB, netMic: 0x6BE70104) received
[LowerTransport] Access Message (akf: 0, szmic: 0, data: 0x6D35D5DDABCC354594F1356432) receieved (decrypted using key: Primary Network Key (index: 0))
[UpperTransport] Upper Transport PDU (encrypted data: 0x6D35D5DDABCC354594, transMic: 0xF1356432) received
[Access] Response Access PDU (opcode: 0x801F, parameters: 0x05110000C00010) receieved (decrypted using key: qasse's Device Key)
[FoundationModel] ConfigModelSubscriptionStatus(status: Insufficient resources, address: 49152, elementAddress: 17, modelIdentifier: 4096, companyIdentifier: nil) received from: 000E

Jan : didReceiveMessage = ConfigModelSubscriptionStatus(status: Insufficient resources, address: 49152, elementAddress: 17, modelIdentifier: 4096, companyIdentifier: nil), from = 000E, to = 0001, withParameters = Optional("05110000C00010")
```
**Questions**
Could you explain what 'Insufficient Resources' means? And how should I counter this?

philips77, 2019-12-17:
The memory available on an embedded device is limited. Usually you have to decide, while worrying fw, how is going to be used, and limit number of features ad you'd run out of memory. Often it is configured to support only one app key, only few groups, etc. You need to modify the settings in your software to match your needs and ensure that all will work at the same time.
This error is thrown from the node after such limit is reached.

philips77, 2019-12-17:
Try pull to refresh to get all current subscriptions. Remove some before adding new one.

philips77, 2021-02-05:
Hi, did you manager to fix your issue? Can it be closed?

closed 2021-05-21 by philips77