nordicsemi/IOS-nRF-Mesh-Library #234: Publish timeout

fiveagle, 2019-12-12:
Hi,

**Describe the bug**
Looks like that having more than 1 device provisioned and with a open bearer (Using NetworkConnection form the sample) makes the config messages like publish, bind and subscribe to fail in timeout. And sending a custom StaticVendorMessage fails too like a timeout but doesn't triggers a delegate call for didFail.
I'm doing like so:
`self.meshManager.send(message, to: model)`

It happens much more on a device with 2 buttons.

Thanks.

philips77, 2019-12-16:
Only acknowledged messages will get timeout. Use `StaticAcknowledgedVendorMesasge` instead.
Regarding the first issue, there is a lot of possible reasons why you get timeouts. Without the log I can't tell more.
Also, I don't think number of buttons makes any difference, unless you also have differences in the firmware.

fiveagle, 2019-12-17:
Ok,
I'm gonna fix my custom message.
Anyway when I try to execute all my operation (publish, bind and subscribe) to a node provisioned after 2 other, I encounter a general slowness and more timeOut calls.
I am having this more on a device with 2 physical buttons and 2 unicast address, one for each button. 

My log are

``````
Prepare provision
Start Scanning
Registered NetworkConnection Delegate: Optional<BearerDelegate>
Setto l'address al device: 111
Discovering services...
Vimar progress: 10.0
Initializing...
Vimar progress: 25.0
Identifying...
Vimar progress: 40.0
New state: Requesting Provisioning Capabilities
Identifying...
Vimar progress: 60.0
New state: Provisioning Capabilities received
CapabilitiesReceived:
Number of elements: 1
Algorithms: FIPS P-256 Elliptic Curve
Public Key Type: None
Static OOB Type: Static OOB Information Available
Output OOB Size: 0
Output OOB Actions: None
Input OOB Size: 0
Input OOB Actions: None
Vimar progress: 70.0
Provisioning...
New state: Provisioning started
Vimar progress: 50.0
New state: Provisioning complete
Provisioning complete.
Vimar progress: 100.0
Registered NetworkConnection Delegate: Optional<BearerDelegate>
Finish provisioning node 1888_592!!!
Prepare key binding
Registered NetworkConnection Delegate: Optional<BearerDelegate>
Starting binding procedure
Requesting Composition Data...
10ab_dbmeshnetwork succed
2019-12-17 18:21:24.912403+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] Central Manager state changed to .poweredOn
2019-12-17 18:21:24.913102+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] Connecting to 1888_592...
Config Device Keys progress: 12.5
2019-12-17 18:21:25.211229+0100 Vimar VIEW Wireless[2609:1596013] [Bearer] -> 0x40076DB54BF1D97DE939B5855FAC3CBB6AFE12F9
2019-12-17 18:21:25.213359+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] -> 0xC04DB7
2019-12-17 18:21:25.323708+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] <- 0x4007C9B7978E1BAE3A5D3C4667EF72DBCC184AA4
2019-12-17 18:21:25.324928+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] <- 0xC087552085EFFF2B7116AC
2019-12-17 18:21:25.636366+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] Connected to 1888_592
2019-12-17 18:21:25.636477+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] Discovering services...
2019-12-17 18:21:25.829642+0100 Vimar VIEW Wireless[2609:1595796] [Bearer] -> 0x4007B43943C2E4476061EEBDBF6DB4ADF14E90EA
2019-12-17 18:21:25.832419+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] -> 0xC03FDF6B7274
2019-12-17 18:21:25.964294+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] Service found
2019-12-17 18:21:25.964409+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] Discovering characteristrics...
2019-12-17 18:21:25.983676+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] <- 0x40078AD4EA020FC8661F8172719BED9C2108D035
2019-12-17 18:21:25.984403+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] <- 0xC00A8F0354E4126FB732F3
2019-12-17 18:21:26.350349+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] Data In characteristic found
2019-12-17 18:21:26.350452+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] Data Out characteristic found
2019-12-17 18:21:26.350508+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] Enabling notifications...
2019-12-17 18:21:26.486660+0100 Vimar VIEW Wireless[2609:1596013] [Bearer] -> 0x400738461D462599FFDE82AEB8DEBE83963CF570
2019-12-17 18:21:26.488670+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] -> 0xC07F2E495B26
2019-12-17 18:21:27.248520+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] <- 0x400760F221402465F8856C00DBBB1E983CDD94E0
2019-12-17 18:21:27.249392+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] <- 0xC02B04CD93CE0B5182BCCE
2019-12-17 18:21:27.252762+0100 Vimar VIEW Wireless[2609:1596106] [Bearer] -> 0x400786F84F346B7B51E6F4D5334D396E5043045A
2019-12-17 18:21:27.254540+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] -> 0xC080C80F4EDA
2019-12-17 18:21:27.337961+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] Data Out notifications enabled
[... 549 lines cut ...]
2019-12-17 18:27:31.209962+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] -> 0xC01898FB1BA1
Success Publish
Config Device Keys progress: 87.5
Publish...
Publishing done
Config Device Keys progress: 100.0
Subscribe...
10ab_dbmeshnetwork succed
2019-12-17 18:27:31.925479+0100 Vimar VIEW Wireless[2609:1597059] [Bearer] -> 0x40074669DEE5B5E9BDD34AFAF750C325FF202AB1
2019-12-17 18:27:31.927992+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] -> 0xC0C7CE668B3A1850940849
2019-12-17 18:27:31.931722+0100 Vimar VIEW Wireless[2609:1597059] [Bearer] -> 0x4007AAB5CEF93A552A1644A6E218DCD4A35353BA
2019-12-17 18:27:31.934299+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] -> 0xC02E1F253B
2019-12-17 18:27:31.997746+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] <- 0x4007B394EC87DA7F99395967E35735731E340709
2019-12-17 18:27:31.997910+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] <- 0xC03A69F84827
Subscribing...
2019-12-17 18:27:32.004524+0100 Vimar VIEW Wireless[2609:1597059] [Bearer] -> 0x4007129999E512143BFE4DDE1E466CF0E4F5D301
2019-12-17 18:27:32.010100+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] -> 0xC0EF47AB2E87EF13
2019-12-17 18:27:32.109509+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] <- 0x400706B248ADD4C96CF1F8632C7A3C17AFA1EA5A
2019-12-17 18:27:32.109857+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] <- 0xC0ADB3B98A3A
2019-12-17 18:27:32.218138+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] <- 0x40074A8C6F42A1A86121354D52BC5246E0A18A5D
2019-12-17 18:27:32.218423+0100 Vimar VIEW Wireless[2609:1595249] [Bearer] <- 0xC0A0B5E23BCC2B10ED
Success subscribe
Key bind done
Provisioning finished after 0:6:14
``````

fiveagle, 2019-12-20:
Could it be that I have too many connection on MeshConnection? 
Anyway, how many connection do you think is reasonable to have open and ready in MeshConnection?

Thanks

philips77, 2019-12-21:
Are you taking about the `NetworkConnection` class from the sample app?
It is always using the first one for sending data, even if connected to more. As the product filter is empty for other, they're not used for receiving either. Just to quickly take over of the first one gets disconnected. I don't think this is relevant.
In your case you'd have to check where the message get lost. Do they reach the destination, why is the response not received. Try enabling more logs on the nodes.

fiveagle, 2020-04-10:
Hi,
I'm reopening this thread because we are struggling on the same issue on Android.
On iOS I've solved implementing `StaticAcknowledgedVendorMessage ` using a dummy responseType as:
`static var responseType: StaticMeshMessage.Type = GenericVimarPropertyStatus.self`
with an opCode labeled as Status in the mesh device documentation.
Is this necessary also in Android? And if it is so, how should be implemented since there is not an equivalent protocol (interface) to which I should conform?

Thanks

philips77, 2020-04-11:
Let's ask @roshanrajaratnam

roshanrajaratnam, 2020-04-14:
Hi @fiveagle, have you looked at [`VendorModelMessageAcked`](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/master/android-nrf-mesh-library/meshprovisioner/src/main/java/no/nordicsemi/android/meshprovisioner/transport/VendorModelMessageAcked.java). You could create your own vendor model message by extending this class or just using it as it is.

fiveagle, 2020-04-14:
@antoniovillanova

antoniovillanova, 2020-04-15:
Hi everyone,

thanks for you advices. On Android actually i'm succesfully using VendorModelMessageAcked.

Bu i'm still having issues when i import a mesh network coming from iOS. I already opened an [issue](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/issues/295) where i better explain my scenario.

philips77, 2020-09-30:
Let's close this issue here then as the mentioned issue was closed as well.

closed 2020-09-30 by philips77