nordicsemi/IOS-nRF-Mesh-Library #235: Many provisioners!

Baothien1301, 2019-12-17:
Hi!

When i create a new provisioner, i must set allocatedUnicastRange, allocatedGroupRange, allocatedSceneRange. And they must differ with another provisioners. So, if we set a large range, we will have less provisioner.
I think difference between provisioners is oder to avoid duplicate unicast address when provisioning. If a provisioner just control, does not provision, this provision will not need any range. So, i should set same range with another provisioners or difference range???

deverickasih, 2019-12-17:
You should give the new provisioner 1 address, that address is his own address. He will not be able to provision other nodes but will be able to access all the other nodes set up by the master provisioner. I believe you could use the following post as reference to this answer : #211 :)

philips77, 2019-12-17:
Yes, @deverickasih is right. This library defines provisioners as nodes that know device keys of other node, therefore can configure them. There is no need for them to actually provision anything, but as they are not provisioned by other provisioners, rather they claim they will use this address, the address needs to be reserved for them.

Baothien1301, 2019-12-18:
@deverickasih thanks you. i think provisioner with 1 address is a right way.
@philips77 may be i ask you one more question:
  - In Android, i see a feature: You can choice provisioner as a current provisioner. But in IOS, have not yet. And my question is: If you create a new provisioner, you have 2 provisioner in your database and Json file. When you import it into another app, how you know which provisioner is current provisioner?

philips77, 2019-12-18:
In iOS you need to click EDIT button and drag&drop the provisioner you want to be the "current" to the top.

philips77, 2019-12-19:
"This provisioner" is the one at index 0, the first found in json. To set a different one as "this", you need to move it to the front.
However, the iOS mesh library is a bit smarter than this. Since version 2.1, for each network that was created or imported in nRF Mesh on iOS, the library stores the UUID of the last "this provisioner". If you then switch the network to another one and import the first again, the same provisioner will be set as "this" even if provisioner order in the imported json is different. This is to make sure each phone is a different and unique provisioner.
If you import a new network, unknown to the app (new mesh network UUID), restoration of the previous provisioner will fail, and the app should create a new provisioner. Have a look at this code:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/45e12224e8b5f2d45659e99dddda966b1ba17789/Example/nRFMeshProvision/View%20Controllers/Settings/SettingsViewController.swift#L219-L242

Baothien1301, 2019-12-19:
Thanks you for your response. 
Thanks you very much.

closed 2019-12-19 by Baothien1301