nordicsemi/IOS-nRF-Mesh-Library #237: LightCTLTemperatureGet times out

sunilloharshoreline1, 2019-12-17:
LightCTLTemperatureGet doesn't return the Status and log says Timed Out and eventually says: Cancelling messages with op code: 33377

Here are the logs : 
2019-12-17 18:29:37.825208+0530 Aether Ble[846:861850] [Access] Response to Access PDU (opcode: 0x8261, parameters: 0x) not received (timeout)2019-12-17 18:29:37.826234+0530 Aether Ble[846:861850] [Model] LightCTLTemperatureGet() sent from: 0001, to: 0002 timed out2019-12-17 18:29:37.827016+0530 Aether Ble[846:861850] [Access] Cancelling messages with op code: 33377, sent from: 0001 to: 00022019-12-17 18:33:28.793108+0530 Aether Ble[846:863316] [Model] Sending LightCTLTemperatureGet() from: Element 0 (0x0001), to: 00022019-12-17 18:33:28.799359+0530 Aether Ble[846:863316] [Access] Sending Access PDU (opcode: 0x8261, parameters: 0x)2019-12-17 18:33:28.804859+0530 Aether Ble[846:863316] [UpperTransport] Sending Upper Transport PDU (encrypted data: 0x0D6F, transMic: 0x2D0E12E1) encrypted using key: App Key 1 (index: 0)2019-12-17 18:33:28.805988+0530 Aether Ble[846:863316] [LowerTransport] Sending Access Message (akf: 1, aid: 0x3F, szmic: 0, data: 0x0D6F2D0E12E1)2019-12-17 18:33:28.810856+0530 Aether Ble[846:863316] [Network] Sending Network PDU (ivi: 0, nid: 0x1D, ctl: 0, ttl: 5, seq: 577, src: 0001, dst: 0002, transportPdu: 0xD5E7B5D5916801, netMic: 0xC1F55063) encrypted using Primary Network Key (index: 0)2019-12-17 18:33:28.818321+0530 Aether Ble[846:863316] [Bearer] -> 0x001D303FF95F2CB7B406D5E7B5D5916801C1F550632019-12-17 18:33:34.140162+0530 Aether Ble[846:863316] [Access] Resending Access PDU (opcode: 0x8261, parameters: 0x)2019-12-17 18:33:34.141668+0530 Aether Ble[846:863316] [UpperTransport] Sending Upper Transport PDU (encrypted data: 0x8733, transMic: 0x8B99C3FD) encrypted using key: App Key 1 (index: 0)2019-12-17 18:33:34.142110+0530 Aether Ble[846:863316] [LowerTransport] Sending Access Message (akf: 1, aid: 0x3F, szmic: 0, data: 0x87338B99C3FD)2019-12-17 18:33:34.144224+0530 Aether Ble[846:863316] [Network] Sending Network PDU (ivi: 0, nid: 0x1D, ctl: 0, ttl: 5, seq: 578, src: 0001, dst: 0002, transportPdu: 0x0F4ECAE395B56B, netMic: 0x0F24033E) encrypted using Primary Network Key (index: 0)2019-12-17 18:33:34.145629+0530 Aether Ble[846:863316] [Bearer] -> 0x001DA4C298E7976D746B0F4ECAE395B56B0F24033E2019-12-17 18:33:44.840034+0530 Aether Ble[846:863405] [Access] Resending Access PDU (opcode: 0x8261, parameters: 0x)2019-12-17 18:33:44.842072+0530 Aether Ble[846:863405] [UpperTransport] Sending Upper Transport PDU (encrypted data: 0x8BDC, transMic: 0x260F1C52) encrypted using key: App Key 1 (index: 0)2019-12-17 18:33:44.843103+0530 Aether Ble[846:863405] [LowerTransport] Sending Access Message (akf: 1, aid: 0x3F, szmic: 0, data: 0x8BDC260F1C52)2019-12-17 18:33:44.847021+0530 Aether Ble[846:863405] [Network] Sending Network PDU (ivi: 0, nid: 0x1D, ctl: 0, ttl: 5, seq: 579, src: 0001, dst: 0002, transportPdu: 0x5F165628A9F0A4, netMic: 0xEFF9D47F) encrypted using Primary Network Key (index: 0)2019-12-17 18:33:44.849348+0530 Aether Ble[846:863405] [Bearer] -> 0x001DD32A3AC2161D12845F165628A9F0A4EFF9D47F2019-12-17 18:34:06.141121+0530 Aether Ble[846:863496] [Access] Resending Access PDU (opcode: 0x8261, parameters: 0x)2019-12-17 18:34:06.142773+0530 Aether Ble[846:863496] [UpperTransport] Sending Upper Transport PDU (encrypted data: 0xCEE7, transMic: 0xA057FFF8) encrypted using key: App Key 1 (index: 0)2019-12-17 18:34:06.143212+0530 Aether Ble[846:863496] [LowerTransport] Sending Access Message (akf: 1, aid: 0x3F, szmic: 0, data: 0xCEE7A057FFF8)2019-12-17 18:34:06.145444+0530 Aether Ble[846:863496] [Network] Sending Network PDU (ivi: 0, nid: 0x1D, ctl: 0, ttl: 5, seq: 580, src: 0001, dst: 0002, transportPdu: 0xE12521E7AC27C7, netMic: 0xE35D9DDB) encrypted using Primary Network Key (index: 0)2019-12-17 18:34:06.146190+0530 Aether Ble[846:863496] [Bearer] -> 0x001DAE438D9C39CBFC1EE12521E7AC27C7E35D9DDB2019-12-17 18:34:08.865112+0530 Aether Ble[846:863496] [Access] Response to Access PDU (opcode: 0x8261, parameters: 0x) not received (timeout)2019-12-17 18:34:08.866198+0530 Aether Ble[846:863496] [Model] LightCTLTemperatureGet() sent from: 0001, to: 0002 timed out2019-12-17 18:34:08.866926+0530 Aether Ble[846:863496] [Access] Cancelling messages with op code: 33377, sent from: 0001 to: 0002

philips77, 2019-12-17:
As you see, the message gets sent and no response is received during 40 seconds. There may be number of reasons for this:
1. The proxy filter is not configured to accept messages targeting 0x0001 (ensure it's white list filter and this address is added, you may also try with an empty black list filter).
2. Sequence numbers are too low and they are rejected by the target. Did you reuse the same provisioner's address on another phone?
3. The request never reaches the target, for example the nodes between 0x0001 and 0x0002 (if any) don't know the Primary Index Key, therefore can't relay message.
4. The node 0x0002 is down, reset or does not have Light CTL Server Model.
5. The Proxy Feature is disabled on the currently connected Proxy (assuming it's not 0x0002 that you're connected to).
6. Perhaps some other options as well.

sunilloharshoreline1, 2019-12-20:
Somehow, it is resolved after reset and reprovisioning. I will close the issue for now.

closed 2019-12-20 by sunilloharshoreline1