nordicsemi/IOS-nRF-Mesh-Library #239: Importing MeshNetwork from Android 1.x

fiveagle, 2019-12-24:
**Describe the bug**
Hi, unfortunately I need to import a meshNetworkData exported from Android Library 1.x 
Is it possible? Right now it fails because my json is expected to be inside a MeshData Object and some types are different.

Thank You in advance.

philips77, 2019-12-25:
You need to use`import` method, not `load()`. Importing expects just the network object. Load, which is using the storage protocol, indeed it's MeshData which used to contain some additional data, but is empty not, besides the network object.

fiveagle, 2019-12-30:
Ok,
But I need to create a MeshNetwork before importing Android. And then does import() overwrites the new provisioner with the imported one? Because I erase all data when I close the app and when I switch between MeshNetworks for different houses.

Thanks

fiveagle, 2019-12-31:
Ok, but do I need to create a new MeshNetwork each time before importing it? Because I erase all data on the device at each run. And does the import() overwrites the provisioner just created with the old one?
My problem is that I need to manage a network created from Android but also created by me 
Thanks.

philips77, 2019-12-31:
Import replaces the current Mesh Network with w new one. No need to created one before importing, it will be overwritten.

closed 2020-02-05 by philips77