nordicsemi/IOS-nRF-Mesh-Library #241: Sense

Baothien1301, 2019-12-28:
I am using sense for my project. And i got a difficult matter when i want to share my senses. Because in json file i don't see anythings about sense. I have to share sense by another way. 
So, i would like to ask you: Do you have any idea for me? Or Do you have a plan about this matter?

philips77, 2019-12-28:
What do you mean by Sense?

Baothien1301, 2019-12-28:
I'm so sorry about my mistake. i mean "Scense" not "Sense".
I am using scense for my project. And i got a difficult matter when i want to share my scenses. Because in json file i don't see anythings about scense. I have to share scense by another way.
So, i would like to ask you: Do you have any idea for me? Or Do you have a plan about this matter?

philips77, 2019-12-29:
Oh, scenes. As they're not implement yet in the sample app, we skipped them from the library as well. They should be in the Json at some point. I'll try to add it in the next release, but I can't promise any date now.

Baothien1301, 2020-01-02:
Thanks you. You are great!

philips77, 2020-09-30:
Sharing scenes is now implemented and available in #290. It will be released as version 3.0 (soon on GitHub, then on AppStore when Android version is updated as well).

closed 2020-09-30 by philips77