nordicsemi/IOS-nRF-Mesh-Library #243: Can Phone control Node directly?

Baothien1301, 2020-01-03:
Hello. It's me again!
A phone can be a provisioner. And when we create a provisioner, we also create a node.
I think a phone can be a node, so a phone can control another node directly and don't need a proxy connection. But I am not sure. 
Could you answer me? And if phone can control a node, how i do ?
If this is possible, i think BLE mesh is greate!

philips77, 2020-01-03:
Unfortunately, this is not possible neither on iOS and Android these days. To use ADV Bearer, a phone would have to advertise with Mesh Message AID (0x2A, see: https://www.bluetooth.com/specifications/assigned-numbers/generic-access-profile/). Currently, the iOS API allows only to set a local name and supported service UUIDs: https://developer.apple.com/documentation/corebluetooth/cbperipheralmanager/1393252-startadvertising
On Android the possible options are bigger, but there's no Mesh Message parameter and no custom AID type, which could be used for this purpose.
Therefore, Android and iOS require a GATT Bearer (a proxy) to communicate with Bluetooth Mesh network.

Baothien1301, 2020-01-04:
Thank for your response.
I wished it's possible. If It's impossible, I have another question.
For now, when we create a provisioner, we also create a node with some elements, some models. When we want to recieve a message - send to a group or a node through a model, we have to create a node with that model and subcribe a address of that group or that unicast address of node (But cannot subcribe a unicast address). How do a provisioner can recieve all message in network without create model and without subcribe all address.

philips77, 2020-01-06:
You need to set the Proxy Filter to black list without adding any address to it. You will receive the [meshNetworkManager(_:didReceiveMessage:sentFrom:to)](https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/45e12224e8b5f2d45659e99dddda966b1ba17789/nRFMeshProvision/Classes/MeshNetworkDelegate.swift#L53) for each message that the proxy node received.

closed 2020-02-05 by philips77