nordicsemi/IOS-nRF-Mesh-Library #250: Long time no see!

Baothien1301, 2020-02-05:
Since 2 months, i have not seen any update from you! I saw many features you are developting. I am looking forward to new version with those features. I would like to ask you: "When do you update a new version?"
Thanks!

philips77, 2020-02-05:
Hello!
We are currently busy with other projects. Unfortunately, mesh needs to wait for some time (few months at least).
What features are you talking about? I think the next step for us is to add Scenes management. Currently there is nothing in the UI. Also, Scenes should be saved in the JSON file when exported, so this has to be done. But other than that, the lib seems to be feature complete. Of course there is still need to add more and more message types and perhaps DFU when the spec is finally released.

philips77, 2020-04-14:
Hello,
I'm about to publish version 2.3.0. It will have fixed IV Index Update (handling Secure Network Beacon with `updateActive` flag set). Also some minor additions to the API. I'm going through the open issues looking for an inspiration. Any ideas?

Baothien1301, 2020-04-21:
Hi. Thanks for asking me. I think some features like Scenes, DFU, more message types are necessary. I am very looking forward to new version.

philips77, 2020-04-23:
IV Index update is now handled correctly by the version on *develop* branch. I have to implement some unit tests, but my tests on the device look promising. This will go to version 2.2.2, leaving other features for the future.

philips77, 2020-09-30:
IV Index Update was released in 2.2.2.
I'm about to prepare 3.0 release (App Store release must wait until the Android lib gets updated and compatible with all breaking changes). There's lot of new features coming. Please go through latest PRs.

philips77, 2021-02-05:
Version 3.0 will be released on Monday. Check out #295 for details.

philips77, 2021-05-21:
Ok, I'm closing the issue. For versions and changes, please check Releases.

closed 2021-05-21 by philips77