nordicsemi/IOS-nRF-Mesh-Library #253: Example app and App Store app get stuck on "Requesting Composition Data"

PhilippeRivard, 2020-02-20:
**Describe the bug**
The example application and app store application gets stuck on "Requesting Composition Data"

**To Reproduce**
Steps to reproduce the behavior:
1. Provision a node
2. Provision a second node

**Expected behavior**
Should be able to provision more than one node and control the bluetooth mesh devices.

**Platform details:**
 - Device: iPhone 11
 - OS: iOS 13
 - Library Version v2.2.1

**Logs / Screenshots**
Add any logs or screenshots that can help us debug the issue faster

philips77, 2020-02-20:
Duplicate of https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/issues/153

philips77, 2020-02-20:
The timeout depends on the network settings, number of nodes, size of composition data, etc.
In the current version of the app, on Proxy tab, you'll find a switch to enable manual connection. Turn it on and connect to your newly provisioned device. Then open Network tab and open the device page to obtain Composition Data directly from this node using GATT Proxy protocol.

philips77, 2020-02-20:
The mesh library allows to configure those settings in the manager. Have a look where the sample app is doing so: 
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/45e12224e8b5f2d45659e99dddda966b1ba17789/Example/nRFMeshProvision/AppDelegate.swift#L46-L57
(parameters are documented in the `MeshNetworkManager` class: https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/45e12224e8b5f2d45659e99dddda966b1ba17789/nRFMeshProvision/Classes/MeshNetworkManager.swift#L58

closed 2020-02-20 by philips77

PhilippeRivard, 2020-02-20:
I really appreciate the quick response, but can you clarify which setting exactly is causing a glitch when adding more than 1 node? Also why is adding the first node seamless with no freeze on “requesting composition data”? If I wanted to make the example app work correctly then what is it I have to do?

philips77, 2020-02-20:
The Composition Data from the first node is sent directly, via GATT. After provisioning second and more nodes, the app keeps the connection to the first gatt proxy. Only provisioning of done directly. Composition Data must not go through this proxy node, which makes it slower and prone to packet lose. Network properties should be tailored per network. If you have large network, perhaps timeouts could be longer, Network Retransmit set, etc. For your network, where you host has l have 2 devices so far, you may check the log and try to find out the bottleneck. It also depends on your nodes' config, their timeouts, etc.

PhilippeRivard, 2020-02-20:
In the sentence: "Composition Data must not go through this proxy node, which makes it slower and prone to packet lose." Do you mean "Composition data must go through this proxy node, which makes it slower and prone to packet lose."?

philips77, 2020-02-21:
Sure, I'll edit my initial response. I'm writing in the phone which adds "not" wherever it wants... ;)