nordicsemi/IOS-nRF-Mesh-Library #256: Optimize Proxy Communication

horsejockey, 2020-02-28:
**Is your feature request related to a problem? Please describe.**
This isn't really a problem but more of an idea for optimization. I'm new to mesh and this library though so I want to make sure I'm on the right track before I fork the repo or work on a PR.  The library allows you to connect to multiple available proxy nodes and also specify a node you would like to use for communication.  I think an easy improvement would be to check at the transmitter level if the destination node is one of the proxy nodes the phone has a connection to. If that is the case it makes the most sense to send communications directly to that node without having to add an extra communication step. Especially if you have some large payloads you are dealing with. At least I think it does (new to mesh and this sdk).

**Describe the solution you'd like**
The solution would be to pass along some extra information with at least the node destination address if it has one to the Transmitter so it could choose the best node to talk to at that point. Right now the transmitter send function only gets the data and pdu type.

philips77, 2020-02-29:
Hello, your feature would be indeed useful, but it's not really feasible. First of all, direct connection breaks really the whole mesh network idea: "I have a mesh network, but it's so unreliable that I prefer to connect directly".
Secondly, the device address or unicast address are not available before connection (on iOS Mac address is not available and devices are free to change it with every packet). The device can advertise with Node Identity, which contains the unicast address, but it usually does it only after provisioning to allow configuration. After connection to a Node it is possible to send a packet with TTL 1 and check the response source address to get device address.
I'll discuss it with the team.

horsejockey, 2020-03-02:
That all makes sense.  I just thought it was a small optimization that could save the user time with large payloads if they happened to have the proxy node already in their connection pool. We often send payloads in the 100 to 150 byte range.

Looks like if I wanted to make it work reliably I would request the unicast address from every node I add to my direct connection pool and then I'd be able to select the node I wanted based on the message destination if it was in the pool.  If not I'd just send to a random node.

philips77, 2020-04-23:
We decided not to implement such feature. As I wrote before, the only way to get proxy's unicast address is to connect to it and check the source address of some message, e.g. proxy filter message. We could display it to the user, but if the user decided to connect to a different node, we are were we began. All nodes, if they advertise the same network id, are similar for us, so we would have to randomly select a new node and try again.

closed 2020-04-23 by philips77