nordicsemi/IOS-nRF-Mesh-Library #257: Key Refresh Procedure, IV Index update.

BubblyNetDev, 2020-03-03:
Hello, I know you're not currently working on this project. But I was wondering when you may be back as well as what the priority would be for Key Refresh Procedure, IV Index update upon your return?

philips77, 2020-03-04:
Hello,
I don't know when can I resume work on mesh, I have bunch of other tasks now. In couple on months, I suppose? 
The features you requested and l are the last of the Big Features, so they have high priority.

philips77, 2020-04-23:
IV Index update is not implemented on *develop* branch.

philips77, 2020-09-30:
Handling IV Index Update was released in version 2.2.2. The library does not initiate this procedure on its own, as it's very unlikely that it would run out of sequence numbers before any other node. After all, mobile phone is only sometimes present as a node, but other nodes communicate all the time.

closed 2020-09-30 by philips77