nordicsemi/IOS-nRF-Mesh-Library #258: Too big unicast address

fiveagle, 2020-03-05:
**Describe the bug**
Hi,
I have been reported a bug on the app I am developing.
Apparently sometimes the nodes gets provisioned with an unicast address way too big. Like 13000. 
The app has the requirement to use addresses starting from 100. Thats why I've asked you, a while back,  a way to get a nextAvailableUnicastAddress after a certain threshhold.

My implementation is below but I am not sure that it is causing the problem.

Can you recommend a way to accomplish this?
As soon as I get the mesh network json I'll add it to this thread

Thank You

```swift
extension MeshNetwork {
    func nextAvailableUnicastAddress(startingFrom offset: Address, using provisioner: Provisioner) -> Address? {
        let sortedNodes = nodes.sorted { $0.unicastAddress < $1.unicastAddress }
        
        // Iterate through all nodes just once, while iterating over ranges.
        var index = 0
        for range in provisioner.allocatedUnicastRange {
            // Start from the beginning of the current range.
            var address = range.lowAddress
            
            if range.contains(offset) && address < offset {
                address = offset
            }
            
            // Iterate through nodes that weren't checked yet.
            let currentIndex = index
            
            for _ in currentIndex..<sortedNodes.count {
                let node = sortedNodes[index]
                index += 1
                
                // Skip nodes with addresses below the range.
                if address > node.lastUnicastAddress {
                    continue
                }
                // If we found a space before the current node, return the address.
                if address + UInt16(1) - 1 < node.unicastAddress {
                    return address
                }
                // Else, move the address to the next available address.
                address = node.lastUnicastAddress + 1
                
                // If the new address is outside of the range, go to the next one.
                if address + UInt16(1) - 1 > range.highAddress {
                    break
                }
            }
            
            // If the range has available space, return the address.
            if address + UInt16(1) - 1 <= range.highAddress {
                return address
            }
        }
        // No address was found :(
        return nil
    }
}
```

philips77, 2020-03-06:
Hello,
The code above looks good to me. You have just added 
```swift
if range.contains(offset) && address < offset {
    address = offset
}
```
correct?
Perhaps the ranges for the provisioner are set this way? Very hard to say.

fiveagle, 2020-03-06:
Hi, 
I'm investigating the issue. 
Could it be caused by the fact that we save the meshNetwork json on cloud and then access it from different iPhones, each time I import the network creating a new provisioner if necessary.

```swift
let provisioner = Provisioner(name: UIDevice.current.name,
                                      allocatedUnicastRange: [AddressRange(0x0001...0x199A)],
                                      allocatedGroupRange:   [AddressRange(0xC000...0xCC9A)],
                                      allocatedSceneRange:   [SceneRange(0x0001...0x3333)])
```

Apparently even if the unicastAddress range upper bound is 6554 I get addresses around 19000

Below the full meshNetwork print

Thanks,

[MeshNetwork.txt](https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/files/4297535/MeshNetwork.txt)

philips77, 2020-03-06:
Yes. In your file I can see an IPadPro_RTA, which has `allocatedUnicastRange`: 0x4CCD-0x6665.
0x4CCD = 19661. Also, this looks like a 3rd provisioner on this device. If you didn't modify the code, the total unicast range is split into 5 provisioners, each getting around 0x199A addresses. So the first one gets 0x0001-0x199A, the second up to 0x3333 and the third one the range above. So it looks, like you were importing this json on another iDevice, which created a provisioner, and then on the iPad, which added 3rd one. And then you have provisioned the device so some of the first addresses from this range were assigned.

fiveagle, 2020-03-06:
Ok,
So there is a way to limit the addresses upper bound without limiting the number of devices?

fiveagle, 2020-03-06:
I mean the number of provisioners..

philips77, 2020-03-06:
Well.. the unicast address range is from 0x0001 to 0x7999. You may split it on how many provisioners you want. We decided to use 5, as then the graph looks ok (when more the ranges are barely visible, and when less, it's hard to demonstrate more provisioners or test on more devices). but when creating a provisioner (like in the code you posted above, you declare the ranges). So if you have a device that's not going to provision any nodes, just assign the range for elements it has. If you want to provision less than 0x199A, use your estimated value. If you have a lot of installers (provisioners), you can split the total range into small ranges.

fiveagle, 2020-03-06:
Ok but I don't know how much provisioners I'm going to add. Is it possible to reduce the provisioner unicast range of the old provisioners each time I add a new one? 
Basically I need to keep the number below a certain threshold without limiting the number of provisioners.

fiveagle, 2020-03-06:
That's what I do for each import with a new provisioner
```Swift
let meshNetwork = try self.meshNetworkManager.import(from: data)
                // Try restoring the Provisioner used last time on this device.
                if meshNetwork.restoreLocalProvisioner() == false {
                    // If it's a new network, try creating a new Provisioner.
                    // This will succeed if available ranges are found.
                    if let nextAddressRange = meshNetwork.nextAvailableUnicastAddressRange(ofSize: 0x199A),
                       let nextGroupRange = meshNetwork.nextAvailableGroupAddressRange(ofSize: 0x0C9A),
                       let nextSceneRange = meshNetwork.nextAvailableSceneRange(ofSize: 0x3334) {
                        let newProvisioner = Provisioner(name: UIDevice.current.name,
                                                         allocatedUnicastRange: [nextAddressRange],
                                                         allocatedGroupRange: [nextGroupRange],
                                                         allocatedSceneRange: [nextSceneRange])
                        // Set it a a new local Provisioner.
                        try? meshNetwork.setLocalProvisioner(newProvisioner)
                        // And assign a Unicast Address to it.
                        if let address = meshNetwork.nextAvailableUnicastAddress(for: newProvisioner) {
                            try? meshNetwork.assign(unicastAddress: address, for: newProvisioner)
                        }
                    }
                }
```

If I set the size for the `nextAvailableUnicastAddressRange` to 1 can I provision more than 1 device? I would say no but then how can I limit the size of the total network unicastAddressRange?

Thanks

fiveagle, 2020-03-06:
And what about overlapping addressRanges? Can 2 provisioner have part of their range in common?

philips77, 2020-03-06:
So in theory a provisioner should never provision a device with address outside of its range and no two ranges of the same type should overlap. This is to avoid conflicts in the mesh, when 2 nodes would be provisioned with the same address by two different provisioners (before the network database is merged).
If you are sure that only one provisioner is provisioning at a time, you may keep available unicast range equal to the number of elements on this provisioner, and just extend it if needed to provision a new node. Then collapse it again. But that would just be a hack.
The unicast range allows to have 31128 addresses (elements) in the network (including provisioners). You need to limit number of provisioners or nodes somehow. You may e.g. grant each new provisioner 100 addresses and extend if necessary (additional address range may be added). This is a better approach, and also requires some efforts to make sure that after expansion all provisioners are synchronized, so that the new range isn't assigned to more than 1 provisioner.

fiveagle, 2020-03-06:
Ok, this approach seems good. I have only one caveat: I need to ensure that the central device of the network is always assigned to unicastAddress 2 when provisioned, independently by which provisioner will do the actual provisioning. 
There is a way to change the unicastAddress after the provisioning, or allowing a provisioner to use unicast 2 no matter what? 
In android looks like I can force an assignment on whatever unicast I need.

Or is it possible to not create a new provisioner at each import on a different phone?

Thank You

philips77, 2020-03-07:
The reason why each phone should be a different provisioner is the sequence number. Each phone should have a different unicast address assigned, as seq isn't sync. Otherwise only one of them, the one with the greatest seq number, would be able to send any messages.
What you can do, is that you can have multiple provisioners, each with a different address, and you assign them unicast ranges from 3 (or 100, or whatever) and none of them gets address 2. When any one wants to provision the central device you add this address to its ranges and then provision the central device.

The address may not be charged after provisioning. You need to unprovision and reprovision the device again.

philips77, 2020-03-07:
In the iOS mesh library you may also assign any address, but this address needs to be in provisioner's ranges, so make sure you add this address before provisioning.

philips77, 2020-04-08:
Hi, did you solve your problem? Can we close the issue?

fiveagle, 2020-04-08:
Yes. I ended deleting all existing provisioners and then creating a new one with a new unique unicast address. Thanks

closed 2020-04-08 by fiveagle

philips77, 2020-04-14:
@fiveagle Have a look at a30cb2a.

fiveagle, 2020-04-15:
Hi,
I'm glad you add my suggestions.
I have been extending the method with the inclusion of the blackList logic.
I don't know if this will be used in the future for the actual SIG management of the blacklist. Either way I'm putting it here for reference.

Thanks

```Swift
func nextAvailableUnicastAddress(startingFrom offset: Address,
                                     excluding blacklist: [Address] = [],
                                     using provisioner: Provisioner) -> Address? {
        
        let sortedNodes = nodes.sorted { $0.unicastAddress < $1.unicastAddress }
        
        // Iterate through all nodes just once, while iterating over ranges.
        var index = 0
        
        for range in provisioner.allocatedUnicastRange {
                                    
            // Start from the beginning of the current range.
            var address = range.lowAddress
            
            if range.contains(offset) && address < offset {
                address = offset
            }
            
            // Iterate through nodes that weren't checked yet.
            let currentIndex = index
            
            for _ in currentIndex ..< sortedNodes.count {
                let node = sortedNodes[index]
                index += 1
                
                // Skip nodes with addresses below the range.
                if address > node.lastUnicastAddress {
                    continue
                }
                
                // If we found a space before the current node, return the address.
                if address + UInt16(1) - 1 < node.unicastAddress, blacklist.contains(address) == false {
                    
                    return address
                }
                // Else, move the address to the next available address.
                address = node.lastUnicastAddress + 1
                
                // If the new address is outside of the range, go to the next one.
                if address + UInt16(1) - 1 > range.highAddress {
                    break
                }
            }
            
            // If the range has available space, return the address.
                        
            if address + UInt16(1) - 1 <= range.highAddress, blacklist.contains(address) == false {
                
                return address
                
            } else if var afterBlack = blacklist.first {
                
                afterBlack += UInt16(1)
                
                if afterBlack <= range.highAddress {
                    return afterBlack
                }
            }
        }
        // No address was found :(
        return nil
    }
```