nordicsemi/IOS-nRF-Mesh-Library #259: [CoreBluetooth] XPC connection invalid

jcarioti, 2020-03-18:
**Describe the bug**
Any time I try to open a PBGattBearer, I get this message and none of my bearer delegate methods fire.

**To Reproduce**
Mimic NRF Mesh iOS app up to the point of searching for a device and opening a provisioning invitation. I *think* this might have something to do with the fact that I'm using all of these toolkits outside of the context of a view controller... I've read that the reference to CBCentralManager needs to be within the context of a view controller and that's about the only thing I can tell that's different between my app and the example program.

**Expected behavior**
The bearer opens successfully.

**Platform details:**
iPhone 7 Plus
iOS 13.4

I've ensured that my info.plist file has the appropriate BT keys, as well as turned on the BLE Accessory capabilities. I've also seen the other open issue about this problem, but nothing in the post helped.

philips77, 2020-03-18:
But do you make any problem, or just this message is printed? I've seen this issue on our side, but it's seems everything is working.

jcarioti, 2020-03-18:
> But do you make any problem, or just this message is printed? I've seen this issue on our side, but it's seems everything is working.

I believe there is an error, because none of the GattBearerDelegate methods fire after calling bearer.open().

```
self.unprovisionedDevice = BluetoothPeripheralScanner.shared.discoveredPeripherals[0].device
// Open the gatt bearer for provisioning... logic continues in func bearerDidOpen()
let bearer = PBGattBearer(target: BluetoothPeripheralScanner.shared.discoveredPeripherals[0].peripheral)
bearer!.logger = MeshNetworkManager.instance.logger
bearer!.delegate = self
bearer!.open()
```

This is the last code that runs before I see that error, and then nothing else happens. None of these methods ever fire:

```
extension BluetoothMeshBridge: GattBearerDelegate {
    func bearerDidConnect(_ bearer: Bearer) {
        DispatchQueue.main.async {
          print("DISCOVERING SERVICES...")
        }
    }
    func bearerDidDiscoverServices(_ bearer: Bearer) {
        DispatchQueue.main.async {
          print("INITIALIZING...")
        }
    }
    func bearerDidOpen(_ bearer: Bearer) {
        DispatchQueue.main.async {
          print("DID OPEN")
          print("Provisioning step 2")
          // Continues if successful from func provisionDevice()... convert the bearer into a provisioning bearer and begin the provisioning process
          BluetoothDeviceProvisioner.shared.bearer = bearer as? ProvisioningBearer
          BluetoothDeviceProvisioner.shared.inviteDeviceForProvisioning()
        }
    }
    func bearer(_ bearer: Bearer, didClose error: Error?) {
        DispatchQueue.main.async {
          print("bearer closed?")
        }
    }
}
```

philips77, 2020-03-19:
Could you post here the logs you get in logger?

jcarioti, 2020-03-20:
My AppDelegate is my LoggerDelegate, but the log function never gets called. Here's part of my AppDelegate code:

```
class AppDelegate: UIResponder, UIApplicationDelegate, LoggerDelegate {
  func log(message: String, ofCategory category: LogCategory, withLevel level: LogLevel) {
    print("Mesh Network Log: \(message) \(category) \(level)")
  }
  ...
}
```

Here are some logs showing the order of things:

```
2020-03-19 20:46:17.549 [info][tid:main][RCTRootView.m:293] Running application BLEMeshApp ({
    initialProps =     {
    };
    rootTag = 1;
})
Creating MeshNetworkManager instance
Set AppDelegate as the MeshNetworkManager logger delegate
Loading mesh network
Saving local elements
Creating NetworkConnection from mesh network
Set AppDelegate as the NetworkConnection logger delegate
Opening network connection
Create a new app key
Add app key to network
Bind app key to network key
Central Manager not ready
Central Manager start scanning for peripherals
Central Manager discovered peripheral (TestBLEDevice)
Provision (TestBLEDevice)
Stop scanning
Saved reference to unprovisionedDevice
Create PBGattBearer instance from peripheral
Opening bearer
2020-03-19 20:47:08.142592-0700 BLEMeshApp[670:419829] [CoreBluetooth] XPC connection invalid
```

That XPC error is the last thing that happens after I try to open the PBGattBearer. None of the delegate methods ever fire.

philips77, 2020-03-20:
You need to set the delegate to the `MeshNetworkManager.logger`:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/45e12224e8b5f2d45659e99dddda966b1ba17789/nRFMeshProvision/Classes/MeshNetworkManager.swift#L50
 and `Bearer.logger`:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/45e12224e8b5f2d45659e99dddda966b1ba17789/nRFMeshProvision/Classes/Bearer/GATT/BaseGattProxyBearer.swift#L44

jcarioti, 2020-03-20:
I have set the logger delegate for MeshNetworkManager as well as PBGattBearer...

```
print("Creating MeshNetworkManager instance")
    meshNetworkManager = MeshNetworkManager()
    // Read properties documentation for details.
    meshNetworkManager.acknowledgmentTimerInterval = 0.600
    meshNetworkManager.transmissionTimerInteral = 0.600
    meshNetworkManager.retransmissionLimit = 2
    meshNetworkManager.acknowledgmentMessageInterval = 5.0
    meshNetworkManager.acknowledgmentMessageTimeout = 40.0

    print("Set AppDelegate as the MeshNetworkManager logger delegate")
    **_meshNetworkManager.logger = self_**
    
    var loaded = false
    do {

      print("Loading mesh network")
      loaded = try meshNetworkManager.load()
    } catch {
        print(error)
    }

    // If load failed, create a new MeshNetwork.
    if !loaded {
        createNewMeshNetwork() // See AppDelegate in Sample App
    } else {
        meshNetworkDidChange() // See AppDelegate in Sample App
    }
```
 
```
print("Create PBGattBearer instance from peripheral")
let bearer = PBGattBearer(target: BluetoothPeripheralScanner.shared.discoveredPeripherals[0].peripheral)
bearer!.logger = MeshNetworkManager.instance.logger
bearer!.delegate = self
      
print("Opening bearer")
bearer!.open()
```

Still nothing is being logged.

jcarioti, 2020-03-20:
I've figured out my issue. From the Example Code, it looked like I needed to open this PBGattBearer directly, but that doesn't seem to be working for some reason. I'm now creating the PGGattBearer instance, setting the logger, casting as a ProvisioningBearer and passing it off to my provisioner to be opened. Since doing that, my bearer delegate methods are firing and I'm now seeing logs from the LoggerDelegate.

Here is the working code:
```
print("Provisioning step 1")
let bearer         = PBGattBearer(target: self.unprovisionedPeripheral)
bearer?.logger = MeshNetworkManager.instance.logger
      
BluetoothDeviceProvisioner.shared.bearer = bearer! as ProvisioningBearer
BluetoothDeviceProvisioner.shared.inviteDeviceForProvisioning()
```
and inside the BluetoothDeviceProvisioner:
```
func inviteDeviceForProvisioning() {
    print("Provisioning step 2")
    
    let manager                                 = MeshNetworkManager.instance
    let primaryNetworkKey               = manager.meshNetwork?.networkKeys[0]
    
    self.provisioningManager            = try! manager.provision(unprovisionedDevice: self.unprovisionedDevice, over: self.bearer)
    self.provisioningManager.delegate   = self
    self.provisioningManager.logger = MeshNetworkManager.instance.logger
   
    self.provisioningManager.networkKey = primaryNetworkKey
    self.bearer.delegate                     = self
    
    self.bearer.open()
  }
```

I'm still unsure as to why opening a PBGattBearer was not working since that's how the Example application seems to be configured (ScannerTableViewController:100)
```
override func tableView(_ tableView: UITableView, didSelectRowAt indexPath: IndexPath) {
        tableView.deselectRow(at: indexPath, animated: true)
        
        let bearer = PBGattBearer(target: discoveredPeripherals[indexPath.row].peripheral)
        bearer.logger = MeshNetworkManager.instance.logger
        bearer.delegate = self
        
        stopScanning()        
        selectedDevice = discoveredPeripherals[indexPath.row].device
        
        alert = UIAlertController(title: "Status", message: "Connecting...", preferredStyle: .alert)
        alert!.addAction(UIAlertAction(title: "Cancel", style: .cancel) { action in
            action.isEnabled = false
            self.alert!.title   = "Aborting"
            self.alert!.message = "Cancelling connection..."
            bearer.close()
        })
        present(alert!, animated: true) {
            bearer.open()
        }
    }
```

Is there something different between opening a PBGattBearer and opening a ProvisioningBearer? They look to be nearly identical.

philips77, 2020-03-20:
No, this shouldn't matter. `PBGattBearer` is a `ProvisioningBearer` so the reference type is irrelevant.

philips77, 2020-04-08:
Hi, did you find a solution? Can we close this issue?

jcarioti, 2020-04-08:
> Hi, did you find a solution? Can we close this issue?

Yes, I restructured the provisioning steps and it started working. I still have an issue where 3/4 provisioning attempts don't work, but the example app doesn't seem to be having that issue so I'm still debugging the difference between the two. This issue can be closed.

philips77, 2020-04-09:
Great. Good luck!

closed 2020-04-09 by philips77

trusbe, 2020-12-01:
@jcarioti @philips77 
I also encountered the same problem, but I can solve it by using the following methods:

1. Set Property
```
private var bearer: PBGattBearer!
```

2. Init
```
bearer = PBGattBearer(target: self.devItem.per)
bearer.logger = MeshNetworkManager.instance.logger
bearer.delegate = self
bearer.open()
```