nordicsemi/IOS-nRF-Mesh-Library #260: Access PDU Op codes computed differently in Android implementation

kiyoshi-brilliant, 2020-03-24:
---

**Describe the bug**
Logic to compute Op codes does not match Android implementation. Also filing on Android repo.
Most concerning is the default / else case treatment but the other cases are slightly mismatched as well.
Please provide guidance as to which is correct.
Android issue here: https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/issues/293

Android:
https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/fd90df15ebb03222ee6f7f7bb52587e1f7e60173/android-nrf-mesh-library/meshprovisioner/src/main/java/no/nordicsemi/android/meshprovisioner/utils/MeshParserUtils.java#L357

iOS:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/68dc144aabdf99da472b809115d94358b843343d/nRFMeshProvision/Classes/Layers/Access%20Layer/AccessPdu.swift#L151

**To Reproduce**
Steps to reproduce the behavior:
1. View Op code computing code

**Expected behavior**
Should be logically equivalent.

**Platform details:**
N/A

**Logs / Screenshots**
N/A

philips77, 2020-04-08:
What Op Code is incorrectly computed? Do you have any example?

kiyoshi-brilliant, 2020-04-08:
@philips77 For opCode: **0xC12008**,
Android's code would fall in the first case and run
``` return new byte[]{(byte) ((opCode >> 16) & 0xFF), (byte) ((opCode >> 8) & 0xFF), (byte) (opCode & 0xFF)};```
iOS' would fall in the default case:
```
accessPdu = Data([ 
                 UInt8(0xC0 | ((opCode >> 16) & 0x3F)), 
                 UInt8((opCode >> 8) & 0xFF), 
                 UInt8(opCode & 0xFF) 
                 ]) + parameters 
```

**The first byte here would be 0xFF in Android, but 0x3F in iOS.**
I don't know which is correct, so I filed this bug on both repos.

Overall comparing the two code chunks:
Android:
```public static byte[] getOpCodes(final int opCode) {
        if ((opCode & 0xC00000) == 0xC00000) {
            return new byte[]{(byte) ((opCode >> 16) & 0xFF), (byte) ((opCode >> 8) & 0xFF), (byte) (opCode & 0xFF)};
        } else if ((opCode & 0xFF8000) == 0x8000) {
            return new byte[]{(byte) ((opCode >> 8) & 0xFF), (byte) (opCode & 0xFF)};
        } else {
            //return new byte[]{ (byte) ((opCode >> 8) & 0xFF), (byte) (opCode & 0xFF)};
            return new byte[]{(byte) opCode};
        }
    }```
iOS:
```
switch opCode {
        case let opCode where opCode < 0x80:
            accessPdu = Data([UInt8(opCode & 0xFF)]) + parameters
        case let opCode where opCode < 0x4000 || opCode & 0xFFFC00 == 0x8000:
            accessPdu = Data([UInt8(0x80 | ((opCode >> 8) & 0x3F)), UInt8(opCode & 0xFF)]) + parameters
        default:
            accessPdu = Data([
                UInt8(0xC0 | ((opCode >> 16) & 0x3F)),
                UInt8((opCode >> 8) & 0xFF),
                UInt8(opCode & 0xFF)
                ]) + parameters
        }```

The cases / if logic don't seem equivalent? Are they supposed to be?

philips77, 2020-04-09:
In the example you provided, the first byte of the returned array would be `0xC1 & 0xFF = 0xC1`.
For iOS, the given code would do really in `0xC0 | (0xC1 & 0x3F) = 0xC0 | 0x01 = 0xC1`.
The 0xC0 prefix in the first byte is to indicate that the op code is a 3-byte vendor messages op code.

kiyoshi-brilliant, 2020-04-09:
@philips77  Oh hm missed the 0xC0 |
Does seem to be correct, will close out this bug.
We resolved our issue between the two clients by passing the op code without the company identifier to the Android VendorModelMessageAcked, though in iOS we able to use our opCode with the company identifier for implementing StaticAcknowledgedMeshMessage. Do you happen to have any insight into which behavior is expected?

Thanks!!

closed 2020-04-09 by kiyoshi-brilliant