nordicsemi/IOS-nRF-Mesh-Library #261: Fatal error: Can't form Range with upperBound < lowerBound

deverickasih, 2020-03-27:
**Describe the bug**
I am trying to create a new provisioner giving it 1 address only, and no addresses for group & scene when i came across a fatal error when setting : 
`
let nextAddressRange = meshNetwork.nextAvailableUnicastAddressRange(ofSize: 0x0001)
`

the error is found in MeshNetwork+Ranges.swift line : 264
`
return RangeObject(lastUpperBound + 1...lastUpperBound + size - 1)
`

what i believed happen is that the above becomes the following : 
`
return RangeObject(6554 + 1...6554 + 1 - 1)
`
The fatal error disappears once i change my code to : 
`
let nextAddressRange = meshNetwork.nextAvailableUnicastAddressRange(ofSize: 0x0002)
`

This also happens for groupAddressRange & sceneRange
`
let nextGroupRange = meshNetwork.nextAvailableGroupAddressRange(ofSize: 0x0000)
`
`
let nextSceneRange = meshNetwork.nextAvailableSceneRange(ofSize: 0x0000)
`
I have to change them to `(ofSize: 0x0002)` to work

In #211 you've stated that there was going to be a removal of validation checks, in particular, for scene and group. Is this already updated yet ? As for nextAddressRange, why is there a need for ofSize: 2 ? Shouldn't it be ofSize: 1 ?

**Question 2**
I'd also like to ask, how can the mesh network know that there is a second phone using the same provisioner in such a way that that second phone can't interact with the mesh network when the first phone is interacting with the mesh network ? (i have two phones pointing to the provisioner @ index 0 from the mesh JSON file) How come the second phone is unable of communicating with the mesh network and how do we prompt the user that is using the second phone, telling him/her that there is another phone using the provisioner @ index 0 ?

philips77, 2020-03-27:
Regarding the bug report, this indeed looks like a bug in the library. I will fix it in the next release (release date unknown). Thank you very much for the report.

Regarding your question. Each device (also each phone) must use a different Unicast Address, as each message sent from one Unicast Address must have Sequence number greater than the last message sent from this error. If you use the same provisioner on 2 phones, and one of them is sending messages, it increases the seq number for this address. Later on, when you try to send messages from the other phone, it will use it's old counter. All other devices will disregard the messages, as the Seq requirement is not fulfilled. This requirement is to prevent from replay attacks, when the same message could be used multiple times. However, if you will try to send more and more messages on the other phone, at some point the Seq number will be bigger than on the 1st phone, so now only this phone will be able to send messages.
To avoid this, each phone should be a different provisioner (or at least have different Unicast Address), but it's much easier to just create a new provisioner object.

philips77, 2020-04-08:
I took nRF Mesh on my plate now. You're first issue is now fixed on *develop* branch. I also added some tests with edge cases, hopefully you'll not find any more issues with this. I should release next version eight today, or next week.

closed 2020-04-08 by philips77