nordicsemi/IOS-nRF-Mesh-Library #262: Cannot get Composition Data

jcarioti, 2020-03-27:
**Describe the bug**
After provisioning, I am unable to get the composition data for my device consistently. 1/10 tries is successful, but the rest result in request timeouts.

**To Reproduce**
Browse for devices, select a device and provision, then request composition data. I don't seem to be getting this issue when using the NRF Mesh iOS app, but my custom code that's based on NRF Mesh doesn't seem to be working consistently.

**Expected behavior**
Composition data is retrieved consistently.

**Platform details:**
iPhone 6s Plus
iOS 13.4

**Logs / Screenshots**
```
-> GET COMPOSITION DATA
Mesh Network Manager Log: Sending ConfigCompositionDataGet(page: 0) to: 0005 foundationModel info
Mesh Network Manager Log: Sending Access PDU (opcode: 0x8008, parameters: 0x00) access info
Mesh Network Manager Log: Sending Upper Transport PDU (encrypted data: 0xE47AC8, transMic: 0x2236133E) encrypted using key: BLETestDevice's Device Key upperTransport info
Mesh Network Manager Log: Sending Access Message (akf: 0, szmic: 0, data: 0xE47AC82236133E) lowerTransport info
Mesh Network Manager Log: Sending Network PDU (ivi: 0, nid: 0x50, ctl: 0, ttl: 5, seq: 27, src: 0001, dst: 0005, transportPdu: 0x912DA33F242167FB, netMic: 0x3A3C4CAC) encrypted using Primary Network Key (index: 0) network info
Mesh Network Manager Log: The bearer is closed. lowerTransport warning
Mesh Network Manager Log: Resending Access PDU (opcode: 0x8008, parameters: 0x00) access debug
Mesh Network Manager Log: Sending Upper Transport PDU (encrypted data: 0x8929F9, transMic: 0xE0C0F116) encrypted using key: BLETestDevice's Device Key upperTransport info
Mesh Network Manager Log: Sending Access Message (akf: 0, szmic: 0, data: 0x8929F9E0C0F116) lowerTransport info
Mesh Network Manager Log: Sending Network PDU (ivi: 0, nid: 0x50, ctl: 0, ttl: 5, seq: 28, src: 0001, dst: 0005, transportPdu: 0x0C632CB1C84EBA32, netMic: 0xB6310CD9) encrypted using Primary Network Key (index: 0) network info
Mesh Network Manager Log: The bearer is closed. lowerTransport warning
Mesh Network Manager Log: Resending Access PDU (opcode: 0x8008, parameters: 0x00) access debug
Mesh Network Manager Log: Sending Upper Transport PDU (encrypted data: 0x63F6A8, transMic: 0xD1D05DDD) encrypted using key: BLETestDevice's Device Key upperTransport info
Mesh Network Manager Log: Sending Access Message (akf: 0, szmic: 0, data: 0x63F6A8D1D05DDD) lowerTransport info
Mesh Network Manager Log: Sending Network PDU (ivi: 0, nid: 0x50, ctl: 0, ttl: 5, seq: 29, src: 0001, dst: 0005, transportPdu: 0xE8418A00C06B90E1, netMic: 0x2AF37FA5) encrypted using Primary Network Key (index: 0) network info
Mesh Network Manager Log: The bearer is closed. lowerTransport warning
Mesh Network Manager Log: Resending Access PDU (opcode: 0x8008, parameters: 0x00) access debug
Mesh Network Manager Log: Sending Upper Transport PDU (encrypted data: 0x5D598A, transMic: 0x4585C29F) encrypted using key: BLETestDevice's Device Key upperTransport info
Mesh Network Manager Log: Sending Access Message (akf: 0, szmic: 0, data: 0x5D598A4585C29F) lowerTransport info
Mesh Network Manager Log: Sending Network PDU (ivi: 0, nid: 0x50, ctl: 0, ttl: 5, seq: 30, src: 0001, dst: 0005, transportPdu: 0xFF23724B0FFB35FD, netMic: 0x35676501) encrypted using Primary Network Key (index: 0) network info
Mesh Network Manager Log: The bearer is closed. lowerTransport warning
Mesh Network Manager Log: Response to Access PDU (opcode: 0x8008, parameters: 0x00) not received (timeout) access warning
Mesh Network Manager Log: ConfigCompositionDataGet(page: 0) sent from: 0001, to: 0005 timed out foundationModel warning
Mesh Network Manager Log: Cancelling messages with op code: 32776, sent from: 0001 to: 0005 access info
-> ERROR: Request timed out.
-> FAILED TO GET COMPOSITION DATA
```

philips77, 2020-03-28:
Add you see, the bearer is closed. Call `bearer.open()`.

geza-pycom, 2020-04-12:
Hello,
I am experiencing the same issue both on iPhone and iPad ios 13.4.1.
How can I see the log ? On Android the log is automatically displayed.
The same device works well with Android app, Composition Data cannot be fetched only when using ios application.

philips77, 2020-04-13:
In the nRF Mesh app, go to Proxy tab, and ensure you're connected. In automatic mode, the app should connect to the first device advertising Network ID. If automatic connection is disabled there, click the 2nd row to select a device to connect to.
If you don't see any devices, make sure at least one work Gatt proxy is provisioned and it's advertising, e.g. not connected to other devices.

jcarioti, 2020-04-14:
> As you see, the bearer is closed. Call `bearer.open()`.

Originally I was having this error because the firmware on our device was crashing. I've fixed that firmware issue, but I'm still having intermittent issues when getting composition data or getting TTL information. I've compared my implementation to the NRF mesh iOS app and cannot figure out why it's not 100% consistent.

Here's my log when everything works as expected (the first thing I do is reset the mesh network):

```
 RESET MESH
[Bearer] (verbose) Cancelling connection...
Creating provisioner instance
Creating new mesh network
2020-04-13 18:21:31.180630-0700 BLETestApp[582:106351] [CoreBluetooth] XPC connection invalid
Saving mesh network
Closing connection
[13034754: BLETestApp.BLETestDeviceColorStatus, 12707074: BLETestApp.BLETestDeviceType, 12838146: BLETestApp.BLETestDeviceStatus]
Saving local elements
Creating NetworkConnection from mesh network
Set AppDelegate as the NetworkConnection logger delegate
Opening network connection
2020-04-13 18:21:31.200406-0700 BLETestApp[582:106352] [CoreBluetooth] XPC connection invalid
Central Manager start scanning for peripherals
<CBPeripheral: 0x283316260, identifier = F760F657-7237-B255-3D16-E5494392B2E6, name = BLETestDevice, state = disconnected>
<CBPeripheral: 0x283316260, identifier = F760F657-7237-B255-3D16-E5494392B2E6, name = BLETestDevice, state = disconnected>
<CBPeripheral: 0x283316260, identifier = F760F657-7237-B255-3D16-E5494392B2E6, name = BLETestDevice, state = disconnected>

 PROVISIONING 1
[Bearer] (info) Central Manager state changed to .poweredOn
[Bearer] (verbose) Connecting to BLETestDevice...
[Bearer] (info) Connected to BLETestDevice
[Bearer] (verbose) Discovering services...
[Bearer] (verbose) Service found
[Bearer] (verbose) Discovering characteristrics...
[Bearer] (verbose) Data In characteristic found
[Bearer] (verbose) Data Out characteristic found
[Bearer] (verbose) Enabling notifications...
[Bearer] (verbose) Data Out notifications enabled
[Bearer] (info) GATT Bearer open and ready
DEVICE TO PROVISION Unknown DDDDBCDD-C2D1-B3DE-0000-000000000000
Optional([Primary Network Key (index: 0)])
[Provisioning] (info) Requesting Provisioning Capabilities
[Provisioning] (verbose) Sending Provisioning Invite (attention timer: 5 sec)
[Bearer] (debug) -> 0x030005
2020-04-13 18:21:42.238680-0700 BLETestApp[582:106351] [CoreBluetooth] XPC connection invalid
Identifying...
[Bearer] (debug) <- 0x03010200010000000000000000
[Provisioning] (verbose) Device Capabilities: Number of elements: 2
Algorithms: FIPS P-256 Elliptic Curve
Public Key Type: None
Static OOB Type: None
Output OOB Size: 0
Output OOB Actions: None
Input OOB Size: 0
Input OOB Actions: None received
[Provisioning] (info) Provisioning Capabilities received
Number of elements: 2
Algorithms: FIPS P-256 Elliptic Curve
Public Key Type: None
Static OOB Type: None
Output OOB Size: 0
Output OOB Actions: None
Input OOB Size: 0
Input OOB Actions: None
[Provisioning] (info) Provisioning started
[Provisioning] (verbose) Sending Provisioning Start (algorithm: FIPS P-256 Elliptic Curve, public Key: No OOB Public Key, authentication Method: No OOB)
[Bearer] (debug) -> 0x03020000000000
[Provisioning] (verbose) Sending Provisioner Public Key (0xCDE1FD0FC2B3BBFA3D3ECEDFC892B0604D5C9ADED1B90E544BB26114E2D01A2DD55BCBE6526F0AA4E611D06C1C839BC07F26991A2D41E6021D6A79303843ABC2)
[Bearer] (debug) -> 0x4303CDE1FD0FC2B3BBFA3D3ECEDFC892B0604D5C
[Bearer] (debug) -> 0x839ADED1B90E544BB26114E2D01A2DD55BCBE652
[Bearer] (debug) -> 0x836F0AA4E611D06C1C839BC07F26991A2D41E602
[Bearer] (debug) -> 0xC31D6A79303843ABC2
[... 350 lines cut ...]
[Bearer] (debug) -> 0xC0EDF6
[Access] (debug) Resending Access PDU (opcode: 0x8008, parameters: 0x00)
[UpperTransport] (info) Sending Upper Transport PDU (encrypted data: 0x75FE84, transMic: 0xBA8FB03B) encrypted using key: BLETestDevice's Device Key
[LowerTransport] (info) Sending Access Message (akf: 0, szmic: 0, data: 0x75FE84BA8FB03B)
[Network] (info) Sending Network PDU (ivi: 0, nid: 0x13, ctl: 0, ttl: 5, seq: 7, src: 0001, dst: 0003, transportPdu: 0x3946640AABB10F9A, netMic: 0x9F041ED3) encrypted using Primary Network Key (index: 0)
[Bearer] (debug) -> 0x40137BE55B803F21422C3946640AABB10F9A9F04
[Bearer] (debug) -> 0xC01ED3
[Access] (warning) Response to Access PDU (opcode: 0x8008, parameters: 0x00) not received (timeout)
[FoundationModel] (warning) ConfigCompositionDataGet(page: 0) sent from: 0001, to: 0003 timed out
[Access] (info) Cancelling messages with op code: 32776, sent from: 0001 to: 0003
FAILURE SENDING MESSAGE
Request timed out.
```

I've been spending hours and hours on this and I cannot figure out why it's failing. It seems like a race condition based on some testing I've done. When I delay before requesting composition data, it almost always fails. Also, when I instantiate the mesh network using the main Dispatch Queue, it always fails. 

When the process is failing, I see these logs from the firmware side:

```
W (24341) BLE_MESH: Ran out of retransmit attempts
W (35021) BLE_MESH: Ran out of retransmit attempts
W (56261) BLE_MESH: Ran out of retransmit attempts
```

Hopefully someone can help point me in the right direction. This has really been driving me crazy! 😞

philips77, 2020-04-14:
Ok, I know why.
Seems like your device (proxy) is only able to have 1 unicast address in the white list filter. First of all, I think I have "fixed" this in the latest release of the library. Have a look at #212. **Are you sure you are using version 2.2.1?** 

> FYI: 
There will be 2.2.2 today, which will fix the IV Index Update procedure, which doesn't work so far.

Have a look at your "success" log:
```
[Proxy] (info) New Proxy connected

// Sending Proxy Filter type: Whitelist:

[Proxy] (info) Sending SetFilterType(filterType: Whitelist) from: 0001 to: 0000
[Network] (info) Sending Control Message (opCode: 0x00, data: 0x00)
[Network] (info) Sending Network PDU (ivi: 0, nid: 0x2B, ctl: 1, ttl: 0, seq: 1, src: 0001, dst: 0000, transportPdu: 0x6742, netMic: 0xEB54E914F93D8950) encrypted using Primary Network Key (index: 0)
[Bearer] (debug) -> 0x022BB39724BC3D02BCDD6742EB54E914F93D8950

// Receiving confirmation, that the filter was set and it's empty:

[Bearer] (debug) <- 0x422B61DD9AFFCB73185A6D9FA785C9B1FB068A9E
[Bearer] (debug) <- 0xC2D698
[Network] (info) Network PDU (ivi: 0, nid: 0x2B, ctl: 1, ttl: 0, seq: 0, src: 0003, dst: 0000, transportPdu: 0x6D9FA785, netMic: 0xC9B1FB068A9ED698) received
[Network] (info) Control Message (opCode: 0x03, data: 0x000000) receieved (decrypted using key: Primary Network Key (index: 0))
[Proxy] (info) FilterStatus(filterType: Whitelist, listSize: 0) received from: 0003 to: 0000

// Your provisioner has 2 elements with addresses 0x0001 and 0x0002. So far they aren't added
// to any group. The library tries to add those 2 addresses + "all nodes" address to the whitelist.
// This is because we want your app to receive all messages targeting any of the elements, 
// and any of the groups that any model if subscribed to (that includes "all nodes").
// "All nodes" address is 0xFFFF, that is 65535 in decimal.

[Proxy] (info) Sending AddAddressesToFilter(addresses: Set([65535, 1, 2])) from: 0001 to: 0000
[Network] (info) Sending Control Message (opCode: 0x01, data: 0xFFFF00010002)
[Network] (info) Sending Network PDU (ivi: 0, nid: 0x2B, ctl: 1, ttl: 0, seq: 2, src: 0001, dst: 0000, transportPdu: 0xFF30311CB67E1D, netMic: 0xAE69CFD4BCE3C71E) encrypted using Primary Network Key (index: 0)
[Bearer] (debug) -> 0x422BAB3E0E05499A35ABFF30311CB67E1DAE69CF
[Bearer] (debug) -> 0xC2D4BCE3C71E

// Receiving confirmation:

[Bearer] (debug) <- 0x422B290222E724B0F80D87E9D037907B1C69DE80
[Bearer] (debug) <- 0xC2285B
[Network] (info) Network PDU (ivi: 0, nid: 0x2B, ctl: 1, ttl: 0, seq: 1, src: 0003, dst: 0000, transportPdu: 0x87E9D037, netMic: 0x907B1C69DE80285B) received
[Network] (info) Control Message (opCode: 0x03, data: 0x000001) receieved (decrypted using key: Primary Network Key (index: 0))
[Proxy] (info) FilterStatus(filterType: Whitelist, listSize: 1) received from: 0003 to: 0000
[Proxy] (warning) Refreshing Proxy Filter...

// Ups! The returned listSize is 1, not 3! The new version of the library should focus on registering 
// at least the primary element address, 0x0001. Instead, yours is retrying registering all again.
// First, let's clear the list by setting the filter to whitelist again:

[Proxy] (info) Sending SetFilterType(filterType: Whitelist) from: 0001 to: 0000
[Network] (info) Sending Control Message (opCode: 0x00, data: 0x00)
[Network] (info) Sending Network PDU (ivi: 0, nid: 0x2B, ctl: 1, ttl: 0, seq: 3, src: 0001, dst: 0000, transportPdu: 0x8238, netMic: 0x5A484BC7CB251BE5) encrypted using Primary Network Key (index: 0)
[Bearer] (debug) -> 0x022BA4C4F11E63957B6582385A484BC7CB251BE5

// Confirmation...

[Bearer] (debug) <- 0x422B539F3181C2981C24ED8C5162E6303DD61348
[Bearer] (debug) <- 0xC23C53
[Network] (info) Network PDU (ivi: 0, nid: 0x2B, ctl: 1, ttl: 0, seq: 2, src: 0003, dst: 0000, transportPdu: 0xED8C5162, netMic: 0xE6303DD613483C53) received
[Network] (info) Control Message (opCode: 0x03, data: 0x000000) receieved (decrypted using key: Primary Network Key (index: 0))
[Proxy] (info) FilterStatus(filterType: Whitelist, listSize: 0) received from: 0003 to: 0000

// And... 2.2.1 should add only address 1. Unless, there's a bug. You add 1, 65535 and 2 
// (but 1 is first, which is important! That's the reason it works later!).

[Proxy] (info) Sending AddAddressesToFilter(addresses: Set([1, 65535, 2])) from: 0001 to: 0000
[Network] (info) Sending Control Message (opCode: 0x01, data: 0x0001FFFF0002)
[Network] (info) Sending Network PDU (ivi: 0, nid: 0x2B, ctl: 1, ttl: 0, seq: 4, src: 0001, dst: 0000, transportPdu: 0xD0F15FC6A4F39B, netMic: 0x275F6E27A29129A5) encrypted using Primary Network Key (index: 0)
[Bearer] (debug) -> 0x422B183D5816B9E53F31D0F15FC6A4F39B275F6E
[... 15 lines cut ...]
In case of the log showing failure, the addresses added in the 2nd attempt were 0xFFFF, 0x0002 and 0x0001 (notice different order in the first line below).
```
// Retrying:

[Proxy] (info) Sending AddAddressesToFilter(addresses: Set([65535, 2, 1])) from: 0001 to: 0000
[Network] (info) Sending Control Message (opCode: 0x01, data: 0xFFFF00020001)
[Network] (info) Sending Network PDU (ivi: 0, nid: 0x13, ctl: 1, ttl: 0, seq: 4, src: 0001, dst: 0000, transportPdu: 0xB4C3501FDAFB0D, netMic: 0x42A733361799E6E0) encrypted using Primary Network Key (index: 0)
[Bearer] (debug) -> 0x421325540BA48E25A5DBB4C3501FDAFB0D42A733
[Bearer] (debug) -> 0xC2361799E6E0

// Receiving confirmation:

[Bearer] (debug) <- 0x4213B6540FC70217FE2CF7D3981A28BDE9540A2C
[Bearer] (debug) <- 0xC20587
[... cut ...]
```
Finally, the 0xFFFF is registered. So when you send messages from 0x0001, the reply is never received, as it's targeting the source address, not "all nodes", and your proxy does not send them over GATT Bearer.

**Solutions:**
1. Update to the latest version 2.2.1. If you are already in this version, please let me know so I can fix it in the 2.2.2.
2. Fix your device, allow it to support multiple addresses on the proxy filter.
3. Configure the proxy filter manually, by using `ProyFilter` class.

philips77, 2020-04-14:
The `addresses` property in the `ProxyFilter` class is a `Set<Address>`, so the actual order sent each time is random.
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/45e12224e8b5f2d45659e99dddda966b1ba17789/nRFMeshProvision/Classes/ProxyFilter.swift#L101-L102

jcarioti, 2020-04-14:
It looks like I have version 2.0.0 in my Podfile.lock. I think the reason I have this version is because with 2.2.1 I am getting build errors:

```
./Pods/OpenSSL-Universal/ios/lib/libcrypto.a(cmac.o)' does not contain bitcode. You must rebuild it with bitcode enabled (Xcode setting ENABLE_BITCODE), obtain an updated library from the vendor, or disable bitcode for this target. for architecture arm64
```

I don't want to have to disable bitcode. Any idea what to do here?

philips77, 2020-04-14:
How about this comment: https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/issues/191#issuecomment-548764309 and also specifying version of the mesh library?

philips77, 2020-04-14:
Also, have a look at #196.

jcarioti, 2020-04-14:
> Also, have a look at #196.

Here's what ended up working for me:

```
  pod 'nRFMeshProvision'
  pod 'OpenSSL-Universal/Framework'
```

No specifying version numbers at all. Bitcode is enabled. Thanks for your help!

closed 2020-04-16 by philips77

jcarioti, 2020-04-16:
Unfortunately, I spoke too soon. Builds with bitcode are working on the simulator only. I still have build failures related to OpenSSL and bitcode when building for a physical device.

For now the only way I can get nRF to compile with 2.2.1 on a physical is to disable bitcode. I've tried 'OpenSSL-Universal' and 'OpenSSL-Universal/Framework'. Both have the same issue when building for a physical device.