nordicsemi/IOS-nRF-Mesh-Library #263: Node reset without removing

fiveagle, 2020-04-06:
Hi,

This time I was wondering why when we send a reset message the target node gets removed from the MeshNetwork object since in the example app looks like I have to manually do it after the message, calling `meshNetworkManager.meshNetwork?.remove(node: deletingNode!)`.

I need to not cancel it after a reset because it looks like that after clearing their Unicast Address the app can eventually use that same address for a new node but all the other nodes in the mesh would not be able to talk whit the new one because of the wrong sequence number. Is it true?
Anyway can I register my self as ModelDelegate for a node model in order to prevent the local deletion?

Thanks a lot.

philips77, 2020-04-06:
The library will remove the node automatically if response to `ConfigResetNode` is received. Very often the device resets before it is able to reply, or the response is lost on its way, so the app still waits and waits for an answer and timeouts eventually. In that case the node is not removed automatically, hence the method to delete a node manually (if user verifies that it actually has reset).

And yes, after reset, the library allows to reuse the same Unicast Address, which can trigger problems, as you said so. The reason for that is that the json db does not store "used" addresses, and it is in your responsibility to avoid assigning the same address twice, or incrementing the Sequence Number for such node to avoid the problem.

fiveagle, 2020-04-06:
> The reason for that is that the json db does not store "used" addresses

Ok, is this something that will be implemented in the future or is completely out task to manage this scenario? I'm implementing my list also for ensure that I do not create 2 provisioners on the same address.

philips77, 2020-04-06:
The json schema is developed by Bluetooth SIG and is still not released. I saw some changes recently, but didn't track what were there related to. Also, keep in mind, that when the spec is released, we will release an update for the library with final json schema. There will be option to migrate version, but the new version will only export in the updated schema.

philips77, 2020-04-06:
Hmm.. There's "blacklisted" parameter in the Node. Perhaps resetting a node should not remove it, but rather change this flag to *true*, as it just claims that it has reset. To truly make sure a Node has been removed you need to update all the Network Keys (App Keys change is also recommended) in all other Nodes, so this Node cannot understand the communication anymore.

fiveagle, 2020-04-07:
I'm not changing the key.
My need is simply to not get an used in the past unicast from `nextAvailableUnicastAddress()`
For now I have implemented a blacklist which I save along side the json.
Thanks

closed 2020-04-10 by fiveagle